universam1/iSpindel #256: 6.0.4 - cloud service parameters not saved

ploufe, 2019-02-18:
Hi,

Since I updated my device in 6.0.4. version I can no longer saved my service cloud configuration.
I'd like to use a HTTP server.
I write all config parameters, clic on "save" button, 
but after reboot my ispindle always display "Ubidot" (instead of "HTTP")...

Any idea to solve it?

universam1, 2019-02-18:
Please send a log output

ckrack, 2019-02-24:
I can confirm the problem on 6.0.2

I'm posting my log - I set the TCP server yesterday and upon first switching the device on it sends data correctly.

```log
12:32:44.040 -> FW 6.0.2
12:32:44.040 -> 3.0.0-dev(c0f7b44)
12:32:44.040 -> Worker run!
12:32:44.040 -> mounting FS... mounted!
12:32:44.040 -> reading config file
12:32:44.040 -> offsets not available
12:32:44.040 -> parsed config:
12:32:44.075 -> {"Name":"Spindel_Stern","Token":"foobar","Sleep":900,"Server":"example.com","API":6,"Port":61085,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Foobar","PSK":"foobar","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake
12:32:44.248 -> 
12:32:44.248 -> woken from deepsleep, normal mode
Samples:42 min:63.64 max:63.67 time:755
12:32:45.014 -> x: -9992 y: 12492 z: -1928
12:32:45.014 -> Tilt: 63.66
12:32:45.014 -> Tacc: 16.72
12:32:45.014 -> Volt: 5.34
12:32:45.048 -> Temp: 19.44
12:32:45.048 -> Gravity: 20.15
12:32:45.048 -> After waiting 201ms, result 3
12:32:45.255 -> IP: 192.168.178.40
12:32:45.255 -> 
12:32:45.255 -> calling TCP
12:32:45.255 -> {"name":"Spindel_Stern","ID":3937792,"token":"foobar","angle":63.66018,"temperature":19.4375,"temp_units":"C","battery":5.338894,"gravity":20.14841,"interval":900,"RSSI":-68}
12:32:45.322 -> Sender: TCP stream

12:32:47.485 -> Final-sleep: 900s; RT: 3512
rld⸮⸮|⸮l⸮|⸮d⸮c|⸮⸮⸮⸮r⸮c⸮#⸮⸮no⸮loo⸮⸮⸮cx⸮⸮drlslp⸮o⸮⸮l⸮⸮co⸮|l⸮⸮#⸮⸮no⸮$⸮⸮l`⸮ool`gr⸮⸮⸮'c⸮l{ےgc⸮l⸮l`⸮⸮o⸮
12:34:41.982 -> FW 6.0.2
12:34:41.982 -> 3.0.0-dev(c0f7b44)
12:34:41.982 -> Worker run!
12:34:41.982 -> mounting FS... mounted!
12:34:41.982 -> reading config file
12:34:41.982 -> offsets not available
12:34:41.982 -> parsed config:
12:34:41.982 -> {"Name":"Spindel_Stern","Token":"foobar","Sleep":900,"Server":"example.com","API":6,"Port":61085,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Foobar","PSK":"foobar","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
rld⸮⸮|⸮d⸮|⸮d⸮c|⸮⸮⸮⸮s⸮c⸮b⸮⸮oo⸮lgg⸮⸮⸮cp⸮dslslp⸮g⸮⸮d⸮⸮cg⸮|l⸮⸮b⸮⸮oo⸮l⸮⸮l`⸮ool`gsǓ⸮nc⸮lr⸮⸮gc⸮l⸮l ⸮⸮o⸮
12:34:42.177 -> FW 6.0.2
12:34:42.177 -> 3.0.0-dev(c0f7b44)
12:34:42.177 -> Worker run!
12:34:42.211 -> mounting FS... mounted!
12:34:42.211 -> reading config file
12:34:42.211 -> offsets not available
12:34:42.211 -> parsed config:
12:34:42.211 -> {"Name":"Spindel_Stern","Token":"foobar","Sleep":900,"Server":"example.com","API":6,"Port":61085,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Foobar","PSK":"foobar","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
;ld⸮⸮|⸮l⸮|⸮d⸮b|⸮⸮⸮⸮r⸮b⸮c⸮⸮no⸮$oo⸮⸮⸮cx⸮⸮drlrlp⸮o⸮⸮l⸮⸮co⸮|l⸮⸮c⸮⸮no⸮d⸮⸮l`⸮ool`gr⸮⸮⸮gc⸮l{⸮⸮gc⸮l⸮l`⸮⸮n⸮
12:34:42.630 -> FW 6.0.2
12:34:42.630 -> 3.0.0-dev(c0f7b44)
12:34:42.630 -> Worker run!
12:34:42.663 -> mounting FS... mounted!
12:34:42.663 -> reading config file
12:34:42.663 -> offsets not available
12:34:42.663 -> parsed config:
12:34:42.663 -> {"Name":"Spindel_Stern","Token":"foobar","Sleep":900,"Server":"example.com","API":6,"Port":61085,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Foobar","PSK":"foobar","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
rld⸮⸮|⸮l⸮|⸮d⸮c|⸮⸮⸮⸮r⸮c⸮#⸮⸮no⸮$oo⸮⸮⸮cx⸮drlslp⸮o⸮⸮l⸮⸮co⸮|l⸮⸮#⸮⸮no⸮$⸮⸮l`⸮ool`gr⸮⸮⸮'c⸮l{ےgc⸮l⸮l`⸮⸮o⸮
12:34:43.021 -> FW 6.0.2
12:34:43.021 -> 3.0.0-dev(c0f7b44)
12:34:43.021 -> Worker run!
12:34:43.021 -> mounting FS... mounted!
12:34:43.021 -> reading config file
offsets not available
12:34:43.058 -> parsed config:
12:34:43.058 -> {"Name":"Spindel_Stern","Token":"foobar","Sleep":900,"Server":"example.com","API":6,"Port":61085,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"Foobar","PSK":"foobar","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":⸮{dl⸮⸮|⸮l⸮|⸮l⸮c|⸮⸮⸮⸮{⸮c⸮c⸮⸮gg⸮dno⸮⸮⸮cp⸮l{d{dp⸮n⸮⸮l⸮⸮bn⸮|d⸮⸮c⸮⸮gg⸮d⸮⸮d`⸮gnd`o{⸮⸮⸮gc⸮l{⸮⸮o#⸮d⸮l`⸮⸮g⸮
12:34:43.334 -> FW 6.0.2
12:34:43.334 -> 3.0.0-dev(c0f7b44)
12:34:43.334 -> Worker run!
[... 105 lines cut ...]
I am connected to the captive right now and it displays Ubidots as service type.
After the timeout and without saving anything on the config, it sends data correctly again.

```log
2:39:56.873 -> *WM: 5.44
12:39:56.873 -> *WM: seconds
12:39:56.873 -> *WM: Timed out connection result: 
12:39:56.873 -> *WM: WL_CONNECTED
12:39:56.873 -> Samples:30 min:80.49 max:80.59 time:514
12:39:56.873 -> x: -2476 y: 15942 z: -1328
12:39:56.873 -> Tilt: 80.53
12:39:56.873 -> Tacc: 24.04
12:39:56.873 -> Volt: 5.34
12:39:56.873 -> Temp: 19.19
12:39:56.873 -> Parse error at 1
12:39:56.873 -> Gravity: 0.00
12:39:56.873 -> IP: 192.168.178.40
12:39:56.873 -> 
12:39:56.873 -> calling TCP
12:39:56.873 -> {"name":"Spindel&#95;Stern","ID":3937792,"token":"foobar","angle":80.53284,"temperature":19.1875,"temp_units":"C","battery":5.338894,"gravity":0,"interval":900,"RSSI":-65}
12:39:56.873 -> Sender: TCP stream
12:39:56.873 -> 
12:39:56.873 -> 
12:39:56.873 -> Final-sleep: 900s; RT: 310959
```

ploufe, 2019-02-24:
Thanks because I don't know how to get the log :)

universam1, 2019-02-27:
@ckrack Would you therefor state that the functionality is fine, only displaying might be wrong?

@ploufe See the FAQ doc how to get the terminal debug log

ckrack, 2019-02-27:
Hi @universam1.
Yes, I could not reproduce an actually incorrect saving - although I thought there was at first, when I "lost" a floating spindle to Ubidots.

It's a problem with the mirroring between the form inputs on the config page.
There is one `<select #API>` and one <input hidden #selAPI>`.
I saved the HTML of the config locally and it works fine, TCP is selected after loading.
However, when loaded via the Captive on the actual device, it's not selected.

I debugged it in the console in several breakpoints.

````
document.getElementById('selAPI').value

"6"
document.getElementById('API').value

""
document.getElementById('API').value

""
document.getElementById('API').value

""
document.getElementById('API').value = document.getElementById('selAPI').value;

"6"
document.getElementById('API').value

""
document.getElementById('API').value

"0"
document.getElementById('API').value = document.getElementById('selAPI').value;

"6"
```

Seems like setting the value of #API (which is the select) does not work as expected.

ckrack, 2019-02-27:
Actually this should work in all browsers - the problem is, that the `<option>`s are rendered twice.
This might confuse browsers.. My fix uses `querySelector`, which will stop after the first option it found.

```html
<select id="API" onchange="sAPI(this.options[this.selectedIndex].value);"><option value="0">Ubidots</option><option value="2">CraftBeerPi</option><option value="3">HTTP</option><option value="4">TControl</option><option value="5">FHEM</option><option value="6">TCP</option><option value="7">iSpindel.de</option><option value="8">InfluxDB</option><option value="9">Prometheus</option><option value="10">MQTT</option><option value="0">Ubidots</option><option value="2">CraftBeerPi</option><option value="3">HTTP</option><option value="4">TControl</option><option value="5">FHEM</option><option value="6">TCP</option><option value="7">iSpindel.de</option><option value="8">InfluxDB</option><option value="9">Prometheus</option><option value="10">MQTT</option></select>
```

ploufe, 2019-02-28:
I'm totaly unskill to understand the discution, but I just share that I downgrade my ispindle on 6.0.1 version and it solve the issue

ckrack, 2019-02-28:
I think this addition leads to the options appearing twice:
https://github.com/universam1/iSpindel/commit/07141b9d9e2a3b6fb4122155f7111a4f485f3431#diff-46b0b7d03759ea011d1640d980af8453R73

universam1, 2019-02-28:
Thanks @ckrack for debugging the conditions!

universam1, 2019-03-07:
@ckrack there was another issue with the order of the calls. Fixed with 6.0.5. Please make sure to `reset` under `Maintenance` menu

universam1, 2019-03-14:
Assuming this is fixed

closed 2019-03-14 by universam1