universam1/iSpindel #259: Blynk support not working

tdsnor, 2019-02-20:
Hi. 
Is the Blynk support released in 6.0.4? I was not able to choose this service in the drop down. But I see that Leo has been working with it.

Thanks.

Grandement, 2019-08-11:
Hi,
I am waiting for this support as well. The pull request looks really promosing. I hope we get it soon into the Master.

mr-bridger, 2019-10-04:
Is this in progress? I use blynk and would love to use that instead of ubidots.

Grandement, 2019-10-04:
Hi mr-bridge,
I have compiled the pull request of Idab myself. Worked very good. I modified his version a little bit because I like Plato more than Gravity ;). Hopefully it will be merged to the master one day.

ldab, 2020-02-06:
https://github.com/universam1/iSpindel/pull/264

Topspliff, 2020-03-11:
Hi is there any update on this? I`m also a Blynk user and this would be most excellent to have it as an option, I gave up with Ubidots as I could not work out how to use it, I`m currently using the Thingspeak option, however, I do not get the gravity reading only temp, tilt and batt showing, no idea why any thoughts most appreciated.

Cheers Alan

closed 2020-06-23 by stale[bot]