universam1/iSpindel #261: iSpindel for Wine Making

heliocosta, 2019-02-28:
Hi everyone,

I'm not sure if I should ask this question here, but I can't find anywhere else.

I've noticed that most or everyone has used iSpindel for brewing beer, however, there is a lack of information or at least I haven't found anyone who used it to make wine.

The wine has a different CO2 from a beer. Is that something the calibration will cover?

What should I do to use iSpindel to use it in a winemaking process?

Thanks.

universam1, 2019-02-28:
You are definitely not the first one to use the iSpindel for wine fermentation. From my experience it tends to be even more precise as the CO2 is an error factor in the measurement that is tried to equalize it out of the equation by the calibration. 
Keep us updated about your progress

heliocosta, 2019-03-01:
Thank you very much @universam1 . I'll keep you'all updated for sure. I've ordered 5 boards from @cherryphilip74 4.0 board. 
BTW, this will be the final project that my wife is working on for her Masters: How to use low-cost technologies for oenology.

BulletMag, 2019-06-13:
I can confirm this works fine for wine fermentation, I have successfully used one in a couple of wine brews and it worked like a dream.

universam1, 2019-06-13:
Great 👍 thanks for the feedback!

What windows of SG/Plato sugar level you are dealing with, and how long is fermentation duration around?

BulletMag, 2019-06-13:
Mine began with a SG of 1. and finished around 1.000 that was from a kit (Solomon Grundy Chardonnay) fermentation was over in around four days (as per the kits instructions) I did leave for a couple more days but both hydrometer and iSpindle readings were the same.
This all depends on the calibration being done correctly in the first place of course.

cherryphilip74, 2019-06-13:
> Mine began with a SG of 1.080 and finished around 1.000 that was from a kit (Solomon Grundy Chardonnay) fermentation was over in around four days (as per the kits instructions) I did leave for a couple more days but both hydrometer and iSpindle readings were the same.
> This all depends on the calibration being done correctly in the first place of course.

Did you use the iSpindel-PCB or other hardware?  I'm any case good results

BulletMag, 2019-06-13:
I used a cherryPhillip74 v4 PCB

heliocosta, 2019-07-02:
Great to hear from you that it worked. I’ve tested it as well and also worked pretty well with white wine. I’m not sure it will work with red wine due the grape skin.

The most difficult challenge to me was the calibration, to be honest, I’m not really sure about it yet. Although the documentation is well written, I’d suggest to create a video to teach how to calibrate it. 

I’ve been using cherryphilip v4 as well with exactly the same components.

BulletMag, 2019-07-02:
I have made ten of these in total, fortunately as i used the same components for all devices i only needed to calibrate one device and use the same formula for the others.
The excel sheet is quite flexible so you don't need to hit the exact plato/SG during the calibration just make sure to record your readings for both plato/SG and tilt for each step then feed this into the table

winetech1, 2019-08-22:
Hi guys, have you heard any commercial winery that is using iSpindel? Cheers

closed 2020-06-23 by stale[bot]