universam1/iSpindel #262: Which resistors to die?

ErikdBr, 2019-03-01:
Hi, I am Erik den Breeyen from The Netherlands and planning to built my first iSpindel from scratch and if all goes well built some more for fellow brewers. Ordered all the parts but now I am not sure about which resistors to use. Like metal or carbon, 0.5w 1 w, voltage etc
Also on the iSpindel @Mikmonken v2 pcb at r3 it says 470. I read somewhere that it is better to use 390 or replace it by a Bat 43 diode Schottky.

Please advies.
Greetings, Erik.

Hammersoe, 2019-03-14:
Hi Erik,
The function of the resistor between D0 and RST is to wake up from deep sleep or powersaver for lack of a better word. You can use both the resistors mentioned (470ohm and 390ohm) or you can do the better alternative and choosing the BAT-43 diode.
The reason why you see different usage is related to the voltage required to wake the iSpindel up from sleep and send data at given interval. On voltage of around 3.7volts they all act the same, but when your battery starts to drain and voltage drops then the 470ohm resistor in some cases wont let enough voltage thorugh to RST for the ESP8266 controller to wake up and go through the sendinginterval.
The BAT-43 diode opens on as low as 0.33volt so it will always be able to wake up the device, as long as there is enough juice left on the battery for the ESP8266 controller to function.

None of the components in the design requires any special high voltage or watt toleranse.
You should be good using 1/4watt (0.25w) metal film resistors.

Hope this helps you in your builds:)

ErikdBr, 2019-03-14:
Hi Hammersoe,
That clarifies it all and is really helpful. 
Thanks and Cheers.

closed 2019-05-18 by ErikdBr