universam1/iSpindel #265: derived variable option missing

nzedred, 2019-03-31:
Hi there. 
Im following the instructions for calibration on ubidots but when i get to part about adding the derived variable this option does not appear? i only get raw and synthetic as options.
any help would be appreciated.
cheers.

ErikdBr, 2019-05-25:
Then you are clearly not talking about ubidots because there you wil get the option if you add a new variable to your device.
[image]

nzedred, 2019-05-25:
Hi, thanks for your reply, but yes, I am taking about unibots. I don't have
that option there is no option for derived when I add a new variable.

ErikdBr, 2019-05-25:
Then I suspect you didn't sign up at ubidots for education but you signed up for a trial or a payed account.

I have an account at https://ubidots.com/education/ there you'll have the derived option,
[Ubidots for education derived variable](https://help.ubidots.com/faqs-and-troubleshooting/derived-variables-with-ubidots-for-education)

If you don't want to sign up at Ubidots for Education or if you allready payed for your account I am pretty sure the synthetic variable is your derived variable as it where and you should use that one to calculate SG etc. https://help.ubidots.com/user-guides/analytics-synthetic-variables-basics

nzedred, 2019-05-25:
Ah, so that option isn't available on the trial version.?that explains it
then. Thanks very much!

closed 2019-11-17 by universam1