universam1/iSpindel #270: InfluxDB not working in 6.0.5

pompel82, 2019-04-20:
InfluxDB reporting is not working in 6.0.5. I reverted to 6.0.2 and it solved the issue. Have seen other issues reported on this e.g. #215, #218 but these are closed and fix included (I assume) in 6.0.5 so seems this is another issue or fix is not working. Don't have any info on the error, could not find any trace of connection in influxdb log and no info found in network packet capture so seems iSpindel is not even connecting to the influxdb server.

SunSun1979, 2019-04-21:
+1, have this problem, revert to 6.0.2

finchlt, 2019-05-04:
I did a little debugging on this issue this morning. Seems that the fix for #257 and #258 changed the URL generation logic to build a complete URL string instead of pass in the host, port, uri separately.
Prior (works for me):
``
http.begin(server, port, uri);
``
Currently:
``
http.begin(server + ':' + port + uri);
``
So it seems that for the current code it doesn't correctly parse the protocol. (i.e. http://). The code looks for the first ':' character and since this won't be until after the server string this won't work. If you include http:// in the server host configuerd in the config webpage it works or if you add http:// to the URL string programmatically. However the Http client library usage of the current method signature below is deprecated so the fix is probably revert the #258 merge the figure more what caused PR #257.
``
http.begin(String url);
``
I can put together a pull request later today or tomorrow.

universam1, 2019-05-05:
Thanks that’s welcome!

Antti1111, 2019-06-29:
Is the same problem 6.0.6 ?

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel","Token":"","Sleep":900,"Server":"192.168.0.192","API":8,"Port":8086,"URL":"","DB":"spindel","Username":"spindel","Password":"spindel","Job":"","Instance":"","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"AABBCC","PSK":"12062013","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-691,"aY":-821,"aZ":1903}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:63.53 max:63.54 time:755
x: 12696 y: 12746 z: 922
Tilt: 63.54
Tacc: 26.65
Volt: 4.17
Temp: 27.00
Gravity: 20.08
After waiting 200ms, result 3
IP: 192.168.0.204

calling InfluxDB
Sending to db: spindel w/ credentials: spindel:spindel
INFLUXDB: posting to db: /write?db=spindel
POST data: measurements,source=iSpindel tilt=63.5372,temperature=27,temp_units="C",battery=4.171011,gravity=20.08476,interval=900,RSSI=-62
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 900s; RT: 1413

mharbach, 2019-09-03:
I also still have this issue in 6.1.1. Here's the iSpindel log

`calling InfluxDB
Sending to db: ispindel w/ credentials: ispindel:m76LBQuty8vR
INFLUXDB: posting to db: /write?db=ispindel
POST data: measurements,source=iSpindelMH
code: 400
calling InfluxDB
Sending to db: ispindel w/ credentials: ispindel:m76LBQuty8vR
INFLUXDB: posting to db: /write?db=ispindel
POST data: measurements,source=iSpindelMH
code: 400
`

and here's the influxDB access.log entry:

`109.41.192.51 - ispindel [03/Sep/2019:11:01:29 +0200] "POST /write?db=ispindel HTTP/1.1" 400 77 "-" "ESP8266HTTPClient" 6adecc26-ce29-11e9-85b2-b827ebdf3768 344
`

I guess the probelm might be that the POST data does not include any actual measurements. Accordingly, the access.log reports only 344 bytes in the request. When I create a manual request using curl, the size is somewhere in the 24000 bytes range. here's the curl command I use (except for password and host ;)):

`curl -X POST --data-binary 'measurements,source=iSpindelMH tilt=65.22439,temperature=23.9375,temp_units="C",battery=4.144943,gravity=19.42556,interval=30,RSSI=-42' -u ispindel:mypass http://myhost:18086/write?db=ispindel 
`

TheNetStriker, 2019-09-25:
+1 I can also confirm this problem. Reverted to 6.0.2.

TheNetStriker, 2019-09-26:
I guess I found out what was causing my problem. My post data only contained measurements,source=iSpindel001, so there was no actual data in the http request. I found a simple fix for the problem. I changed the line in SenderClass::sendInfluxDB
from:
for (const auto &kv : _doc.to<JsonObject>())
to:
for (const auto &kv : _doc.as<JsonObject>())

Now the post contains data again and it also get's saved by the InfluxDB.

commit 8150afa9 (2019-10-27): make InfluxDB work again (fixes #270) (#306)
closed 2019-10-27 by universam1