universam1/iSpindel #275: Request ro reopen issue #217 due to this post and following posts

ErikdBr, 2019-05-22:
My console monitor shows after calibration (in Arduino Serial Monitor):

```
Testing device connections...
MPU6050 connection successful
averaging 1000 readings each time
expanding:
	XAccel			YAccel				ZAccel
[0,0] --> [22675,22408]	[0,0] --> [25171,24816]	[0,0] --> [5656,5912]
[-1000,0] --> [13502,22408]	[-1000,0] --> [16975,24816]	[0,0] --> [5654,5912]
[-2000,0] --> [4133,22408]	[-2000,0] --> [8599,24816]	[0,0] --> [5647,5912]
[-3000,0] --> [-5257,22408]	[-3000,0] --> [225,24816]	[0,0] --> [5649,5912]
[-3000,0] --> [-5474,22408]	[-4000,0] --> [-8169,24816]	[0,0] --> [5641,5912]
	XAccel			YAccel				ZAccel
[-3000,0] --> [-5474,22034]	[-4000,0] --> [-8169,24416]	[0,0] --> [5641,5644]
[-3000,0] --> [-5474,22657]	[-4000,0] --> [-8169,25141]	[0,1000] --> [5641,14207]
[-3000,0] --> [-5474,22661]	[-4000,0] --> [-8169,25142]	[0,2000] --> [5641,22945]

closing in:
	XAccel			YAccel				ZAccel
[-3000,-1500] --> [-5474,8894]	[-4000,-2000] --> [-8169,8747]	[1000,2000] --> [14581,22945]
[-3000,-2250] --> [-5474,1694]	[-4000,-3000] --> [-8169,190]	[1000,1500] --> [14581,18674]
[-2625,-2250] --> [-1893,1694]	[-3500,-3000] --> [-4082,190]	[1000,1250] --> [14581,16613]
[-2437,-2250] --> [-238,1694]	[-3250,-3000] --> [-2121,190]	[1125,1250] --> [15491,16613]
[-2437,-2343] --> [-238,666]	[-3125,-3000] --> [-1060,190]	[1187,1250] --> [15992,16613]
	XAccel			YAccel				ZAccel
[-2437,-2390] --> [-238,247]	[-3062,-3000] --> [-516,190]	[1218,1250] --> [16278,16613]
[-2437,-2413] --> [-238,3]	[-3031,-3000] --> [-257,190]	[1218,1234] --> [16278,16422]
[-2425,-2413] --> [-95,3]	[-3015,-3000] --> [-122,190]	[1226,1234] --> [16359,16422]
[-2419,-2413] --> [-43,3]	[-3007,-3000] --> [-55,190]	[1230,1234] --> [16383,16422]
[-2419,-2416] --> [-43,2]	[-3003,-3000] --> [-20,190]	[1230,1232] --> [16383,16401]
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2417,-2416] --> [-17,2]	[-3001,-3000] --> [-7,190]	[1231,1232] --> [16377,16401]
[-2417,-2416] --> [-26,2]	[-3001,-3000] --> [-10,190]	[1231,1232] --> [16364,16401]
-------------- done --------------
a/g:	2	-8	16362
saving config...{"Name":"iSpindel002","Token":"97utjst0v334l8he32pgu89xxx","Sleep":30,"Server":"brew-spy.com","API":3,"Port":80,"URL":"/api/ispindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":198.1,"TS":0,"OWpin":12,"SSID":"WiFi-Arnet-rxxx","PSK":"xxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2417,"aY":-3001,"aZ":1231}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel002","Token":"97utjst0v334l8he32pgu89xxx","Sleep":30,"Server":"brew-spy.com","API":3,"Port":80,"URL":"/api/ispindel","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":198.1,"TS":0,"OWpin":12,"SSID":"WiFi-Arnet-rxxx","PSK":"xxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2417,"aY":-3001,"aZ":1231}
Boot-Mode: Software/System restart

Double Reset detected

going to Config Mode
reboot RFCAL
rd
```

_Originally posted by @stefschin in https://github.com/universam1/iSpindel/issues/217#issuecomment-494221475_

universam1, 2019-05-23:
Please describe the issue clearly

ErikdBr, 2019-05-24:
Hallo Samuel,

The offset tilt after calibration seems not to be saved after calibration.
After calibration my offset tilt was 89.9xxx. After saving the config all was well. After switching the power off and on the next day the ofsset went back from 89.9xxx to 67.
I started the calibration again and laid the Ispindel on the exact same location where I measured the offset before. Outcome was about the same tilt of 89,xxx. Saved the configuration. turned off.
Yesterday I turned my Ispindel on and laid it flat on a 100% spirit level surface. I expect to see a tilt of about 90 but I got 68.
Voltage was above 4V and version of the firmware is 6.05

universam1, 2019-05-24:
Could you please go ahead and reset the configuration completely and try again. I suspect your config is broken and does not update.

This option you can find unter Maintenance

ErikdBr, 2019-05-25:
I did it and tried several times with 3 different versions of the firmware, 6.0.5 and 6.0.6 but  I had to downgrade to 5.8.6 in order to get the ispindel to hold the offset tilt after switching off. I will post a more details report later on, I am now going to sleep. Thanks.

ErikdBr, 2019-05-25:
So here is my report on my findings.

Firmware version 6.0.5 battery [SAMSUNG INR18650-35E](https://www.batterijservice.nl/nl/samsung-inr18650-35e-3500mah-unprotected.html) voltage@4,51
1. reset configuration
2. configured ispindel
3. calibrate ispindel, offset tilt 89.9xxx
4. turn off ispindel wait 10 seconds turn on, offset tilt **68.xxx**
Conclusion: ispindel did not store the offset tilt in memory correctly

Firmware version 6.0.6 voltage@4,51
1. reset configuration
2. configured ispindel
3. upgraded firmware ota from 6.0.5 to 6.0.6
3. calibrate ispindel, offset tilt 89.9xxx
4. turn off ispindel wait 10 seconds turn on, offset tilt **58.xxx**
Conclusion: ispindel did not store the offset tilt in memory correctly

Because I have seen a couple of postings from people with similar problems where they didn't have any problems after downgrading to firmware 5.8.6 I did the same.

Firmware version 5.8.6 voltage@4,51
1. reset configuration
2. configured ispindel
3. downgraded graded firmware ota from 6.0.6 to 5.8.6
3. calibrate ispindel, offset tilt 89.9xxx
4. turn off ispindel wait 10 seconds turn on, offset tilt 89.9xxx
5. repeated #4 a couple of times, every time offset tilt 89.9xxx
Conclusion: ispindel stored the offset tilt in memory correctly

So somewhere in a version of the firmware after version 5.8.6 some code got lost?
That is a pitty(sehr schade), for me mainly because major updates regarding use of power are in 6.0.0 and above, for example I see the blue light flashing more when sending data.

ErikdBr, 2019-06-03:
Any news on this one?
I did a search on "tilt" and the following closed issues came up: #217 and #233 
Although they are closed the issue still remains the same: on some(?) Ispindels the offset tilt of about 90 degrees is not saved properly, not read back properly or not displayed properly.
https://github.com/universam1/iSpindel/issues/233#issuecomment-470693817 says "Fixed with 6.0.5" but as you can see above I haven't been able to verify that.

I had to downgrade to 5.8.6, as I have seen several people do, in order to get it working.

I was worried about battery life with this 5.8.6 version because I didn't remove the led but after a 9 day 24/7 test to calibrate the Ispindel with a 2 minute update interval (it is in a fermenting sugar solution and fermenting took a while to get going and still is going) battery voltage has dropped from 4.50V to 4.25V so I am not worried about that anymore :-)

However it would be nice if this could be solved in a new version so we can all stay up to date with the firmware.

Greetings, Erik.

cstewy5, 2019-06-05:
I've got the same issue as ErikdBr, the 'calibration offset' doesn't survive a restart.
1. Power on iSpindel after it has been turned off. Showing ~16° when vertical
2. Maintenance menu  'calibrate offset' , sit iSpindel as per diagram (horizontal)
3. Running 'calibrate offset'
4. After 'calibrate offset' horizontal correct showing ~90°
5. After 'calibrate offset' vertical also correct showing ~0°
6. Power off, then power back on. Horizontal showing 86°
7. Vertical back to incorrect again. Showing ~16* when vertical again

ErikdBr, 2019-06-17:
@universam1 Hallo Samuel, have you had the time to look into this one?

universam1, 2019-06-17:
I could not reproduce that issue so far. Only thing that could help if you post here the console log before and after the power off, maybe that gives some insight

ErikdBr, 2019-06-17:
I will come back to that, it is in a brew right now :-)
[image]

mpjames01, 2019-06-30:
I have exactly the same problem with mine, when I turn it off it loses the offset calibration.

ErikdBr, 2019-07-01:
@cstewy5 @mpjames01 Maybe one of you guys can do what @universam1 suggested and  post the console log before and after the power off on here? Because my iSpindel is dead since saturday due to experementing with wireless charging and/or causing a short circuit by disconnecting the battery with pliers. During the wireless charging the thing got so hot that the black tape melted!

So I have to build another one first before I can post my console log.

[image]

mpjames01, 2019-07-05:
sorry but I don't know how to find the console log. If you can explain it maybe I could try to post it. I'm trying to erase the memory and to install the firmware again but it doesn't work. I'm thinking that could be a problem of the Wemos or of the gyroscope...

ErikdBr, 2019-07-11:
@cstewy5 @mpjames01 @universam1 

I think I discovered(@homebrewtalk forum) the solution to this problem. It seems that sometimes it looks like the flashing went fine while it actually didn't. Sometimes it will then work with an earlier version of the software like 5.8.6 but with my 2nd build it didn't work at all.

So after I blew up my 1st iSpindel I build a 2nd one. With this one I had the problem that the blue led was permanently lit up after flashing with nodemcu flasher thru usb and the iSpindel didn't work at all. After searching the internet for this problem I found this thread:

https://www.homebrewtalk.com/forum/threads/ispindle-diy-electronic-hydrometer.598187/page-19#post-8366882

and I followed the instructions of user gromitdj which is basically to initialize the internal memory of the esp8266 first before(or after when you are having problems) flashing it with the iSpindel firmware.

So I flashed the ESP with the following settings (if you want to do this, check that you'll have the adresses selected correctly).

[image]

Then, I flashed the ESP with the iSpindel bin like so:

### (**Don't mind the version, I actually used 6.0.5** I just nicked the pics of the other forum for your convenience)

[image]

And guess what, it didn't only solved the problem of the iSpindel not working at all, after calibrating and a power off and on it keeps the tilt of the calibration also!

So @cstewy5 @mpjames01 maybe you can try this and if it works for you too we then can ask @universam1  to maybe add this as an extra step in the documentation.

I am very curious if this works for you all.

universam1, 2019-07-11:
Good job @ErikdBr for presenting a solution and discovering the workaround!
I’m happy to accept this info as part of the FAQ for instance, if you mind coming up with a pull request

mpjames01, 2019-07-11:
@ErikdBr thanks, I will try! I have a question: did you leave the default settings in the advanced menu?

ErikdBr, 2019-07-11:
> Good job @ErikdBr for presenting a solution and discovering the workaround!
> I’m happy to accept this info as part of the FAQ for instance, if you mind coming up with a pull request

Give me some time to figure out how to make a pull request and when I have I will do that.

ErikdBr, 2019-07-11:
> @ErikdBr thanks, I will try! I have a question: did you leave the default settings in the advanced menu?

I only set the baud rate to 115200 as suggested in the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md)

cstewy5, 2019-07-13:
I tried this procedure and it didn't work:
1.  Connected with nodemcu flasher thru usb
2. Set settings as per Erik's suggestion:
CONFIG
1st Line "INTERNAL://NODEMCU" Address "0x00000"
2nd Line "INTERNAL://BLANK" Address "0x01000"

ADVANCED
Baudrate: 115200
Flashsize:1MByte
Flash speed: 40Mhz
SPI Mode: DOUT

3. clicked flash and let it finish
4. Flashed firmware:

CONFIG
1st Line "C:\firmware.bin" Address "0x00000"

5. Connected and then checked angles
6. Showing ~16° when vertical

Maintenance menu 'calibrate offset' , sit iSpindel as per diagram (horizontal)
Running 'calibrate offset'
After 'calibrate offset' horizontal correct showing ~90°
After 'calibrate offset' vertical also correct showing ~0°
Power off, then power back on. Horizontal showing 86°
Vertical back to incorrect again. Showing ~16* when vertical again

7. i also tried downgrading to 5.8.6, same issue. Tried upgrading again to 6.0.6. Same issue

I suspect your hardware change was what solved the original issue, not the initialize the internal memory before flashing.

cstewy5, 2019-07-13:
Here is my serial output while running 'calibrate offset':

*WM: Sent iSpindel info page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: offset
Testing device connections...
MPU6050 connection successful
averaging 1000 readings each time
expanding:
	XAccel			YAccel				ZAccel
[0,0] --> [19452,19112]	[0,0] --> [-3357,-2842]	[0,0] --> [3163,3521]
[-1000,0] --> [10651,19112]	[0,0] --> [-3357,-2842]	[0,0] --> [3165,3521]
[-2000,0] --> [1639,19112]	[0,0] --> [-3359,-2842]	[0,0] --> [3158,3521]
[-3000,0] --> [-7349,19112]	[0,0] --> [-3360,-2842]	[0,0] --> [3157,3521]
[-3000,0] --> [-7349,18865]	[0,0] --> [-3360,-3358]	[0,0] --> [3157,3158]
	XAccel			YAccel				ZAccel
[-3000,0] --> [-7349,19458]	[0,1000] --> [-3360,5812]	[0,1000] --> [3157,11589]
[-3000,0] --> [-7349,19456]	[0,1000] --> [-3360,6014]	[0,2000] --> [3157,20214]

closing in:
	XAccel			YAccel				ZAccel
[-3000,-1500] --> [-7349,6220]	[0,500] --> [-3360,1420]	[1000,2000] --> [11954,20214]
[-2250,-1500] --> [-644,6220]	[250,500] --> [-967,1420]	[1500,2000] --> [15996,20214]
[-2250,-1875] --> [-644,2504]	[250,375] --> [-967,123]	[1500,1750] --> [15996,18197]
[-2250,-2062] --> [-644,941]	[312,375] --> [-424,123]	[1500,1625] --> [15996,17186]
[-2250,-2156] --> [-644,76]	[343,375] --> [-161,123]	[1500,1562] --> [15996,16636]
	XAccel			YAccel				ZAccel
[-2203,-2156] --> [-365,76]	[359,375] --> [-11,123]	[1531,1562] --> [16351,16636]
[-2179,-2156] --> [-160,76]	[359,367] --> [-11,67]	[1531,1546] --> [16351,16478]
[-2167,-2156] --> [-48,76]	[359,363] --> [-11,33]	[1531,1538] --> [16351,16416]
[-2167,-2161] --> [-48,3]	[359,361] --> [-11,13]	[1534,1538] --> [16379,16416]
[-2164,-2161] --> [-10,3]	[359,360] --> [-11,10]	[1534,1536] --> [16379,16394]
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2164,-2162] --> [-10,10]	[359,360] --> [-8,10]	[1535,1536] --> [16375,16394]
[-2163,-2162] --> [-3,10]	[359,360] --> [-10,10]	[1535,1536] --> [16368,16394]
[-2163,-2162] --> [0,10]	[359,360] --> [-13,10]	[1535,1536] --> [16364,16394]
-------------- done --------------
a/g:	6	-14	16370
saving config...{"Name":"iSpindel001","Token":"","Sleep":900,"Server":"192.168.1.XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX","POLY":"0.000000941*tilt^3 + -0.000130507*tilt^2 + 0.007714405*tilt + 0.875149754","aX":-2163,"aY":359,"aZ":1535}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel001","Token":"","Sleep":900,"Server":"192.168.1.XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX","POLY":"0.000000941*tilt^3 + -0.000130507*tilt^2 + 0.007714405*tilt + 0.875149754","aX":-2163,"aY":359,"aZ":1535}
Boot-Mode: Software/System restart

woken from deepsleep, normal mode
Samples:39 min:89.88 max:89.96 time:677
x: 4 y: 16378 z: -16
Tilt: 89.93
Tacc: 20.93
Volt: 4.09
Temp: 18.19
Gravity: 1.20
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 1394

MTkeg, 2019-07-13:
I just completed 2 builds and am having the same Offset Calibration Issue.  If I perform the Offset Calibration then turn the device off it reverts back to the original tilt (pre-offset calibration).  I performed the method @ErikdBr presented; this did not resolve the issue.  Below are the logs when I performed the Offset Calibration and after powering down and restarting.

[After Offset Calibration.txt](https://github.com/universam1/iSpindel/files/3388484/After.Offset.Calibration.txt)
[After Power Down.txt](https://github.com/universam1/iSpindel/files/3388485/After.Power.Down.txt)

Let me know what you guys think.

MT

MTkeg, 2019-07-14:
I saw something interesting today when I was trying to figure out what is going on.  My battery voltage is 4.12v.  I totally cleared the D1 mini using the erase feature using the ESP8266 Flash Download Tool.  I used the NodeMCU Firmware Programmer and programmed Firmware v6.0.6 to 0x00000.  

It appears to me that aX, aY, and aZ successfully save but after a power cycle they are not being applied to x, y, or z.  Was there a snippet of code accidentally removed between v5 and v6 that could cause this problem?

Here is a console snippet for the fresh install:

```
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,
"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"",
"Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,
"SSID":"XXXXX","PSK":"XXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054",
"aX":0,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:61.35 max:61.53 time:756
x: -1488 y: 15054 z: -8128
Tilt: 61.45
Tacc: 33.95
Volt: 4.23
Temp: 27.00
Gravity: 19.01

```
I then performed the offset calibration.  Here is a snippet of the console output when the offset calibration was complete:

```
-------------- done --------------
a/g:    -14     -10     16376
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0
,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"",
"Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,
"SSID":"XXXXX","PSK":"XXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054",
"aX":-1391,"aY":1671,"aZ":2325}
saved successfully
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

   FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,
"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"",
"Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,
"SSID":"XXXXX","PSK":"XXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054",
**"**aX":-1391,"aY":1671,"aZ":2325****}
Boot-Mode: Software/System restart

woken from deepsleep, normal mode
Samples:40 min:89.85 max:89.97 time:700
x: 6 y: 16360 z: -6
Tilt: 89.91
Tacc: 33.29
Volt: 4.23
Temp: 26.56
Gravity: 33.52
```
So far so good.  Now I power cycle the D1 mini, it appears that aX, aY, and aZ are saved but are not being applied to x, y, or z after a power cycle (see snippet below)

```
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,
"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"",
"Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,
"SSID":"XXXXX","PSK":"XXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054",
"aX":-1391,"aY":1671,"aZ":2325}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:40 min:68.45 max:68.53 time:701
x: -13710 y: 5624 z: -3068
Tilt: 68.50
Tacc: 30.83
Volt: 4.23
Temp: 26.00
Gravity: 22.64
```

mpjames01, 2019-07-15:
I've tried several settings and several flashing but there is nothing to do...If you want I can post the log file (sorry but I did't saw the procedure in the FAQ before) but is the same as @MTkeg. I'm thinking to change the Wemos D1 mini and try again, any suggestion about the settings in nodemcu? I have to change only the boudrate to 115200?

cstewy5, 2019-07-15:
in NodeMCU flasher I used the following during 'memory initialization' and firmware flash:
Baudrate: 115200
Flash size: 1MByte
Flash speed: 40MHz
SPI Mode: DOUT

JETET, 2019-07-16:
I have the following problem does not stay connected the wemos D1 Mini NodeMcu 4 M bytes
[image]

JETET, 2019-07-16:
It keeps giving me error I don't know what to do anymore, some idea
[image]

mpjames01, 2019-07-16:
if it could help:
1:
[before offset calibration.txt](https://github.com/universam1/iSpindel/files/3399210/before.offset.calibration.txt)
2:
[offset calibration.txt](https://github.com/universam1/iSpindel/files/3399215/offset.calibration.txt)
3:
[after offset calibration.txt](https://github.com/universam1/iSpindel/files/3399216/after.offset.calibration.txt)
4:

ErikdBr, 2019-07-16:
> I have the following problem does not stay connected the wemos D1 Mini NodeMcu 4 M bytes
> [image]

First of all you have 4mb of memory then your setting in nodemcu also should be 4mb.
After that you have to go into the config of the iSpindel and fill in your token that you get from ubidots for your device. It is a unique identifier so ubidots can recognize it is your device.

JETET, 2019-07-17:
thanks for answering, I do not want to use any server I just want to see it on the website can it?

ErikdBr, 2019-07-17:
> thanks for answering, I do not want to use any server I just want to see it on the website can it?

If you want to see it on ubidots website you'll have to fill in the ubidots token (which is under api credentials in ubidots) in the config page of the iSpindel.

ErikdBr, 2019-07-17:
[image]
[image]

JETET, 2019-07-17:
No, I want to see it only in my network by putting the ip that my network provides in: 192.168.1.105

ErikdBr, 2019-07-17:
Over HTTP I guess

JETET, 2019-07-17:
I don't want to use any server but when the wemos connect to my network I can connect to the wemos using any browser without any server
[image]

ErikdBr, 2019-07-17:
Sorry don't know how to do that.

JETET, 2019-07-17:
Swim in place on the Bat 43 and keep going into sleepmode and there's no way to wake up coming in from the IP, just resetting

*WM: iSpindel
*WM: Sent iSpindel info page
*WM: Handle root
*WM: Info
17:21:50.356 -> *WM: Sent info page
Samples:30 min:24.36 max:25.70 time:520
17:23:18.333 -> x: 4054 y: 5464 z: -14146
17:23:18.379 -> Tilt: 25.01
17:23:18.379 -> Tacc: 40.32
17:23:18.379 -> Volt: 4.48
17:23:18.379 -> Temp: 29.62
17:23:18.379 -> Parse error at 1
17:23:18.379 -> Gravity: 0.00
17:23:18.379 -> IP: 192.168.1.100
17:23:18.379 -> 
17:23:18.379 -> calling Ubidots
17:23:18.379 -> {"tilt":25.01286,"temperature":29.625,"battery":4.483837,"gravity":0,"interval":3600,"RSSI":-49}
17:23:19.227 -> Sender: Ubidots posting
17:23:19.227 -> POST /api/v1.6/devices/iSpindel000?token=BBFF&#45;Gz8tphJp02LXbR7WKE25senkUtDOkw HTTP/1.1
17:23:19.227 -> Host: things.ubidots.com
17:23:19.227 -> User-Agent: ESP8266
17:23:19.227 -> Connection: close
17:23:19.227 -> Content-Type: application/json
17:23:19.227 -> Content-Length: 96
17:23:19.227 -> 
HTTP/1.1 403 Forbidden
17:23:20.070 -> Server: nginx
17:23:20.070 -> Date: Wed, 17 Jul 2019 15:23:16 GMT
17:23:20.070 -> Content-Type: application/json
17:23:20.070 -> Transfer-Encoding: chunked
17:23:20.070 -> Connection: close
17:23:20.070 -> Vary: Accept-Encoding
17:23:20.070 -> Vary: Cookie
17:23:20.070 -> Allow: GET, POST, HEAD, OPTIONS
17:23:20.070 -> 
17:23:20.070 -> 3b
17:23:20.070 -> {"detail": "Authentication credentials were not provided."}
17:23:20.070 -> 0
17:23:20.070 ->

lbussy, 2019-07-17:
> I don't want to use any server but when the wemos connect to my network I can connect to the wemos using any browser without any server

I understand what you mean however this is not possible.  The device is sleeping in between checking the gravity.  At that time it is only "awake" briefly, transmits, and goes back to sleep.  You must have what it called an endpoint (such as Ubidots) to receive the information, save it, and allow you to view it.

Options I know about currently are:

- Ubidots
- Brewer's Friend
- Fermentrack
- BrewPi Remix (in Alpha testing)

Alternatively, if you have a computer available on the network and only want to save the data you can create your own endpoint.

cstewy5, 2019-07-17:
JETET you are dragging this issue off topic

JETET, 2019-07-18:
Perdon

brendongough, 2019-10-07:
@universam1 any luck with this issue? I'm not sure if there is a way to get this to save the same way the other settings are saved or if there is a longer delay which can be implemented to make sure these are written to flash 100%. At the moment, this is one of the bigger issues people are facing as the majority of good quality gyros are still out by a around 5-7 degrees.

universam1, 2019-10-07:
my problem is that I cannot reproduce this problem at all. So I really have no hook yet to understand whats going wrong on your devices

cstewy5, 2019-10-07:
Is there any more data/logs I can collect to help solve the issue? console logs posted above.

universam1, 2019-10-09:
I've written a test Firmware that adds a confirmation to the calibration offsets.
1. it sets the offset, outputting `applying offsets: ...`
2. reading back from the sensor the values, output `confirming offsets: ...`

This way I hope we can shed some light into whether the sensor or the Firmware is the issue.

Please update here your  results @ErikdBr @cstewy5 @brendongough or whoever is willing to test

[firmware.bin.zip](https://github.com/universam1/iSpindel/files/3708798/firmware.bin.zip)

cstewy5, 2019-10-10:
Thanks, i'll try it on Sunday and upload my results.

cstewy5, 2019-10-13:
I re-intialized the memory (just to be safe) then flashed the debug firmware here is the output:
`1st column		2nd column
1st Line:	INTERNAL://NODEMCU	0x00000
2nd Line:	INTERNAL://BLANK	0x01000`
This is after flashing debug firmware then running the calibration (ignore the wifi connection issue, I hadn't set the password yet):

`*WM: Maintenance page
*WM: Sent iSpindel info page
*WM: offset
Testing device connections...
MPU6050 connection successful
averaging 1000 readings each time
expanding:
	XAccel			YAccel				ZAccel
[0,0] --> [18979,18620]	[0,0] --> [-3580,-3514]	[0,0] --> [2773,3164]
[-1000,0] --> [10167,18620]	[0,0] --> [-3579,-3514]	[0,0] --> [2774,3164]
[-2000,0] --> [1177,18620]	[0,0] --> [-3582,-3514]	[0,0] --> [2776,3164]
[-3000,0] --> [-7822,18620]	[0,0] --> [-3585,-3514]	[0,0] --> [2769,3164]
[-3000,0] --> [-7822,18407]	[0,0] --> [-3585,-3584]	[0,0] --> [2769,2772]
	XAccel			YAccel				ZAccel
[-3000,0] --> [-7822,18981]	[0,1000] --> [-3585,5587]	[0,1000] --> [2769,11210]
[-3000,0] --> [-7822,18981]	[0,1000] --> [-3585,5791]	[0,2000] --> [2769,19836]

closing in:
	XAccel			YAccel				ZAccel
[-3000,-1500] --> [-7822,5769]	[0,500] --> [-3585,1200]	[1000,2000] --> [11588,19836]
[-2250,-1500] --> [-1125,5769]	[250,500] --> [-1190,1200]	[1500,2000] --> [15611,19836]
[-2250,-1875] --> [-1125,2023]	[375,500] --> [-103,1200]	[1500,1750] --> [15611,17820]
[-2250,-2062] --> [-1125,457]	[375,437] --> [-103,487]	[1500,1625] --> [15611,16804]
[-2156,-2062] --> [-402,457]	[375,406] --> [-103,224]	[1562,1625] --> [16254,16804]
	XAccel			YAccel				ZAccel
[-2109,-2062] --> [-19,457]	[375,390] --> [-103,73]	[1562,1593] --> [16254,16492]
[-2109,-2085] --> [-19,199]	[382,390] --> [-4,73]	[1577,1593] --> [16361,16492]
[-2109,-2097] --> [-19,102]	[382,386] --> [-4,29]	[1577,1585] --> [16361,16432]
[-2109,-2103] --> [-19,48]	[382,384] --> [-4,13]	[1577,1581] --> [16361,16402]
[-2109,-2106] --> [-19,25]	[383,384] --> [-3,13]	[1579,1581] --> [16381,16402]
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2109,-2107] --> [-19,9]	[383,384] --> [-6,13]	[1579,1580] --> [16381,16396]
[-2109,-2108] --> [-19,8]	[383,384] --> [-8,13]	[1579,1580] --> [16377,16396]
[-2109,-2108] --> [-9,8]	[383,384] --> [-7,13]	[1579,1580] --> [16378,16396]
-------------- done --------------
a/g:	-8	6	16374
saving config...{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v78a1a66e
~ld

FW 6.1.2.debug
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}
Boot-Mode: Software/System restart

woken from deepsleep, normal mode
Samples:39 min:89.89 max:89.97 time:678
x: -8 y: 16388 z: -10
Tilt: 89.93
Tacc: 24.96
Volt: 4.15
Temp: 22.25
Gravity: 33.53
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 1339
`
Vertical and horizontal reading good now:
`FW 6.1.2.debug
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"XXXX","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:0.31 max:0.85 time:757
x: -108 y: -120 z: -16448
Tilt: 0.53
Tacc: 23.46
Volt: 4.16
Temp: 21.37
Gravity: -13.76
After waiting 5968ms, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 7304
`
Now after a power cycle, vetical reading bad again (15 degrees):

`FW 6.1.2.debug
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"XXXX","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:14.73 max:15.37 time:756
x: -230 y: 4486 z: -16376
Tilt: 15.23
Tacc: 23.84
Volt: 4.14
Temp: 21.94
Gravity: -5.64
IP: 192.168.1.102

calling MQTT
Attempting MQTT connection
Connected to MQTT

MQTT publish: XXXX

Closing MQTT connection

Final-sleep: 900s; RT: 1338
`

universam1, 2019-10-13:
Now this is interesting, the offsets were not saved to your flash storage for some reason. You can see that later „offsets not available“ is displayed that is nothing can be restored. Maybe your flash is corrupted?

cstewy5, 2019-10-13:
The new firmware uploads fine though, isn't that also stored in the flash memory? There was a theory floating around that there was no enough time given to writing the offsets to the same memory.

universam1, 2019-10-13:
Thank you for testing, this is actually very helpful!
I am hunting a library bug possibly. I've extended the testing firmware to give more output and flush the file system before continuing. Also rearranged and added some error tests in order to see if there is a flash issue involved. If there is not enough time to flush out the storage that should be covered now. Could you please test once again?
[firmware.zip](https://github.com/universam1/iSpindel/files/3721580/firmware.zip)

cstewy5, 2019-10-13:
re-initalized as before then flashed the new debug firmware. I then ran the offset calibration and it got stuck in a boot loop. Here are the logs, it kept repeating until I disconnected the wemos.

```
'''
*WM: Maintenance page
*WM: Sent iSpindel info page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: offset
Testing device connections...
MPU6050 connection successful
averaging 1000 readings each time
expanding:
	XAccel			YAccel				ZAccel
[0,0] --> [19124,19082]	[0,0] --> [-3538,-3464]	[0,0] --> [2892,3444]
[-1000,0] --> [10310,19082]	[0,0] --> [-3538,-3464]	[0,0] --> [2893,3444]
[-2000,0] --> [1319,19082]	[0,0] --> [-3538,-3464]	[0,0] --> [2892,3444]
[-3000,0] --> [-7677,19082]	[0,0] --> [-3539,-3464]	[0,0] --> [2893,3444]
[-3000,0] --> [-7677,18526]	[0,0] --> [-3539,-3538]	[0,0] --> [2893,2895]
	XAccel			YAccel				ZAccel
[-3000,0] --> [-7677,19129]	[0,1000] --> [-3539,5632]	[0,1000] --> [2893,11326]
[-3000,0] --> [-7677,19126]	[0,1000] --> [-3539,5839]	[0,2000] --> [2893,19944]

closing in:
	XAccel			YAccel				ZAccel
[-3000,-1500] --> [-7677,5913]	[0,500] --> [-3539,1249]	[1000,2000] --> [11710,19944]
[-2250,-1500] --> [-973,5913]	[250,500] --> [-1142,1249]	[1500,2000] --> [15735,19944]
[-2250,-1875] --> [-973,2174]	[375,500] --> [-59,1249]	[1500,1750] --> [15735,17938]
[-2250,-2062] --> [-973,605]	[375,437] --> [-59,536]	[1500,1625] --> [15735,16928]
[-2156,-2062] --> [-257,605]	[375,406] --> [-59,272]	[1562,1625] --> [16376,16928]
	XAccel			YAccel				ZAccel
[-2156,-2109] --> [-257,129]	[375,390] --> [-59,120]	[1562,1593] --> [16376,16616]
[-2132,-2109] --> [-54,129]	[375,382] --> [-59,45]	[1562,1577] --> [16376,16488]
[-2132,-2120] --> [-54,46]	[375,378] --> [-59,5]	[1562,1569] --> [16376,16416]
[-2126,-2120] --> [-4,46]	[376,378] --> [-13,5]	[1565,1569] --> [16383,16416]
[-2126,-2123] --> [-4,11]	[377,378] --> [-16,5]	[1565,1567] --> [16383,16394]
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2126,-2124] --> [-4,13]	[377,378] --> [-18,5]	[1565,1566] --> [16383,16399]
[-2125,-2124] --> [-3,13]	[377,378] --> [-21,5]	[1565,1566] --> [16380,16399]
[-2125,-2124] --> [-2,13]	[377,378] --> [-22,5]	[1565,1566] --> [16381,16399]
-------------- done --------------
a/g:	-22	-28	16378
new offsets: -2125:377:1565
saving config...
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"XXXX","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v818a55f5
~ld

FW 6.1.2.debug2
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file

Exception (28):
epc1=0x40225c06 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff1070 end: 3fff1300 offset: 01a0
3fff1210:  00000000 00000000 00000001 4020512f  
[... 110 lines cut ...]
epc1=0x40225c06 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff1070 end: 3fff1300 offset: 01a0
3fff1210:  00000000 00000000 00000001 4020512f  
3fff1220:  40220900 00000000 000003e8 4020f8b1  
3fff1230:  3fff25d4 3fff2614 00000000 00000000  
3fff1240:  3ffefe08 00000000 3fff273c 3fff273c  
3fff1250:  3fff2a1c 3fff2b3c 3fff2b2c 3fff2a1c  
3fff1260:  3ffe86a0 00000000 3ffefb0c 00000000  
3fff1270:  cafebabe 00000000 3ffefb2c 3ffefc4c  
3fff1280:  3ffefb0c 00000000 3ffefb2c 402068d4  
3fff1290:  feefeffe feefeffe feefeffe feefeffe  
3fff12a0:  feefeffe feefeffe feefeffe feefeffe  
3fff12b0:  feefeffe feefeffe feefeffe feefeffe  
3fff12c0:  feefeffe feefeffe feefeffe feefeffe  
3fff12d0:  feefeffe feefeffe feefeffe 3ffefc4c  
3fff12e0:  3fffdad0 00000000 3ffefc1c 40211e94  
3fff12f0:  feefeffe feefeffe 3ffe8680 4010107d  
<<<stack<<<

```
```

universam1, 2019-10-13:
Looks like your SPIFFS is corrupted. You can call a format of Spiffs via Reset in Maintenance menu

cstewy5, 2019-10-14:
I cleared the flash using NodeMCU pyFlasher. I then reflashed 6.0.6. I then opened maintenance and clicked 'factory reset'. Then I ran offset configuration. I then power cycled and the offsets were lost again. Here are the logs after the offset calibration. Not sure what the "need to format SPIFFS: 1" line means.
'''
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2138,-2136] --> [-3,15]	[375,376] --> [-4,13]	[1551,1552] --> [16376,16391]
[-2137,-2136] --> [-1,15]	[375,376] --> [-3,13]	[1551,1552] --> [16376,16391]
[-2137,-2136] --> [0,15]	[375,376] --> [-5,13]	[1551,1552] --> [16378,16391]
-------------- done --------------
a/g:	-8	-2	16368
saving config...
need to format SPIFFS: 1
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"","PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-2137,"aY":375,"aZ":1551}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

'''

universam1, 2019-10-14:
Here is a key difference, you can see that now the offsets were saved fine!
Search for aX":-2137,"aY":375,"aZ":1551

However the logs of the restore are not included which are the next step we need to figure

cstewy5, 2019-10-15:
calibration offset once complete ~0 degrees when vertical:
```
XAccel			YAccel				ZAccel
[-2071,-2070] --> [0,21]	[377,378] --> [-21,1]	[1569,1570] --> [16374,16385]
[-2071,-2071] --> [0,7]	[377,378] --> [-24,1]	[1569,1570] --> [16365,16385]
-------------- done --------------
a/g:	0	-34	16366
saving config...{"Name":"xxxx","Token":"","Sleep":900,"Server":"xxxx","API":10,"Port":xxxx,"URL":"","DB":"xxxx","Username":"xxxx","Password":"xxxx","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"xxxx","PSK":"xxxx","POLY":"0.000000941*tilt^3 + -0.000130507*tilt^2 + 0.007714405*tilt + 0.875149754","aX":-2071,"aY":377,"aZ":1569}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"XXXX","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX"}
Boot-Mode: Software/System restart

woken from deepsleep, normal mode
Samples:39 min:89.82 max:89.96 time:676
x: 2 y: 16374 z: -32
Tilt: 89.90
Tacc: 24.24
Volt: 4.11
Temp: 21.56
Gravity: 1.20
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 1360

```

Nower cycle, (then a reboot to start capturing logs again). Back to ~15 degrees when vertical, offsets not recalled.
```
FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"XXXX","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX","POLY":"0.000000941*tilt^3 + -0.000130507*tilt^2 + 0.007714405*tilt + 0.875149754","aX":-2071,"aY":377,"aZ":1569}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:85.78 max:87.53 time:756
x: -1586 y: 21022 z: 136
Tilt: 86.39
Tacc: 23.64
Volt: 4.11
Temp: 21.62
Gravity: 1.17
IP: 192.168.1.102

calling MQTT
Attempting MQTT connection
Connected to MQTT
MQTT publish: XXXX/86.38782
MQTT publish: XXXXtemperature/21.625
MQTT publish: XXXXtemp_units/C
MQTT publish: XXXXbattery/4.108446
MQTT publish: XXXXgravity/1.174288
MQTT publish: XXXXinterval/900
MQTT publish: XXXXRSSI/-44
Closing MQTT connection

Final-sleep: 900s; RT: 1283

```

universam1, 2019-10-15:
You are running FW 6.0.6!?

cstewy5, 2019-10-15:
So far I've tried 6.0.6, 6.1.2 debug, 6.1.2 debug2, 6.1.0, 5.8.4. All have the same issue

cstewy5, 2019-10-15:
I reflashed 6.1.2 debug2 again. Here are the results:

Running calibrate offset:
```
averaging 10000 readings each time
	XAccel			YAccel				ZAccel
[-2191,-2189] --> [-13,3]	[377,378] --> [-13,6]	[1571,1572] --> [16381,16400]
[-2191,-2190] --> [-13,4]	[377,378] --> [-16,6]	[1571,1572] --> [16381,16400]
[-2191,-2190] --> [-13,4]	[377,378] --> [-16,6]	[1571,1572] --> [16382,16400]
-------------- done --------------
a/g:	-2	-28	16388
new offsets: -2191:377:1571
saving config...
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX"}saved successfully
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v818a55f5
~ld

FW 6.1.2.debug2
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX"}
Boot-Mode: Software/System restart

woken from deepsleep, normal mode
Samples:39 min:89.84 max:89.95 time:678
x: 2 y: 16376 z: -14
Tilt: 89.89
Tacc: 24.61
Volt: 4.10
Temp: 22.00
Gravity: 33.51
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 1341

```

after reset button, offsets still working ~0 degrees vertical:
```
FW 6.1.2.debug2
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"XXXX","Token":"","Sleep":900,"Server":"XXXX","API":10,"Port":XXXX,"URL":"","DB":"ispindel","Username":"XXXX","Password":"XXXX","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"XXXX","PSK":"XXXX"}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:88.57 max:89.96 time:756
x: 86 y: 16396 z: -14
Tilt: 89.50
Tacc: 23.89
Volt: 4.11
Temp: 22.06
Gravity: 33.31
After waiting 2972ms, result 3
IP: 192.168.1.102

calling MQTT
Attempting MQTT connection
Connected to MQTT

MQTT publish: XXXX/temp_units/C

```

Power cycle, reverts back to ~15 degrees vertical
```
FW 6.1.2.debug2
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"xxxx","Token":"","Sleep":900,"Server":"xxxx","API":10,"Port":xxxx,"URL":"","DB":"ispindel","Username":"xxxx","Password":"xxxx","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"xxxx","PSK":"xxxx"}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:88.27 max:88.94 time:757
x: -488 y: 21070 z: 118
Tilt: 88.64
Tacc: 24.41
Volt: 4.11
Temp: 22.56
Gravity: 32.88
After waiting 2969ms, result 3
IP: 192.168.1.102

calling MQTT
Attempting MQTT connection
Connected to MQTT

MQTT publish: ispindel/iSpindel001/temp_units/C

Closing MQTT connection

Final-sleep: 900s; RT: 4455

```

cstewy5, 2019-10-15:
I might buy another wemos and try replacing that and a new gyro to see that resolves it.

universam1, 2019-10-15:
Thank you for the test! Again we can see that the offsets are not stored! Maybe this is indeed a SDK bug we are hitting here

universam1, 2019-11-10:
Some progress on this one, I was able to spot an issue with the library that is the driver of the MPU6050. Need to find some more time to work out a solution @cstewy5

Nitnelav, 2019-11-14:
> Some progress on this one, I was able to spot an issue with the library that is the driver of the MPU6050. Need to find some more time to work out a solution @cstewy5

Can you give any information on this issue you found ? We may have the same problem and we have a few days to work on a solution...

[6 more replies]

closed 2019-11-18 by universam1