universam1/iSpindel #277: Using multimeter to test ispindel-battery life

Gkire, 2019-05-24:
I bought a new 3000mah 18650 battery  with soldering strips from a respectable vendor in the Netherlands, and loaded it fully until the charger indicated blue.
Source: https://www.123accu.nl/UltraFire-18650-batterij-met-soldeerlippen-2-stuks-3000-mAh-i11094.html 

I configered my ispindel to 6.0.6 and placed it in the water since 22 may with intervals of 900seconds

This morning my Ispindel stopped sending reports regularly:
[image]

I have bought a multi meter to establish what might be the problem.
However, apart from soldering and following instructions I have no electronic knowhow.

Could someone point me in the right direction as to what things to measure with the multimeter to find the root cause?

universam1, 2019-05-24:
did you follow the build instructions to remove the LED?

stefschin, 2019-05-24:
Did you set battery calibration factor? Ispindel turn off when it detects under 3V, but you should have to calibrate the sensor with this factor. Look at [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md)

ErikdBr, 2019-05-26:
Any news on this?
Cause I didn't think that this was very important because I use a 3500mah battery. But after seeing this post I measured the voltage of the battery with a multimeter and I found that there was a difference of 0.4V between the displayed voltage(4.11) and the measured voltage(4.51). So I used the formula to calculate a new battery conversion factor and now the displayed value = measured value.

The formula to do this is you'll find in the FAQ that stefschin pointed out.

Other then that when I look at your figures I see a voltage drop of 0.6V in 3 hours and 20 minutes. I started a test 24 hours ago and I have a voltage drop of 0.03V with no leds removed what so ever. I don't know what that might be but I would try the conversion factor first.

Gkire, 2019-05-28:
Update: 
To previous questions: Yes all leds are removed with the exception of the wemos itself. 
Additionally I did 2 things in the last 4 days:

1. I corrected the battery conversion factor on my "3- Zwart Geel" ispindel from 191.8 to 192.2. I recharged it and had an other run.  Its progress has been as follows:
27 may 2019 13:13: 4.14V
27may 2019 23:59: 3.96V
27 may 2019 13:09 3.78V
28 may 2019 20:48: 3.51V

2. I built a second Ispindel with the second battery from the pack I bought I mentioned in my first post.
This one is called "6-Geel-Wit". Its progress has been as follows:
27 may 2019 14:26: 3.96V
27 may 2019 23:58: 3.96V
28 may 2019 14:30: 3.93V
28 may 2019 20:42 3.91V

My second Ispindel seems to use less energy. Both are on 6.0.6 so I am sure it is hardware related.
Any ideas what to measure in order to establish which component might be the culprit?

ErikdBr, 2019-06-02:
Other then that I agree that this is a hardware issue I only can say check all soldering points.
I am happy to say my battery (samsung 3500mah) is performing well, it is in a bucket with a fermenting sugar solution since last sunday with a 2 minutes update interval and here are the results of the battery:
[image]
So I am pretty sure it can survive a week of fermenting beer at a 15 minutes interval.

JETET, 2019-10-05:
how do I measure the battery on or off?  
with what charge do I do the battery correction factor?

closed 2019-10-09 by universam1