universam1/iSpindel #280: D1 Mini powers up in REST mode, no way to configure AP/iSpindel

PJRanger, 2019-06-08:
I love this project and have 2 iSpindels built however I cannot get the ESP8266 into AP mode to access the configuration page.  It powers on in the REST mode, the AP is visible on wifi but cannot connect to the config page at 192.168.4.1   Holding the RESET button for 5 seconds does nothing other than blink the light.  Please help.

ErikdBr, 2019-06-08:
You have to connect to the ap first and then go to the ip adress.

PJRanger, 2019-06-08:
Erik, this is the issue, when I load the firmware to the ESP8266 the AP will allow a connection however will not accept an IP connection at the address 192.168.4.1 (config HTTP page does not display). In the documentation it states with the iSpindel Firmware loaded the ESP8266 will boot in the rest mode and has to be set to AP mode by the reset button, however the reset button has no effect and does not go into AP mode (LED flashing at 1 second). I can access the AP but cannot enter configuration mode to proceed with any calibration, ubidots configuration or connection to my wifi network.

universam1, 2019-06-08:
Post the condole log output here. Anything else is wild guessing

PJRanger, 2019-06-08:
These are screen shots of my NodeMCU loading of the firmware and a text file of the log.

PJRanger, 2019-06-08:
Here are all the screenshots and the log from NodeMCU, is there another log file I should be looking for and if so how do I get to it?

[image]
[image]
[image]
[image]
[image]
[Log from NodeMCU.txt](https://github.com/universam1/iSpindel/files/3268864/Log.from.NodeMCU.txt)
[image]

ErikdBr, 2019-06-08:
After the initial flash you have to flash it with the Ispindel firmware. Then you will see "ispindel" in your wifi list and not "ESP_A3B626".

ErikdBr, 2019-06-08:
Firmware you will find here: https://github.com/universam1/iSpindel/releases

PJRanger, 2019-06-09:
I actually figured it out.  The flashing procedure using NodeMCU should mention to remove or deselect the NODEMCU flash information and just select the iSpindel firmware.  Once I did that my iSpindel worked as advertised.  I now have two communicating iSpindels using FERMENTRACK and will also work to connect them to CraftBeerPi.  Thank you for the assistance.  If needed, I do know some German and could help in writing documentation in English.

closed 2019-06-09 by PJRanger

universam1, 2019-06-09:
feel free improve the docs as you find, always welcome