universam1/iSpindel #281: Device will not reboot

Quernon, 2019-06-11:
Firstly, thank you for all your ongoing efforts on this great project. 

I've been trying to get this to work for a while now but keep running into the same issue. I can configure the device to connect to my WiFi network and send data, and the data is correctly recorded. However, data is only sent once and only after pressing the reset button. After that, the device never wakes up again and never sends data.

I've looked at the serial log using putty. When the reset button is pressed it shows everything correctly. Then, after the time out, it just displays one line of unrecognisable characters and stops there.

I started with a resistor between RST and D0 but have changed that to a BAT43 diode (Cathode at D0) but the issue persists. I have now tried 3 different D1 Mini boards but all have had the same issue.

Looking at the regulator on the boards, it looks like they have the weak 150mA version. Could this be the cause? If so, do you know where it's possible to source boards which carry the correct regulator?

universam1, 2019-06-11:
For testing short cut the RST and D0 and see if it changes. How about the voltage

Quernon, 2019-06-11:
Do you mean that I should connect the RST and D0 pins directly, without the use of a diode or resistor? 

And where should I be measuring the voltage between? RST and D0?

Thanks for your help.

Quernon, 2019-06-11:
Okay, so I've measured the voltage across D0 and RST. On first plugging in / turning on the D1 Mini it stays at a flat 0V. If I press the reset button it briefly flicks up to 1-2V but then stays at 0V thereafter.

If I hot wire the RST and D0 pins then the behaviour stays the same.

Quernon, 2019-06-13:
I have ordered an official board from Wemos/Lolin's aliexpress page and will update this when it arrives. If there is anything else I should try in the meantime please let me know.

universam1, 2019-06-13:
Yes short cut the RST and D0 to see if it’s about the resistor. Check that supply voltage is above 4V. If it is then still not working you got a hardware issue

Quernon, 2019-06-13:
I tried shorting RST and D0 yesterday but it made no difference. I'll try a more robust short this evening (I was trying to avoid soldering anything when I last tried).

Pretty sure supply voltage was above 4V but I'll double check and get back to you. This would be measured across 5V and GND, correct?

Quernon, 2019-06-17:
Shorting the RST and D0 pins made no difference. Supply voltage was >4V.

Looks like I have a hardware issue, probably due to the knockoff boards that I bought. I've ordered replacements and will update here when they arrive.

trafim-uakraine, 2019-06-17:
I bought 5 esp8266 https://ru.aliexpress.com/item/Smart-Electronics-D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based/32659152063.html ? spm = a2g0s.9042311.0.0.31ad33edDinub0
Two weeks suffered and came to the conclusion that these boards are not compatible with deep sleep.
DOIT.AM is written on the WiFi module.
[image]
[image]

ErikdBr, 2019-06-17:
I use these: € 3,17 | LOLIN D1 mini V3.1.0-WEMOS WIFI Internet van Dingen development board gebaseerd ESP8266 4 mb MicroPython Nodemcu Arduino Compatibel
https://s.click.aliexpress.com/e/pbB4qQQZd

Quernon, 2019-06-17:
@ErikdBr , those are the official ones from Wemos/Lolin so should be a safe bet. I've ordered the same.

pamarovis, 2019-06-19:
Hello, have you got any news? I have the same mistake with sending data only once.
I bought this board : https://www.aliexpress.com/item/32845084675.html?spm=a2g0s.9042311.0.0.12e24c4dH0a1nj

Quernon, 2019-06-19:
> 
> 
> Hello, have you got any news? I have the same mistake with sending data only once.
> I bought this board : https://www.aliexpress.com/item/32845084675.html?spm=a2g0s.9042311.0.0.12e24c4dH0a1nj

I'm still waiting for my replacement boards to arrive but will post here when they do.

ErikdBr, 2019-06-19:
When you follow the D1 mini link on here http://www.ispindel.de/docs/Parts_en.html you go to https://wiki.wemos.cc/products:d1:d1_mini
On there click on th buy now button and you'll go to: https://nl.aliexpress.com/store/product/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/1331105_32529101036.html

pamarovis, 2019-06-26:
Can you see anything weird in my putty log? 

[image]

pamarovis, 2019-06-30:
Ok, so i solved my problem ! The problem was with BAT43 diode soldered wrong way. Check yours ! :)

Quernon, 2019-07-11:
Just confirming that the issue was with the clone board. Using the official one from Lolin has fixed the issue for me.

closed 2019-07-11 by Quernon