universam1/iSpindel #282: Ability to change MAC address?

BulletMag, 2019-06-13:
I want to use several iSpindel in multiple brews, however I purchased five D1 mini clones from a Chinese seller and while they work it appears they all have the same MAC address so obviously I can't run them all at the same time.

Can a feature be added to change/spoof the MAC address so I can use these devices as I'm sure others will find this issue considering the amount of D1 mini clones out there.

ErikdBr, 2019-07-11:
I could be wrong but I think when you flash the d1 with nodemcu flasher after you build your ispindel the software will assign a mac adress
[image]

BulletMag, 2019-07-11:
Yes this is true, but i have five D1 minis that I have erased multiple times but when I go to flash them they all get the same AP MAC and STA MAC.
The five I mentioned above all came from the same reseller so likely from the same batch.
I have ten others from various other sellers all with completely different AP MAC and STA MAC so there  is clearly a possibility that the clones are literally that including the programmed MAC address.
My OQ refers to the fact that in the SDK there is a command 'wifi_set_macaddr' that allows the MAC address to be changed, unfortunately I do not know enough to know if this is permanent or per session hence why i asked if it could be added as a feature therefore if it was per session this could be done from the firmware.

closed 2020-06-23 by stale[bot]