universam1/iSpindel #283: ispindel not working after flashing

dihutton, 2019-06-27:
recently flashed my ispindel to version 6.0.6 and its now dead as the proverbial dodo
any ideas ?

dihutton, 2019-06-28:
is there anyway to get it back up and running please

ErikdBr, 2019-06-28:
Flash it with an earlier version? Like 5.8.6

dihutton, 2019-06-28:
nah that didn't work , flashes three times then nothing

ErikdBr, 2019-06-28:
Sometimes I have similar problems when I put it on after a while. Put in on and off a couple of times, the 2nd time it could work again. Other thing that might help, put it on and if it does nothing after a couple of flashes press and hold the reset button until something happens(that helped me once)

dihutton, 2019-06-28:
there is no reset button just the on off switch

dihutton, 2019-06-28:
tried that now still nothing

ErikdBr, 2019-06-28:
Then I don't know sorry.

networkingmv, 2019-06-28:
Not sure if it is related but I have a newbuild project and uploaded 6.0.6 to my Wemos. I connected a serial and got the following output and errors. Please not this is only the Wemos with nothing connected as I am at step 1 to flash the image then connect to iSpindel SSID. 

Hope this can add some value 

""
FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Acc Test Connection ERROR!
Boot-Mode: Exception
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
""

ErikdBr, 2019-06-28:
It won't work until you got the Ispindel completely build.

networkingmv, 2019-06-28:
@ErikdBr - Agh Ok, this is promising and makes total sense. If it does end up working after the build the docs are not correct as it states one should first flash then connect to WiFi iSpindel then start the build if one reads it from top to bottom. I am going to give it a bash as I am quite stuck right now as I even tried older versions of code with no luck;)

Thanks for the prompt feedback

Marius

ErikdBr, 2019-06-28:
[DIY_Spindel_Anleitung_En.pdf](https://github.com/universam1/iSpindel/files/3339554/DIY_Spindel_Anleitung_En.pdf)
I used this one. Be aware, it has the V2.1 of the iSpindel platina.

networkingmv, 2019-06-28:
@ErikdBr - You are a rockstar Sir :) I managed to build it up and I now have connectivity and all is hunky dory. I would however suggest a doc edit as it really reads to first flash then connect then assemble and it clearly does not work in this order.

Now to get this monster calibrated 

Thanks again 

Marius

dihutton, 2019-06-28:
all sorted now
thanx guys your input helped

ErikdBr, 2019-06-28:
> all sorted now
> thanx guys your input helped

So what was the problem?

dihutton, 2019-06-28:
was using esp8266 flasher and had not placed the update in the right line, should have been line 1, but I used line 2, works a treat now 
testing on ubidots before full calibration

JETET, 2019-07-14:
Good I installed the version iSpindel-6.1.0 and it starts working after a few minutes it shuts down and no longer transmits data the only thing that remains is the green light of the Position Sensor thanks.
A do not remove any of the leds

Any solution for this?

JETET, 2019-07-15:
this is the error that gives me

Samples:30 min:63.51 max:63.52 time:523
18:30:16.493 -> x: 11652 y: 11732 z: 902
18:30:16.493 -> Tilt: 63.52
18:30:16.493 -> Tacc: 42.20
18:30:16.493 -> Volt: 4.30
Temp: 32.13
18:30:16.527 -> Parse error at 1
18:30:16.527 -> Gravity: 0.00
18:30:16.527 -> IP: 192.168.1.100
18:30:16.527 -> 
18:30:16.527 -> calling HTTP
18:30:16.527 -> {"name":"iSpindel000","ID":8034015,"token":"vd6n1r6eop1sh6f9dmceibpufc","angle":63.51804,"temperature":32.125,"temp_units":"C","battery":4.296142,"gravity":0,"interval":3600,"RSSI":-48}HTTPAPI: posting
code: 200
18:30:16.905 -> {
18:30:16.905 ->   "status" : 200,
18:30:16.905 ->   "ts" : "2019-07-15 12:30:09",
18:30:16.905 ->   "data" : { }
18:30:16.940 -> }

18:30:17.007 -> Final-sleep: 3600s; RT: 367876

CyberfoxDK, 2019-08-11:
Got the same issue as JETET - running Firmware: 6.0.6

JETET, 2019-08-11:
once it has been flattened by default, it is 15 minutes, so when you flachees wait 15 minutes to see if it flashes, if you want faster you have to enter the configuration of the ispindel and put it for example in 60 seconds to see what works and not have to wait 15 minutes

closed 2019-11-10 by universam1