universam1/iSpindel #284: won't save wifi credentials

estrauss, 2019-07-09:
I've built up a board with the v4.0 PCB
I uploaded the 6.0.6 firmware via the NodeMCU firmware programmer
The only variation from most builds is:
Wemos D1 Mini Development Board ESP8285 V1.0.0
Also, I started with the 470 Ohm resistor and changed to a 330 Ohm.  BAT43 diodes are on order.
Once I get the BAT43 diodes, I plan to build another using the standard D1 Mini ESP8266

connects to ubidots and all sensors report reasonably
after power cycling, wifi credentials are lost (This is the problem) and it goes back into setup mode
battery has been fully charged
I only need to take a Wemos board (as received) and flash the firmware to 0x00000 correct?
Or is there some basic firmware I need to flash before?
Is there some sort of reset sequence I need to follow?

WiFi setup
no special characters for the network or password
set up a dedicated 2.4 GHz network
set an IP reservation

I have no idea how to get a console output to troubleshoot

Alternate fix?
Is there a preferred way to build my own .bin with the Wifi credentials hardcoded?

ErikdBr, 2019-07-12:
Try initializing it first by flashing it like this before flashing it with the firmware
[image]
Also fill in the appropriate settings in the Advanced tab for your particular board, I think you have 1MB memory, set the flash size accordingly. Baudrate 115200 is used for esp8266, not sure what it should be for your board, but you can try.

MTkeg, 2019-07-12:
How do you determine the memory size?

estrauss, 2019-07-12:
I did find the re-initializing method somewhere else and did try it and to no avail.  I might have skipped a step or didn't know what I was doing.
Do you need to press reset after each flash?  And for just a second or 5 seconds?  I've just been power cycling after flashing.
Also, here were my settings with the Wemos D1 Mini Lite (Version 1.0.0)
Baudrate: 115200
Flash size: 1MByte
Flash speed: 40MHz
SPI Mode: DOUT

It could also just be an issue with updating Ubidots.  I decreased the update time to 30 seconds just to test and I think accessing locally via IP address it might be functioning ok.  I really don't care about Ubidots, as it will eventually report directly to my BrewPiLess controllers.  I just need to know that it's operating correctly.

I plan to build a standard Wemos D1 Mini version this weekend with the BAT43 diode and see if it makes a difference.

ErikdBr, 2019-07-12:
> How do you determine the memory size?

It is in the specs of the board. For this specific board, estrauss used the ESP8285, I googled it but I think the nodemcu flash software determines this automaticaly.

ErikdBr, 2019-07-12:
> Do you need to press reset after each flash?
No, it should be operational right away. You'll have to go in configuration mode the first time offcourse and after the first time flash you can flash it to a newer version  OTA(over the air) from the maintenance menu in the config page of the iSpindel.

estrauss, 2019-07-13:
I used the Arduino serial monitor and I'm assuming the Wemos itself might be bad or the accelerometer board might be bad:
`FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 FF 6D 5F 65 15 1 2A
  Chip = DS18B20
  Data = 1 91 1 4B 46 7F FF C 10 70  CRC=70
  Temperature = 25.06 Celsius, 
Boot-Mode: External System
power-cycle or reset detected, config mode

going to Config Mode
reboot RFCAL
`

ErikdBr, 2019-07-13:
FS stands for file system of the esp8266. It is where the iSpindel program and data are stored. Why you get this error I don't know.

ErikdBr, 2019-07-13:
[image]
If I understand this correctly, there isn't anything on pin 5?

estrauss, 2019-07-13:
I'm not sure what to think.  I build another board with a Wemos D1 Mini (ESP8266) and although that build is still not keeping it's battery or temperature unit settings, I have it paired to my BrewPiLess right now.  I'm essentially scrapping the D1 Mini Lite build.
My ultimate goal was just to prove out the connection and not need Ubidots, so I'm at least where I need to be.  I'll do the gravity calculation on the BrewPiLess and I really don't care what the iSpindel temperature reports.  Hopefully I can get the battery voltage working though.

estrauss, 2019-07-15:
Closing this issue, due to suspicion that the original Wemos board had issues, it wasn't really resolved. Other issues I'm having are not part of the title.

closed 2019-07-15 by estrauss