universam1/iSpindel #285: Fairylink-#####

mossyrocks, 2019-07-09:
IV recently soldered an ispindle, i have it together but can seem to flash the firmware. the LED blinks when uploading but after i press the reset button the WIFI name is Fairylink and not Ispindle so it has not flashed correctly. 

i have recently flashed a brewpiless and it works prefectly so im not sure what i am doing wrong,
Am i missing something really obvious? i have tried reflashing with earlier releases.

thanks in advance!

closed 2019-07-09 by mossyrocks

ErikdBr, 2019-07-09:
Well that clarifies it.

mossyrocks, 2019-07-10:
Sorry erik,

I was having issues with the firmware but i downloaded 5.8 and it worked so i ttied to delete my question

ErikdBr, 2019-07-10:
Yes I am having issues as well. Up until 5.8.6, I am fine, anything above that doesn't hold all the configuration data. But only some of us are having that issue.