universam1/iSpindel #287: Unit of Temperature Will Not Save

MTkeg, 2019-07-13:
I just built 2 new iSpindels (first ones) and I would like to change the Unit of Temperature from Degrees Celsius to Degrees Fahrenheit.  Unfortunately, when I change these settings in the dropdown the configuration doesn't seem to save.  I have attached the console output of me making the change.

[Unit of Temperature Change.txt](https://github.com/universam1/iSpindel/files/3388494/Unit.of.Temperature.Change.txt)

Let me know what you think.

MT

OpenSourceDistilling, 2019-11-12:
I had this issue and followed the below link and it fixed it. You need to populate the 2 addresses in the first step then re-flash the firmware in the second step. After that, my unit of temp saved correctly. Before that, the unit of temp would reset each time I powered on/off the iSpindel.  https://www.homebrewtalk.com/forum/threads/ispindle-diy-electronic-hydrometer.598187/page-19#post-8366882

universam1, 2019-11-17:
Firmware 6.2.0: The configuration file routine has been improved to support larger strings that were clipped at some point

MTkeg, 2020-02-15:
> Firmware 6.2.0: The configuration file routine has been improved to support larger strings that were clipped at some point

I just got the chance to upload he 6.2.0 firmware and it appears that the calibration and unit of temperature issues are fixed.  Thanks for pulling this together!

closed 2020-02-15 by MTkeg