universam1/iSpindel #288: reading in water changes

arjepsen, 2019-07-13:
Hey.
I'm trying to calibrate my ispindel, and I've put it in water outputting to ubidots every 20 seconds.
I had to leave it there for a while due to family activites - no problem, right.
But now when I'm looking at the graph in ubidots, it starts out averaging around 74.42.
However, currently it's averaging around 74.54
I think the temperature has gone up 0.6 from 23.1 degrees celcius to 23.7
Is that what changes the tilt here??

ErikdBr, 2019-07-13:
Yep that's it. Other than that, your tilt is way to high, in water it should be around 20 (+/-5).
You'll need to add some weight at the bottom of your iSpinde like a nut and washers until you reach the desired tilt.
[image]

arjepsen, 2019-07-13:
hmm.. I read somewhere that it should be around 25 degrees. I attached a few pieces of metal  to the bottom, and measured the tilt to around 20 degrees from the vertical plane. This gave me the tilt reading of  74.54.

ErikdBr, 2019-07-13:
@arjepsen a reading of 74.54 in plain water or in a sugar solution?

arjepsen, 2019-07-13:
plain water

ErikdBr, 2019-07-13:
I am not getting how you measured the 20 degrees. Calibrating is: go in the iSpindel config menu and go to the maintenance menu. Lay the iSpindel flat on a flat level surface and click calibrate. After that is ready it should give you a tilt of about 90 degrees.
Then with the iSpindel in operating mode put it in plain water and add weight until you get a tilt of about 20 degrees, plus or minus 5 degrees.
So how you measured a tilt of 74 in plain water is a riddle to me.

arjepsen, 2019-07-13:
I might be influenced by working as a math teacher....
I used a degree-measuring-tool (not sure what it's called in english).
According to the ispindel docs, you have to add weights untill the inclination is 25 degrees from vertical.
I added a bit more weight, so now my inclination is about 35 degrees to vertical... and the tilt measures 60.82

ErikdBr, 2019-07-13:
> I used a degree-measuring-tool (not sure what it's called in english).
> According to the ispindel docs, you have to add weights untill the inclination is 25 degrees from vertical.

Yes 25 degrees( + or - 5 ) **tilt** in plain water measured by the iSpindel itself, not by hand with a measuring tool.
If your tilt is 70-ish in plain water it will float upside down in your wort, lol.

arjepsen, 2019-07-14:
> If your tilt is 70-ish in plain water it will float upside down in your wort, lol.
mmm not really.

ErikdBr, 2019-07-14:
I hope this will clearify it. The drawing represents an iSpindel in plain water.
[image]

ErikdBr, 2019-07-14:
The angle of 20 in water represents sg 1.000. When you put it in wort, the angle increases to let's say 60 degrees which could be let's say sg 1.068. When fermentation progresses the angle of the iSpindel decreases because the density of the wort decreases ergo has less resistance ergo will sink a little deeper in the wort.

ErikdBr, 2019-07-14:
A few pics to clearify it more, yesterday afternoon I pitched and got this reading
[image]

Just now I got this:
[image]

Compare the two angles. The angle decreases as fermentation progresses.
Don't mind the temperature it is completely accurate and on purpose because I used a special yeast called Hornindal Kveik which ferments at high temperatures but that is a hole different story.
[image]

ErikdBr, 2019-07-14:
> > If your tilt is 70-ish in plain water it will float upside down in your wort, lol.
> 
> mmm not really.

Sorry about that.

arjepsen, 2019-07-14:
Thanks for your descriptions. I guess I got confused, since it seems the need for weights etc. is only mentioned in the FAQ, not in the general guidelines on the ispindel website. 
Anyways, managed to get it to 25 degrees in clean water. Thanks for your help! :-)

closed 2019-10-09 by universam1