universam1/iSpindel #289: Ispindel from localhost

JETET, 2019-07-18:
Hello I open this post why I can not find information to access from localhost can you tell me something thanks

stefschin, 2019-07-18:
You can use Breepi with a raspberry or Brewpiless to collect data locally, but from iSpindel you can just enter to configuration mode with Reset and it will give you instant data with an IP, but your battery wont last. 
Did you read iSpindle Documentation?

JETET, 2019-07-18:
I would like to use this but I do not see any guide on the subject , do not need any more devices?
[image]

MTkeg, 2019-07-18:
I would start here:  [https://github.com/vitotai/BrewPiLess](url)

JETET, 2019-07-18:
That link does not take me anywhere

stefschin, 2019-07-18:
Just copy and paste in your browser `https://github.com/vitotai/BrewPiLess` or click here [Brepiless](https://github.com/vitotai/BrewPiLess)

JETET, 2019-07-18:
Thanks, I'm going to try

JETET, 2019-07-18:
Looks like I'm clumsy I've got the last of brePiless down, but it makes me wrong.
[image]

stefschin, 2019-07-18:
this is iSpindel topic, not Brewpiless

JETET, 2019-07-18:
Okay, you know how to put the ispindel on localhost?

stefschin, 2019-07-18:
> 
> 
> You can use Breepi with a raspberry or Brewpiless to collect data locally, but from iSpindel you can just enter to configuration mode with Reset and it will give you instant data with an IP, but your battery wont last.
> Did you read iSpindle Documentation?

I told you options you have, there is no localhost possible because iSpindel must save battery and can't be on all the time, it goes to sleep and every x intervals, take measurements, send data to a server and after that goes to sleep mode

JETET, 2019-07-18:
what's the simplest thing?

stefschin, 2019-07-18:
simplest thing is to use some made server, I use BrewSpy app

JETET, 2019-07-18:
but it doesn't continuously update I now have it set to 5s and I don't see it being updated

ErikdBr, 2019-07-18:
I also use the brewspy app. The minimum update frequency is 15s for a maximum of 2 hours for test purposes. After 2 hours it only accepts data every 5 minutes. But that is all in the documentation that can be found at brew-spy.com

JETET, 2019-07-19:
Do you have the program in pro version?

JETET, 2019-07-19:
you know if there is any free software and that it is full

ErikdBr, 2019-07-19:
> Do you have the program in pro version?

Yes.

ErikdBr, 2019-07-19:
> you know if there is any free software and that it is full

https://ubidots.com/education/

JETET, 2019-07-21:
Good morning today I woke up and I found this message
[image]

JETET, 2019-07-21:
I already found the fault is from my Internet line

ErikdBr, 2019-07-21:
> Good morning today I woke up and I found this message

But you got the iSpindel working, very good!

JETET, 2019-07-21:
everything ok ok I have to graduate it

ErikdBr, 2019-07-21:
> everything ok ok I have to graduate it

That beer in you avatar, is that the Diablo Rojo? I would like the recipe :-)

JETET, 2019-07-21:
Yes, send me a pribado, look what I took today to make a red devil tomorrow
[image]

ErikdBr, 2019-07-21:
> Yes, send me a pribado, look what I took today to make a red devil tomorrow

Unfortunately it is not possible to send private messages on github :-(

JETET, 2019-07-21:
jetet34@gmail.com 
this is my e-mail

closed 2019-10-09 by universam1