universam1/iSpindel #291: Leer Ispindel desde otra Wemos

JETET, 2019-07-28:
Seria posible ver los datos del Ispindel en otra wemos como si esta fuese un servidor 

Gracias

universam1, 2019-10-07:
That is technically possible - however if it makes sense that is a different topic. There are number of server solutions already supported and can be referenced incase there is another new one.

closed 2019-10-07 by universam1

JETET, 2019-10-14:
I would like to see the information of an ispindel on this server seeing this information as it is shown by the original ispindel in this server code and to be able to see it in a browser by putting the ip that the wemos d1 mini provides when configuring it Thank you
[image]

JETET, 2019-10-14:
I don't know where to start This is the server
/SERVIDOR
#include <ESP8266WiFi.h>

const char WiFiClave[] = "";  //Sin clave
const char AP_Nombre[] = "Jopapa_ESP8266";

const int LED_PIN = 15; // LED rojo
int val;
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(AP_Nombre, WiFiClave);  
  server.begin();
}

void loop() {
  WiFiClient client = server.available();   // ¿Hay un cliente conectado?
  if (!client) { return;}
  String req = client.readStringUntil('\r');  // Lee la primera linea de la petición del cliente
  if (req.indexOf("/led/0") != -1) val=0;
  if (req.indexOf("/led/1") != -1) val=1;
  digitalWrite(LED_PIN, val); 
  client.flush();}

JETET, 2019-10-14:
I don't want to save Nothing just to visualize