universam1/iSpindel #294: Minimalistic iSpindel tool

plut0nium, 2019-08-20:
Hi,
Not an issue, but I  created a small GUI app that listen to iSpindel devices on the network (through TCP protocol) in order to quickly test and calibrate, without needing to create an account on Ubidots or similar, or to switch between multiple devices in configuration mode.

The reason for this is that I have to assemble and calibrate 10 iSpindels for my homebrewer association...

It is not meant for continuous fermentation monitoring.... there are plenty of other (better) apps or services available for this task.

Fell free to test and comment 🍻
https://github.com/plut0nium/iSpindelTool

JETET, 2019-08-21:
thanks for the application but i don't know what i have to run on the pc thanks

plut0nium, 2019-08-21:
This is a python app.
Just install python3 interpreter and run the *.py script.

ErikdBr, 2019-08-21:
Thanks, I will use this nect time I have to configure an iSpindel.

closed 2019-10-27 by universam1