universam1/iSpindel #295: Issue with flashing new build

moorsey12, 2019-08-28:
I am a novice at any sort of programming, and for that matter, assembling electronics. I can, or at least I thought I could, follow instructions as a reasonably practical person.

I have completed the iSpindel build, and flashed the unit with the latest full release (6.0.6), and whilst I get the iSpindel network appearing, I can connect but not connect to the config screen, 192.164.4.1.

I have pulled up the log on Putty terminal screen, and very much somebody would be kind enough to lend me a hand. Much appreciated in advance.
Ian

{ld▒▒<▒d▒|
▒l▒
#|▒▒▒{▒c▒c▒▒ng▒lng▒▒▒cp▒▒${l{lx▒o▒
▒l▒▒
b
n▒<l▒
▒b▒▒gn▒d▒▒l ▒l orǛ▒n
b▒lsۓoc▒l▒▒8▒▒▒`▒o▒
                   FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 17 83 79 A2 16 3 63
  Chip = DS18B20
  Data = 1 A1 1 55 5 7F A5 A5 66 B  CRC=B
  Temperature = 26.06 Celsius,
Boot-Mode: External System
power-cycle or reset detected, config mode
...........
           ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
{ll▒▒|▒$▒|
▒l▒
c<ǃ▒▒r▒b▒
b▒▒gn▒dgn▒▒▒
#x▒▒lrdrdp▒g▒
▒
$
▒▒
cg▒|d▒
▒c▒▒gn▒d▒▒l ▒$`n;Ǜ▒n
#▒ls▒▒n
b▒l▒l▒x▒▒▒`▒g▒
              FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 17 83 79 A2 16 3 63
  Chip = DS18B20
  Data = 1 A1 1 55 5 7F A5 A5 66 B  CRC=B
  Temperature = 26.06 Celsius,
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
           ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: DUP AP: BTHub6-JMX8
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

moorsey12, 2019-08-28:
Apologies, typo on the thread. I typed 192.168.4.1 into my phones web browser, but it just times out on every occasion when connected to the iSpindel network.

Is it connected to:

Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 17 83 79 A2 16 3 63
Chip = DS18B20
Data = 1 A1 1 55 5 7F A5 A5 66 B CRC=B
Temperature = 26.06 Celsius,
Boot-Mode: Deep-Sleep Wake

universam1, 2019-10-27:
You might try to connect with a different device again

closed 2019-10-27 by universam1