universam1/iSpindel #297: Can't get web page for initial configuration to work.

taquitosensei, 2019-09-04:
I flashed it with the 6.0.6 firmware. It's in configuration mode, the blue led is blinking. The SSID is ESP_30F855 instead of ispindel. I get the correct IP information though. My IP is 192.168.4.2 and the Gateway is 192.168.4.1  I can ping the .4.1 address. But whenever I try to browse to it. I get an error "network change has been detected". If I try it in Edge I get "page could not be loaded."

ErikdBr, 2019-09-04:
Did you build your iSpindel completely, because that is your problem I think

taquitosensei, 2019-09-04:
Not sure what you mean by that? It's all assembled but not in the drawer or tube. I'm 3d printing the drawer this afternoon. Or is there a software step I'm missing? The guide goes straight to flashing the firmware and configuring after assembling the ispindel.

ErikdBr, 2019-09-04:
The SSID should be iSpindel, not ESP_30F855. If it is completely assembled then the flashing went wrong.

taquitosensei, 2019-09-04:
I just flashed it again. Using esptool in linux instead of NodeMCU in windows. It looks like it flashed ok. But I still don't have the iSpindel SSID, just the ESP_30F855 

sudo ./esptool -vv -cd nodemcu -cb 921600 -cp /dev/ttyUSB0 -ca 0x00000 -cf ./firmware.bin 
[sudo] password for tomb: 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 921600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 440336 bytes from ./firmware.bin to flash at 0x00000000
	erasing flash
	size: 06b810 address: 000000
	first_sector_index: 0
	total_sector_count: 108
	head_sector_count: 16
	adjusted_sector_count: 92
	erase_size: 05c000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 1000
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 18% ]
................................................................................ [ 37% ]
................................................................................ [ 55% ]
................................................................................ [ 74% ]
................................................................................ [ 92% ]
...............................                                                  [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader

ErikdBr, 2019-09-04:
It looks like it flashed ok, but if you don't get iSpindel as SSID then it didn't. The baudrate should be 115200.

taquitosensei, 2019-09-04:
Sorry for being a pain. I switched the baud rate to 115200, looks like that wasn't necessary default is 115200. It looks like it flashed ok, it went slower this time, but still getting the wrong SSID. Is there a way to figure out why it's not flashing correctly?

ErikdBr, 2019-09-04:
Not at all, I get it that you want it to work.
Try "What should I do if my iSpindel isn't working properly after flashing it with the firmware?" in the  [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md)

taquitosensei, 2019-09-04:
Ok so here is what I get from Serial Log in the Arduino IDE. If I'm reading this right. I've got something hooked up wrong.   ERROR no Wifi credentials
FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Boot-Mode: Deep-Sleep Wake

There's a line of ascii above that, that won't copy properly.

vytux-com, 2019-09-04:
Looks like you haven't attached a temperature sensor or it is not working

taquitosensei, 2019-09-04:
It's attached and all the solder joints look good. Guess I'll start from scratch and see what happens. I bought extra components for a reason lol

ErikdBr, 2019-09-04:
Good luck

piratebriggs, 2019-09-06:
Sorry for the late reply, I saw the "No devices found!" error when I'd connected the temperature sensor but not the appropriate pull up resistor. Maybe that's something that you could check?

taquitosensei, 2019-09-06:
I'm going to try it again this weekend. I'll keep an eye out for that. It should have been connected, but I had a custom pcb that had 6 connections to the charger and my charger only had 4, so I had to tie the negatives together, and the positives together. I've got the correct charger now though.

closed 2019-10-09 by universam1