universam1/iSpindel #300: Temp Sensor Calibration

Novastart171, 2019-09-18:
Hi, Is there an option for temp sensor calibration.  I have two iSpindels built and at 20.0C I have one reading 18.8C and the other reading 16.9C  I have verified the 20C temp results with 5 different temp sensors inside my Brewpiless controlled Freezer.  I have also tried different Temperatures. Currently using release 6.0.6

SunSun1979, 2019-09-18:
Also have this problem, t from iSpindel < real t ~4-6C, bad china DS sensor?

Novastart171, 2019-09-18:
> Also have this problem, t from iSpindel < real t ~4-6C, bad china DS sensor?

Quite possible, had them for a while and cant recall where i got them from, probably eBay.   With that being said they seem to be consistently out by same amount across the temp range and it will be difficult but not impossible to replace the sensor on the assembled boards

ErikdBr, 2019-09-18:
Built 2 iSpindels up to now, both with a ds18b20 sensor from aliexpress and both giving the exact temperature of the wort +/- 0.1 C.
But this might help you make a derived variable from your temp readings:
https://www.letscontrolit.com/wiki/index.php/Basics:_Calibration_and_Accuracy

universam1, 2019-10-09:
remember you will have bad reading if your interval is under 2min. If that does not apply we can reopen

closed 2019-10-09 by universam1

SunSun1979, 2019-10-10:
30m interval

Novastart171, 2019-10-10:
Just an update bit the bullet and replaced both DS temp sensors with ones from a more reputable source and they are +/- 0.1 C of each other and about  +/- 0.5 of other temp sensors at a 2 min interval and gets better as the interval increases.  Problem solved and lesson learned thank you.

universam1, 2019-10-10:
Very interesting- so there are actually counterfeits out there

nadir-dogan, 2025-02-16:
I have both original Dallas and Chinese DS18B20s. I used them on Leonardo and on ESP32-C6 unfortunately everytime results is 3~4 C higher than actual temperature. I'm working on this issue about one week. Not found any solution yet. If anyone have helpful knowledge I will be more happy. Thanks...