universam1/iSpindel #302: Status page displays 0.00 tilt but displays tilt on Ubidots

flowidc, 2019-10-05:
I have built several units and found that on some of them the status page works correctly and displays the tilt value but on some units it shows a 0.00. All of the units send the correct tilt value to Ubidots and Craftbeerpi so the issue seems to be isolated to the status page. Could this be a hardware/soldering issue or software? I have tried using v6.0.6 and 6.1.1 and get the same results with both versions. Thanks.

flowidc, 2019-10-06:
Here is the PuTTY terminal screen. It seems to be repeating "Acc Test Connection ERROR!". It doesn't have this error on my other device. 

             FW 6.0.6
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 6D 60 79 A2 0 3 E6
  Chip = DS18B20
  Data = 1 53 1 55 5 7F A5 A5 66 AA  CRC=AA
  Temperature = 21.19 Celsius,
Acc Test Connection ERROR!
Boot-Mode: Deep-Sleep Wake

Double Reset detected

going to Config Mode
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: Scan done
*WM: DUP AP: Brewhouse24
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!

flowidc, 2019-10-07:
It seems that the getDeviceID() in MPU6050_Base::testConnection() does not equal 0x34. It was equal to 57 on one of my units. 
In the flash() function testAccel() is called every second and since it returns false it does not read the tilt angle in the info page. By removing the if testAccel() line in the flash() function, the tilt information is reading on the info page.

universam1, 2019-10-07:
Interesting - can you double check which sensor you are using, is it reproducible with other iSpindels?
I'm happy to add another device code if that makes sense

universam1, 2019-10-07:
also interested if it can be related with #275

flowidc, 2019-10-11:
Sorry for the late reply. Here is the gyro sensor I am using. 
https://www.amazon.com/gp/aw/d/B00LP25V1A?psc=1&ref=ppx_pop_mob_b_asin_title

So far 2 of the 3 sensors I have had this issue.

pppedrillo, 2019-10-23:
@flowidc  
Here is excerpt from the specs - its WHO_AM_I must be 0x34 not 0x57 as you mentioned. Moreover bits 0 and 7 are reserved (hardcoded to 0 according to specs), and 0x57 has bit 0 set which is against the specs. Looks like these HiLetgo either build against these specs or (coz you mentioned only 2 of total 3 yours gives you wrong 57 value) modules are initially faulty.

@universam1 

[image]

universam1, 2019-10-27:
Thanks for the look up.
It’s questionable to add support for such an exception as it these devices might have other issues as well

closed 2019-11-17 by universam1

Naesstrom, 2019-12-04:
I'm having the same issue with some of these gyros (https://banggood.app.link/sBD0Z82v91) that I've installed. Atleast they show 0.0 tilt. Can you give me some step by step instructions what you did @flowidc to get them to work!

flowidc, 2019-12-05:
@Naesstrom, I don't think we were having the same issue. My issue was isolated to the Info page on the iSpindel device. On that page I would get a 0.00 for the tilt value. It worked fine when sending the tilt data to Ubidots or Craftbeerpi. In any case what I did to get it to work was to comment out the if testAccel() under the flash() function (in the file pio/src/ispindel.cpp) so it looks like:
```
void flash()
{
  // triggers the LED
  Volt = getBattery();
  //if (testAccel())
  getAccSample();
  Tilt = calculateTilt();
  Temperatur = getTemperature(false);
  Gravity = calculateGravity();
  requestTemp();
}
```
Then compile the source code and reflash your device. 

Again, this probably not the same issue you are having and my "fix" for it is really just a band-aid that doesn't address the root cause. It seems that the gyros I was using was not built to spec therefore ordering another gyro might be more appropriate.

Naesstrom, 2019-12-06:
@flowidc yeah I had a couple that sent the same data to ubidots like above but the rest that I tested had the same problem as yours with just the config screen showing 0.0 but the right values being sent!

Would be great to be able to eliminate what's messing it up, and be able not to worry about getting the wrong ACC's. The ones I have "look" like the right ones but apparently there's something strange!