universam1/iSpindel #304: Service type blank

taylora1-1, 2019-10-08:
Hi, I have upgraded to revision 6.0.5 and there appears to be an issue with the drop down box for service type when i connect to my device over wifi.  I have tried MS Edge, IE11, Chrome, Opera and firefox on my windows 10 machine and either the drop down box doesn't work or, if it drops down the list of service options are blank.  I have also tried with an android device and chrome / opera browsers but with teh same result.  I have attached an image with a screenshot.

I have also tried rolling the firmware back to an earlier version but I still have the same problem.

any help would be much appreciated.

[New Microsoft Word Document.docx](https://github.com/universam1/iSpindel/files/3703968/New.Microsoft.Word.Document.docx)

universam1, 2019-10-08:
How about trying the current version?

taylora1-1, 2019-10-08:
Hi Sam, thanks for getting back to me.  I have just tried 6.0.6 and 6.1.1 and u fortunately both versions still give me the same problem.  This did work OK previously and strangely, the drop down box for the temperature scale does appear to work OK in the different browsers so I am not sure what the 'service type' doe not work correctly. Kind regards, Andrew

taylora1-1, 2019-10-09:
Hi Sam,

An update, ihave my unit working with version 5.9.3 from the following link but when I then load 6.0.6 from your profile I get the problem with the service type.

https://docs.brewersfriend.com/devices/ispindel

Do you have any further thoughts?

taylora1-1, 2019-10-10:
Hi Sam, a further update, today I tried reloading version 5.9.3 followed by your release of 6.0.6 and its now all working properly.  I have the drop down box with the options for service type.  This ticket can be closed. Many thanks.  Andrew.

closed 2019-10-27 by universam1