universam1/iSpindel #309: HTTP configuration concatenates server URL

dualcyclone, 2019-11-08:
I am currently building an Amazon AWS Lambda to accept data from my iSpindel, and send the data to a google spreadsheet.

However, when I edit the configuration to use the Amazon AWS Lambda URL, it concatenates it.

Original: `XXXXXXXX.execute-api.eu-west-2.amazonaws.com`
After saving: `XXXXXXXX.execute-api.eu-west-2.amazon`

The portal should not concatenate this URL

universam1, 2019-11-08:
AWS Lambda is an interesting use case - would be nice if you could post a working example?

Could you state the exact length of the url - that would help to determine where the limit happens.

lbussy, 2019-11-08:
Luke are you using AWS to get around the https requirement?

thorrak, 2019-11-08:
Sadly, my solution is basically the same thing - just not hosted on AWS. 👎 

HTTPS is expensive (for an ESP8266) :(

dualcyclone, 2019-11-09:
I'm using AWS Lamba to post data to a google spreadsheet.

I have mapped to my own domain now though, but no data. Reading from the serial port I'm getting:

```
calling HTTP
{"name":"Black","ID":7870127,"angle":62.7542,"temperature":25.125,"temp_units":"C","battery":4.442127,"gravity":19.67928,"interval":600,"RSSI":-77}HTTPAPI: posting
code: -1
[HTTP] POST... failed, error: connection refused
```

...So the iSpindel is calling over HTTP and not HTTPS @thorrak ?

If you guys know how I can work around this, that'd be handy

thorrak, 2019-11-09:
Yes, I believe so. The question I have (and without speaking for him, the one I believe @lbussy was hinting at) is: Why are you using a lambda at all, rather than going directly to google sheets?

My assumption was that it was because Google Sheets requires HTTPS and the ESP can’t handle it on its own. Are you using the lambda to do something else?

lbussy, 2019-11-09:
> So the iSpindel is calling over HTTP and not HTTPS

That’s it basically.  I think @thorrak had a way around it with a proxy sort of thing, but I’ve not figured it out yet.  Granted, my “work” on it so far has been Googling.  

I’d think you need a system to which you can post the data, which inherently understands TLS.  I’d probably make an endpoint in Python (or even keep it easy and do PHP) to receive the post, then turn around and post the same payload to the https endpoint.  

Then again I may be making it more difficult than it needs to be. :)

thorrak, 2019-11-09:
> I’d think you need a system to which you can post the data, which inherently understands TLS. I’d probably make an endpoint in Python (or even keep it easy and do PHP) to receive the post, then turn [...]

That’s exactly what my hack is doing. I’ll post a gist with it when I get home - it’s very short. :)

thorrak, 2019-11-09:
That said, I think the initial point still applies - the iSpindel may be truncating the URL.

dualcyclone, 2019-11-09:
Well the iSpindel can't send data directly to a Google sheet without me providing some basic authentication, which I've done using a lambda.

Everything works just fine, but I didn't realise the iSpindel cannot use HTTPS, anybody know why?

So I've discovered I can use an ALB to route HTTP traffic, but it's becoming pretty complicated.

I've written the lambda in JavaScript, very straightforward, it's literally a few lines.

lbussy, 2019-11-09:
> Everything works just fine, but I didn't realise the iSpindel cannot use HTTPS, anybody know why?

The libs which support HTTPS are clunky and huge and don't lend themselves well to having any other functionality.

Consider how a "regular" OS already has root CA's as part of it's installation.  That allows the OS to trust other certificates which can be traced to an accepted root CA.  A controller has none of that so you sort of have to build it all.

universam1, 2019-11-10:
@dualcyclone @thorrak Firmware 6.1.3 now supports URL length up to 256 character. I believe this issue can be closed then

dualcyclone, 2019-11-10:
> @dualcyclone @thorrak Firmware 6.1.3 now supports URL length up to 256 character. I believe this issue can be closed then.

No - the server address is still truncated to 39 characters,

Original: iSpindelHTTP-XXXXXXXXXX.eu-west-2.elb.amazonaws.com
AFTER: iSpindelHTTP-XXXXXXXXXX.eu-west-2.elb.a

dualcyclone, 2019-11-10:
I've just updated to my CNAME record again, but now I'm getting a rolling reboot:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

   FW 6.1.3
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file

Exception (28):
               epc1=0x40227472 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

                                                                                                  >>>stack>>>

                                                                                                             ctx: cont
                                                                                                                      sp: 3fff0e80 end: 3fff1120 offset: 01a0
                                                                                                                                                             3fff1020:  00000000 00000000 3fff1058 402052f7
                                                                                                                                                                                                             3fff1030:  40222170 00000000 000003e8 40210934
                          3fff1040:  3fff180c 3fff184c 00000000 00000000
                                                                          3fff1050:  3ffefb84 3ffef8ac 3fff2444 3fff2444
                                                                                                                          3fff1060:  3fff2714 3fff2844 3fff2834 3fff2714
                                                                                                                                                                          3fff1070:  3ffef8a0 00000000 00000000 3fff23e1
                                                                                                                                                                                                                          3fff1080:  3f007d0a 00000000 3ffef88c 00000000
                                       3fff1090:  c4a54769 5db2e741 3ffef8ac 3ffef9cc
                                                                                       3fff10a0:  3ffef88c 00000000 3ffef8ac 40207064
                                                                                                                                       3fff10b0:  feefeffe feefeffe feefeffe feefeffe
                                                                                                                                                                                       3fff10c0:  feefeffe feefeffe feefeffe feefeffe
    3fff10d0:  feefeffe feefeffe feefeffe feefeffe
                                                    3fff10e0:  feefeffe feefeffe feefeffe feefeffe
                                                                                                    3fff10f0:  feefeffe feefeffe feefeffe 3ffef9cc
                                                                                                                                                    3fff1100:  3fffdad0 00000000 3ffef99c 40213530
                                                                                                                                                                                                    3fff1110:  feefeffe feefeffe 3ffe8680 401010e5
                 <<<stack<<<
```

Anybody know how to force a reset?

lbussy, 2019-11-10:
> Anybody know how to force a reset?

I suspect @universam1 missed a place where that string is copying which causes the error.  You may have to erase the flash and start over to get rid of that string which is causing the crash.

Assuming you are on Linux, you can erase with the following command:

```
esptool erase_flash
```

If it's not installed, install with:

```
pip install esptool
```

That assumes your device is the only controller plugged in.  If you have more than one be sure to specify the port or else it will erase the first one it finds.

If you are on Windows, you can either use `esptool` from Python, or use Espressif's Flash Download tool which has an `Erase` button.

dualcyclone, 2019-11-10:
Looks like the instructions to install firmware are out of date
```
esptool.py --port=/dev/tty.usbserial-XXXXXX  write_flash  -fm=dio -fs=32m 0x00000 ~/Downloads/firmware.bin
```

lbussy, 2019-11-10:
They are not out of date, I only showed the command to erase the flash.  You would have to follow that with flashing the firmware.  Re-flashing the firmware alone will not erase the configuration.

The command line shown is more explicit with the port declaration.  As I said, if you only have one device plugged in, `esptool` will autmatically find it.

universam1, 2019-11-10:
@dualcyclone A little bit of more effort from your side might help to solve your issue. For instance your terminal log file would help.

Yes the URL length is extended @lbussy. Again, if you would provide some logs it would have been clear from the start that you are actually referring to the server address not the URL which is a different field.

dualcyclone, 2019-11-10:
I'll get you some logs @universam1 , but the original opening description would have been clear I was referring to a domain name.

With regards to effort, I've learned and written Amazon AWS Lambda functions to write to Google Sheets, deployment, Amazon Application Load Balancers because the iSpindel can't send requests to HTTPS addresses, all with sparse documentation. I opened this issue when I discovered the server domain was being truncated.

I've even deployed 6.1.3 onto my iSpindel, where it broke, erased it, and deployed new firmware, where I have gone to the effort to state the command I had to use, because the one on https://github.com/universam1/iSpindel/blob/master/docs/Firmware_en.md under `Linux` did not work, so I have provided it here to be helpful.

dualcyclone, 2019-11-10:
Here is a log - I have opened the configuration page, updated the `Server Address` to the ALB domain, and saved.

It has then rebooted, and attempted to send the initial requests. You will see on the line `saving config...` that the Server domain is truncated as I have described before
```
*WM: Handle root
*WM: HUAWEI-B310-AF38
*WM: -63
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: blue
*WM: Parameter
*WM: sleep
*WM: 600
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: tempscale
*WM: 0
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: selAPI
*WM: 3
*WM: Parameter
*WM: token
*WM:
*WM: Parameter
*WM: server
*WM: iSpindelHTTP-XXXXXXXXXX.eu-west-2.elb.amazonaws.com
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM: /
*WM: Parameter
*WM: db
*WM: ispindel
*WM: Parameter
*WM: username
*WM:
*WM: Parameter
*WM: password
*WM:
*WM: Parameter
*WM: job
*WM: ispindel
*WM: Parameter
*WM: instance
*WM: 000
*WM: Parameter
*WM:
*WM:
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.26
*WM: seconds
*WM: Connection result:
*WM: WL_CONNECTED
saving config...{"Name":"blue","Token":"","Sleep":600,"Server":"iSpindelHTTP-XXXXXXXXXX.eu-west-2.elb.a","API":3,"Port":80,"URL":"/","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"HUAWEI-B310-AF38","PSK":"XXXXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}saved successfully
Samples:36 min:81.72 max:82.06 time:627
x: -2586 y: 17402 z: 478
Tilt: 81.88
Tacc: 23.06
Volt: 4.35
Temp: 20.37
Gravity: 29.47
IP: 192.168.8.111

calling HTTP
{"name":"blue","ID":7870127,"angle":81.87832,"temperature":20.375,"temp_units":"C","battery":4.348279,"gravity":29.47396,"interval":1,"RSSI":-71}HTTPAPI: posting
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 1s; RT: 50967
{ll��|�d�|�$�c|����r�#�c��'o�$gn���#x��l{l{lp�n��d��b'�|$��c��gn�d��$`�'l`or�ۓn#�$r��'c�d��ll��l �g�
                                                                                                    FW 6.1.1
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
offsets not available
parsed config:
{"Name":"blue","Token":"","Sleep":600,"Server":"iSpindelHTTP-2109475467.eu-west-2.elb.a","API":3,"Port":80,"URL":"/","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"HUAWEI-B310-AF38","PSK":"NTY3502TRF9","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:81.78 max:82.04 time:758
x: -2570 y: 17376 z: 474
Tilt: 81.90
Tacc: 22.97
Volt: 4.35
Temp: 20.37
Gravity: 29.48
IP: 192.168.8.111

calling HTTP
{"name":"blue","ID":7870127,"angle":81.89736,"temperature":20.375,"temp_units":"C","battery":4.348279,"gravity":29.4836,"interval":600,"RSSI":-69}HTTPAPI: posting
code: -1
[HTTP] POST... failed, error: connection refused

Final-sleep: 600s; RT: 1354
```

dualcyclone, 2019-11-10:
I have pulled the project down, and tried to fix this for you - but alas I do not have permissions to push my branch to your project.

Alas, it was a minor fix anyway; I've attached a patch.

I provide this completely untested.

universam1, 2019-11-17:
Thank you @dualcyclone the fix is included in 6.2.0

closed 2019-11-17 by universam1