universam1/iSpindel #310: Update to 6.1.3 unable to save POLYN

grahambayley, 2019-11-15:
Hi 

Just upgraded from older version (6.0.6) to 6.1.3 and i can save my POLYN

I started off the upgrade with a flash of the 8266 following with nodemcu flasher thru usb

CONFIG
1st Line "INTERNAL://NODEMCU" Address "0x00000"
2nd Line "INTERNAL://BLANK" Address "0x01000"

Then Flashed the 6.1.3 firmware, then did a factory reset and re-entered my standard setup, all was ok.

Did a Tilt Calibration of the unit and then re-calibrated the POLYN the result was entered and saved 

But it did not save the POLYN (see below)
[iSpindel_v1.log](https://github.com/universam1/iSpindel/files/3853054/iSpindel_v1.log)

What did i do wrong  ?

Many thanks in advance

Graham

`*WM: 
*WM: Parameter
*WM: POLYN
*WM: 0.003427333*tilt^2+0.233015213*tilt-7.46648431
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.52
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...
{"Name":"iSpindel_v1","Token":"","Sleep":10,"Server":"10.0.1.20","API":6,"Port":9501,"Channel":0,"URL":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"SSID":"YYYY","PSK":"XXXXXXX"}saved successfully
Samples:30 min:77.75 max:77.89 time:509
x: 3760 y: 15854 z: -56
Tilt: 77.82
Tacc: 25.36
Volt: 5.34
Temp: 22.81
Gravity: 31.42
IP: 10.0.1.64

calling TCP
{"name":"iSpindel_v1","ID":1870319,"angle":77.81593,"temperature":22.8125,"temp_units":"C","battery":5.338894,"gravity":31.4194,"interval":1,"RSSI":-54}
Sender: TCP stream
6

Final-sleep: 1s; RT: 82572`

universam1, 2019-11-15:
I can confirm that there is a configuration file size limitation that can hit you especially saving long strings. 
Working on a solution on soon coming update

kahunalux, 2019-11-16:
I couldn't change the polynomial too so I have tried to put it in the URL like this:

192.168.4.1/wifisave?POLYN=0123455....

Problem seems every other values are blank now and the ispindel blinks a bit quickly with the blue strong light. Impossible to go in the configuration menu.... Wifi seems to not work anymore.

Is it a possibility to make an hard factory reset or to use nodemCu to do it?

kahunalux, 2019-11-16:
It's ok, ESP8266 download tool has an erase bottom.
Now it works...

universam1, 2019-11-17:
Firmware 6.2.0 is out: the configuration file routine has been improved to support larger strings that were clipped at some point

grahambayley, 2019-11-18:
Thanks Sam - works again ;) time to prove it and brew another !

closed 2019-11-18 by grahambayley