universam1/iSpindel #316: IP-address and MAC-address

mcmellow, 2019-12-02:
How can I discover the IP-address and MAC-address of my configured iSpindel? 
In line with this. Can I assign a fixed or static IP-address to my iSpindel?

ErikdBr, 2019-12-02:
Yes you can in the configuration of your router or accesspoint.

mcmellow, 2019-12-02:
> Yes you can in the configuration of your router or accesspoint.
Thanks, that is the dynamic IP-address given by my router. Can I configure a static IP address in the iSpindel?

universam1, 2019-12-02:
not supported, the IP reservation should suffice

ErikdBr, 2019-12-02:
In my Zyxel router I can configure that a device gets a static ip from the router so no need to configure the device itself. It just gets the same ip from the router. Or am I not getting something here?

mcmellow, 2019-12-02:
Thanks, I get it.  I will check this out.

closed 2019-12-02 by mcmellow