universam1/iSpindel #322: Gravity: 0, "Parse error 12"

lbussy, 2019-12-09:
I calibrated vs a hydrometer at three points:

```
"angle": 22.9584
"gravity": -0.075496
Hydrometer:  1.0000

"angle": 43.16552
"gravity": -0.218134
Hydrometer:  1.029

"angle": 62.21345
"gravity": -0.326197
Hydrometer:  1.063
```

I put the points in the [web page](http://www.ispindel.de/tools/calibration/calibration.htm) and got the following:

Degree 2:y = 0.000008873x^2 + 0.000849961x + 0.975778578
Degree 3:y = 0.000001408x^3 + -0.000171842x^2 + 0.008040437x + 0.888885446

I tried both `0.000008873x^2 + 0.000849961x + 0.975778578` and `0.000001408x^3 + -0.000171842x^2 + 0.008040437x + 0.888885446` in the configuration portal and only get "0" as a gravity (although you can see my numbers pre-calibration, they are not always "0".)

The JSON I read out of the device matches what I see in the serial monitor:

```
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel-Yellow","Token":"","Sleep":1,"Server":"brewpi.local","API":3,"Port":80,"Channel":0,"URI":"/brewpi-api.php","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":1,"OWpin":12,"POLY":"0.000008873x^2 + 0.000849961x + 0.975778578","SSID":"Bohica24ghz","PSK":"xxxxxxxx","Offset":[-910,-468,1584,181,35,3]}
applying offsets: -910:-468:1584
confirming offsets: -910:-468:1584
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:44.16 max:44.59 time:748
x: -504 y: 11412 z: 11714
Tilt: 44.36
Tacc: 32.47
Volt: 4.03
Temp: 28.25
Parse error at 12
Gravity: 0.00
IP: 192.168.168.136

calling HTTP
{"name":"iSpindel-Yellow","ID":2027057,"angle":44.36437,"temperature":82.85,"temp_units":"F","battery":4.03024,"gravity":0,"interval":1,"RSSI":-45}
HTTPAPI: posting
code: 200
string(8) "200 (Ok)"

Final-sleep: 1s; RT: 2571
```

All I can see is the `Parse error at 12` but some searching for that phrase netted me nothing.  Fairly certain I'm doing something wrong, just not sure what it might be.

ErikdBr, 2019-12-10:
I have to ask, did you perform the 1st calibration step where you put the iSpindel on a flat surface where you get an angle of about 89-90-ish?

lbussy, 2019-12-10:
Yes, sir.  Don't worry about asking questions you think are basic.  Clearly I missed something. :)

sergioguimaraes, 2019-12-10:
You must remove all spaces from your equation and replace x to tilt on it.

lbussy, 2019-12-10:
> You must remove all spaces from your equation and replace x to tilt on it.

Thank you, I'll try that here in a little bit.

ErikdBr, 2019-12-10:
Yes that is it. It should look like:

`0.000008873*tilt^2+0.000849961*tilt+0.975778578`

lbussy, 2019-12-10:
Well, that makes sense, however when I tried that I had (almost the) same result.  Not being one to feel helpless, I stepped through the problem and realized you gave me the final answer (re-writing the formula for me).  However, I did not comprehend that `x` should be replaced with `* tilt` instead of just `tilt`.  I was completely missing the `*` part.  It was only when I plugged it all in Excel that it made sense.

To make up for my stubborness, I created a patch for the calibration page which will avoid this issue for future people such as myself.  @universam1 you can use or discard this as you see fit:

```
363,365c363,365
<                 $("#secondpoly").html(secondRegression.string);
<                 $("#thirdpoly").html(thirdRegression.string);
<                 // calcuate errors
---
>                 $("#secondpoly").html(secondRegression.string.split("x").join("*tilt"));
>                 $("#thirdpoly").html(thirdRegression.string.split("x").join("*tilt"));
>                 // caluate errors
```

When so patched, the page will display formulas which may be safely copy/pasted into the configuration portal without interpretation:

```
Degree 2:y= 0.000008912*tilt^2 + 0.000845855*tilt + 0.975883228
Degree 3:y= -1.2e-7*tilt^3 + 0.000024316*tilt^2 + 0.000233117*tilt + 0.983283673
```

closed 2019-12-10 by lbussy

universam1, 2019-12-10:
Thank you for the contribution