universam1/iSpindel #323: Ispindel sends data once then will not send anything else

LRowley97, 2019-12-16:
Hi all, 
firstly just wanted to say what a great project.  
Currently, my Ispindel will send information fine while in configuration mode giving correct values etc. However when I take it out of configuration mode and to the Ubidots interface it will only send data once then will not send it again unless the reset button is pressed.  

I am aware that this is a common problem with connection between D0 and Rst pins on the Wemos D1 mini but I have tried several of the suggestions from other users including 470-ohm, 330-ohm, 270ohm resistors, bat 46 diode and a Bat 85 diode none of these seem to have done the trick and I still get the same error. 

In addition to changing the diodes etc. I have also tried several different D1 mini's in an attempt to solve this issue but always end up with the same problem. I have also experimented with a few different firmware options although I have now gone back to the firmware from this repository. 

I have used putty to check the output of the Wemos D1 mini it starts after I push the reset button it goes through all the normal operations then gets to the sleep stage waits for the prescribed amount of time in the example 60s then seems to fail as it produces random characters. Which does suggest to me that the issue is still with the reset problem but I am just out of ideas as to what could be causing it. 

[image]

Also linked below is the circuit diagram and a PCB layout that I am using for this project. Just to make sure I am not missing something extremely obvious in the implementation 

[image]

[image]

Any ideas that people could offer or places to go looking for information would be much appreciated. I hope that I have included enough information if there is anything missing or there are any issues with my post please let me know. 

Cheers 
Logan

sqztom, 2019-12-28:
#metoo :(
8266 going asleep and waking up in programming mode. Found similar issues:
https://github.com/esp8266/Arduino/issues/5892
https://github.com/esp8266/Arduino/issues/6007
And i made some more tests: 
- board with pogo pins ( for fast module change)
- several borrowed ESP with the same iSpindel code inside
effects:
modules marked DOIT.AM - 5 tested, 0 working
marked Ai - 6 tested, 4 working ( the two that failed seems to be a poor copies)
WeMos unshielded D1 modules - 3 tested, 2 working

Modules that passed test just once are working flawlessly.
First one is now "fermenting in sugar" and gathering Ubi dots :)

Looks like modules we are buying are pure lottery. Some of them ale copies of junk clones made on monday morning.
Solution: get another module

LRowley97, 2019-12-29:
Thanks, @sqztom  yeah have switched to a new board and a put into a breadboard. Seems to be working well at the moment will double-check the circuit and then hopefully put everything back together and make it work.

closed 2020-06-23 by stale[bot]