universam1/iSpindel #325: Tilt: 0.00°

sbonaime, 2019-12-20:
Hi,

I just built my Ispindel with PCB 4.0 and Weemos D1 Mini V3.  With Firmware 5.8.6, I have 
Tilt: 0.00°
temperature: -127°

With telnet I have the following messages:

I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
I2C ERROR: 0 con:0

I understand than the lower chip which takes care of tilt and temperature can't connect through I2C. What can I check ? Thank you

[image]
[image]

sbonaime, 2019-12-20:
I did some wrong soldering on the Weemos ! Now It is working correctly
[image]

closed 2019-12-20 by sbonaime

langerc, 2020-05-25:
The wiring diagram is inconsistent the pins D3 and D6 have to be connected. In most manuals they are left out which results in
error can not load json
or the error above because it can not read the i2c bus.