universam1/iSpindel #326: Circuit diagram has errors

DanielO, 2019-12-26:
Hi,
I was following the English diagram (https://github.com/universam1/iSpindel/blob/master/docs/circuit_diagram_en.md) and I think it has a few errors
1 - The 1 wire line is pulled up to 5V, which means the ESP8266 sees 5V on an IO (which I believe is out of spec)
2 - The DS18B20 is powered by 5V not 3.3V (after fixing the above problem)
3 - The battery monitor line (A0) is connected to the output of the DC/DC converter rather than the battery.

I was going to make an updated version but I couldn't find the TP4056 or 18650 battery part in Fritzing..

universam1, 2019-12-26:
Some thoughts on that:
1. The battery line will not see 5V but maximum the battery voltage which is <4.2V. With the 4k7 res. and because D6 is floating I considered it an acceptable condition, necessary because of next:
2. The DS18B20 runs not under 3V, or a battery voltage of ~3.2V. However the ESP does work fine with ~0.4V lower, which increases the lifetime to a good degree.
3. Check the datasheet of the battery charge module (which is not a DC/DC) you will see that the earth is switched and the + is connected. With other words they are the same line

DanielO, 2019-12-26:
Oh, I thought it was a DC/DC converter - the English documentation just says "Lipo Lademodul" the top hit on Google for that is https://www.banggood.com/TP4056-Micro-USB-5V-1A-Lithium-Battery-Charging-Module-Board-TE585-Lipo-Charger-p-1225993.html?rmmds=buy&cur_warehouse=CN which is a combination LiPo charger and DC/DC converter.

Also, the diagram shows OUT-/+ which the TP4056 charging only modules do not seem to have (only IN-/+ and BAT-/+) eg https://www.ebay.com.au/itm/Lipo-Lion-1S-Akku-Mini-USB-Lademodul-TP4056-fuer-Solar-charger-Arduino-/401179690746

Perhaps the documentation should have a "theory of operation" section or similar which explains how it is supposed to work which would make this sort of thing clearer.

universam1, 2019-12-27:
Attention here that the 4 vsr 6 pin version of TP4056 are completely different animals.
As often with chinese parts they are all but normed hence its hard to describe the part list accurately - so far in the parts list https://github.com/universam1/iSpindel/blob/20d6b3c7b54ca0bb8394dd7d9a4172a2480ea557/docs/Parts_en.md I've mentioned it as "Lipo charger TP4056 (6 pin recommended)". If you have suggestions for documentation you are welcome to raise a PR

DanielO, 2019-12-28:
Do you have a schematic or something for the one you are using?

DanielO, 2019-12-28:
So, I think the 6 pin modules have a TP4056 (to charge the battery) and a DW01a & MOSFET (to protect against over current / shorts).
I confused myself earlier and thought they must have a DC-DC converter.

Perhaps the documentation could be amended to add a photo of a known working unit and a description highlighting the above parts.
I can do a pull request for the documentation but I don't [yet] have a photo of a known good unit :)

Relatedly - it would be very helpful to have a photo of the modification of the Wemos board for reference.

universam1, 2020-01-18:
happy to accept a PR that improves the documentation

closed 2020-01-18 by universam1

DanielO, 2020-02-23:
FYI I have been making another one with the right battery module this time and taking lots of photos so when I'm done I'll create some more documentation :)