universam1/iSpindel #329: I can't configure/ no Ipindel wifi signal

rackam74, 2019-12-28:
Hi,
I try to configure an ispindel and it's impossible to have the ispindel wifi signal. do you have any ide of the issue???

¦¦¦F¦*WM: Scan done
*WM: No networks found
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
sdd¦¦|¦l¦|
¦l¦c|!¦;¦c¦
c¦¦g'¦lgn¦¦¦cx¦¦${$slx¦o¦¦
d
¦¦
b
g¦|l¦
¦#¦¦gn¦l¦¦l ¦'l`gsU¦g#
¦ls¦¦g#
¦l¦¦ld¦$`¦n¦sdd¦¦|¦d¦<
¦
l¦
c|¦¦¦r¦c¦
b¦¦'o¦dog¦¦¦
cp¦lrl{dp¦g¦
¦l
¦¦
c
n¦|d¦
¦c¦¦oo¦dcd`¦no
d`n{¦¦¦o
c¦l;¦¦o
c
¦l¦¦ll¦d`¦¦g¦;ll¦¦|¦l¦|
¦l¦c|!¦r¦c¦
b¦¦'o¦lno¦¦¦
cp¦l;lsdp¦o¦
¦l
¦¦
c
'¦|l¦
¦b¦¦no¦l¦¦l ¦nl gs¦¦¦'
c¦d{?n
c¦$¦¦$l¦l ¦o¦sdd¦¦|¦l¦|
¦
l¦c|¦¦¦s¦c¦
b¦¦no¦dog¦¦¦
cpcl;l{dp¦g¦
¦l
¦¦
c
n¦|d¦
¦c¦¦og¦d¦¦d`¦n
l gs¦¦¦gb¦ds¦¦g#
¦l¦¦ld¦¦d`¦g¦;ll¦¦|¦l¦|
¦l¦c<¦¦¦;¦c¦
c¦¦gn¦l'o¦¦¦cx¦¦l{$sdx¦o¦
¦$
¦¦
c
'¦|l¦
¦c¦¦og¦d¦¦d`¦oo
d`o;¦¦¦o
c
¦lr¦¦o
c
¦l¦¦ll¦d`¦g¦sd$¦¦|¦d¦|
¦
d¦
#|¦¦¦s¦b¦c¦¦og¦dog¦¦¦
cpcl;l{dp¦o¦
¦$
¦¦
c
g¦|l¦
¦#¦¦gn¦l¦¦l`¦gd`o;¦¦¦o
c
¦lr¦¦o
c
¦l¦¦ld¦d`¦¦g¦{$l?|¦$¦|
¦l¦c<¦¦¦{¦c¦
c¦¦gg¦lgg¦¦¦
#p¦¦dsl{lp¦g¦
¦
lG
co¦|d¦
¦c¦¦og¦$¦¦d`¦g$`orU¦g#
¦ls¦¦g#
¦l¦¦dd¦$`¦n¦
            FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS...{ll?|¦$¦|
¦d¦
#|¦¦¦s¦#¦c¦¦no¦doo¦¦¦
cp¦l;lsd8¦o¦
¦l
¦¦
c
'¦|l¦
¦b¦¦'o¦l¦¦l ¦nl gs¦?'
c¦d{¦¦n
c¦$¦¦$l¦l ¦g¦
             FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS...{ll?|¦$¦|
¦d¦
#|¦¦¦s¦c¦c¦¦no¦doo¦¦¦cp¦l;lsdp¦o¦
¦l
¦¦
c
n¦|d¦
¦c¦¦oo¦d¦¦d`¦
l`'{¦¦¦o
c
¦lr¦¦o
#
¦l¦¦ld¦$`¦'¦
            FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS...rll¦¦|¦l¦<
¦
d¦
#|¦¦¦s¦b¦c¦¦og¦$g'¦¦¦
bp¦dsd;lx¦'¦¦
d
¦¦
b
g¦|l¦
¦b¦¦'n¦l¦¦l`¦g$`o;¦¦¦o
c
¦ls¦¦g
#
¦l¦¦ld¦d`¦g¦
            FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mountin¦rll¦¦<¦l¦|
¦l¦c<¦¦¦{¦c¦c¦¦og¦dog¦¦¦
cpcl;lsd8¦o¦¦
d
¦¦
#g¦<l¦¦c¦¦gg¦l¦¦$`¦oo
d`o{¦¦¦o
c
¦ls¦¦gb¦d¦¦d$¦l`¦o¦
                   FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS...sdd¦¦|¦l¦<
¦l¦c|¦¦¦{¦c¦
c¦¦og¦$g'¦¦¦
bx¦dsd;lx¦'¦
¦
l¦¦
co¦<$¦¦c¦¦g'¦l¦¦l`¦'l`gs¦¦¦g
c¦d{?'
c¦d¦¦dl¦¦l ¦g¦
              FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
ERROR: failed to load json config

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 AD 7F 79 A2 0 3 6A
  Chip = DS18B20
  Data = 1 5D 1 55 5 7F A5 A5 66 13  CRC=13
  Temperature = 21.81 Celsius,
offsets not available
Boot-Mode: External System
power-cycle or reset detected, config mode

Double Reset detected
...........
           ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
{$l?|¦$¦|
¦l¦c|!¦r¦c¦
b¦¦'n¦lno¦¦¦c8¦¦l;lsdp¦g¦
¦
lG
co¦|$¦
¦c¦¦g'¦l¦¦$`¦g$`o;¦¦¦o
c
¦lr¦¦o
c
¦l¦¦ll¦d`¦'¦
            FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
ERROR: failed to load json config

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 AD 7F 79 A2 0 3 6A
  Chip = DS18B20
  Data = 1 5E 1 55 5 7F A5 A5 66 D6  CRC=D6
  Temperature = 21.87 Celsius,
offsets not available
Boot-Mode: Deep-Sleep Wake
...........
           ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

sbonaime, 2020-01-02:
You can try what I did in #324 :
firmware 5.8.6
initializing the memory as described in the FAQ
=> 6.2.0

rackam74, 2020-01-02:
thank I'll try it!

closed 2020-06-23 by stale[bot]