universam1/iSpindel #331: CraftBeerPi ServiceType don't work if we using HTTAuth plugin (Http Basic Auth)

victorsoliveira, 2020-01-09:
Hi @universam1 !

I'm using HttpAuth plugin to add Http Basic Auth security to my CBPI, so, i tried to configure iSpindel with to work with my CBPI and i could not do it. Exists any workaround? 

Thanks!

victorsoliveira, 2020-01-14:
Any idea @universam1 ?

Thanks!

duttonw, 2020-02-05:
I'd think you need to allow username/password to be passed to the sender
https://github.com/universam1/iSpindel/blob/master/pio/lib/Sender/Sender.cpp#L194

Most iot don't deal with security, but instead use [security through obscurity](https://en.wikipedia.org/wiki/Security_through_obscurity), i.e a really long url or apikey in the header.

AWS command line uses hmac signing of payloads where there is a pre-shared key which is never passed over the wire and even then aws suggests to rotated these keys every 90 days just to ensure no one cracks it or stop a leaked key causing too much damage. 

I do think it would be nice to allow improved security to be ready on the clients so that when the server side gets updated its ready, but saying that you kinda also want to be using https and i'm unsure what power draw that would have on the battery when compared to plain communications.

closed 2020-06-23 by stale[bot]