universam1/iSpindel #333: How to use Service Type: HTTP

ps915, 2020-01-10:
I can't find any documentation about the HTTP Service.

Is it possible to use it with my own HTTP Webserver?

What if i set up a .php script to log the data in a MySQL Database. 

http://beerserver.de/log.php?mode=iSindel_No_1&battery=4.15&gravity=39.9&interval=1&rssi=87.00&temp=22.5&tilt=82.5

Is this possible?

mp-se, 2020-01-11:
Yes that is possible. The data will be sent as a http post and json format. This is an example of what you should expect in the request: 

{"name":"iSpindel002","ID":11795794,"token":"fermentrack","angle":46.20663,"temperature":17.625,"temp_units":"C","battery":4.160584,"gravity":11.02123,"interval":300,"RSSI":-85}

ps915, 2020-01-11:
Thanks a lot, for the information, that's usefull! Is it somehow possible (or planned for the future) to support GET additional to the POST method?

ps915, 2020-01-14:
How do i configure the Service correctly?

What does "token" mean?

[image]

victorsoliveira, 2020-01-14:
@ps915 It's correctly! Token info is needed if your URI is authenticated token based, so, when you provide a token info when iSpindel make a POST at http://www.server.de/log.php an authorization header will be provided for request with this token information.

ps915, 2020-01-14:
Ok, understand but unfortuantly i cant get it to work. 
I dont want to use the token at all so i leave it blank.

I configured the iSpindel in these two ways.

If i set up the Server Adresse like:
[image]
the request URL looks good.
[image]

But if i configure it like this:
[image]
The request does not go to the .php script, just to the main server.
[image]

I wrote this little PHP Script to test the Output of the iSPindel, but none of the both configurations create a POST Data.

When i test ist with my POST Form via HTML, everything works but just not with the iSpindel. Any idea what could be wrong?

[image]

`<?php

//check for Post data
if (!empty($_POST["data"])) {
	
	echo "Data received";
	file_put_contents("post.log", print_r($_POST, true), FILE_APPEND);

} 

//Show log
$log = file_get_contents('post.log');
echo "<pre>".$log."</pre>";

?>

<form action="http://server.de/logtest.php" method="post">
	<input type="text" name="data">
	<input type="text" name="data2">
	<input type="submit" value="SEND">
</form>`

victorsoliveira, 2020-01-14:
@ps915 Try append a slash at "Path / URI" info... Use "/logtest.php" instead of "logtest.php"

ps915, 2020-01-14:
i tried
Server Adress: **http://server.de/**
Path: **/logtest.php**

The iSpindel still just  makes a request to **http://server.de/** instead of **http://server.de/logtest.php**

victorsoliveira, 2020-01-14:
When i configure it out, works fine with samples info below
**Server Address:** 192.168.0.105
**Path:** /ispindel

try to remove slashes and protocol from **Server Address** info

ps915, 2020-01-14:
no sucess so far!  I tried to use it in my Local Area Network but same issue here.
I also tested it without the script path, so i created a index.php on my webserver.

tested this Server Adresses:

- 192.168.0.30
- http://192.168.0.30/
- http://192.168.0.30/index.php

victorsoliveira, 2020-01-14:
@ps915 Yes, the POST data was transmitted.

Looking better now , you have a condition expecting POST BODY at "$_POST['data']", try without that condition, the JSON info from POST request come in at BODY.

Sou, try it out.

`<?php
//check for Post data

echo "Data received";

 $data = json_decode(file_get_contents('php://input'), true);
var_dump($data)

file_put_contents("post.log", print_r($data, true), FILE_APPEND);

//Show log
$log = file_get_contents('post.log');
echo "<pre>".$log."</pre>";

?>

ps915, 2020-01-14:
dammit, totally my bad! I Fcked the $_POST['data'] up! I apologize and thanks you very much for your help! =) i owe you a beer!

closed 2020-01-14 by ps915

victorsoliveira, 2020-01-14:
Relax! @ps915.
Good beers for u!