universam1/iSpindel #334: Corrupt Config File - Offset Calibration

OpenSourceDistilling, 2020-01-16:
If you try and do an Offset Calibration before saving the configuration page the offset calibration doesn't save after you power off and back on again the iSpindel. Everything appears fine in the user interface when doing the offset calibration, but in the Arduino serial monitor, it throws errors about corrupt JSON and config files. Once you go to iSpindel > Configuration > Save I believe the config file is created and then the offset calibration information can then be saved and retrieved correctly. I produced this issue on firmware 6.2.0. Are we able to create the required file(s) prior to saving? Sorry, I didn't save the log files but I'll be able to reproduce the log files in the next week or two when I do my next iSpindel build.

pppedrillo, 2020-01-18:
I've just checked with 6.2 fw and completely wiped D1 (so no prev configs available on SPIFFS) and it looks like everything ok.
After calibration config gets saved along with read offsets.
Here is couple of logs (2 consequent runs) from terminal:

1.-------------------------------------------------------------------------------------------
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Maintenance page
*WM: Sent iSpindel info page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: offset
Testing device connections...
MPU6050 connection successful
>**......>......
at 600 Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS    -3098,     698,    2846,     199,     -52,     -18  

new offsets: -3098:698:2846       
confirming offsets: -3098:698:2846
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"","PSK":"","Offset":[-3098,698,2846,199,-52,-18]}
saved successfully
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: iSpindel
*WM: Sent iSpindel info page

2. -----------------------------------------------------------------------------
Testing device connections...
MPU6050 connection successful
>**......>......
at 600 Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS    -3796,     730,    1938,     199,     -52,     -17

new offsets: -3796:730:1938
confirming offsets: -3796:730:1938
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"","PSK":"","Offset":[-3796,730,1938,199,-52,-17]}
saved successfully
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: iSpindel
*WM: Sent iSpindel info page

OpenSourceDistilling, 2020-01-18:
@pppedrillo Thanks for that. I will try and reproduce the issue and post the logs. Hopefully, it was some sort of one-off thing that will never happen again.

pppedrillo, 2020-01-19:
@OpenSourceDistilling pls note that to be at the same point like you were first time start (eg without any stored config on the device) you'll need to completely erase whole D1 memory before flashing the firmware. Only fw flashing will not reset SPIFFS memory and you'll still "old" config there.

Cheers and have a nice beer! :)

OpenSourceDistilling, 2020-04-23:
This is an old issue. Sorry I haven't tested and reported back. I will see if I can reproduce the error and update this. I will be building more iSpindels in the near future.

closed 2020-06-29 by stale[bot]