universam1/iSpindel #337: Voltage Display factor

ldab, 2020-02-03:
Hi,

Is there a reason for the Voltage Factor be limited to 20%? 

https://github.com/universam1/iSpindel/blob/069b12274ee2e2a9b0724e4e3e15ca67eeec59aa/pio/src/iSpindel.cpp#L422

I currently designed on the [wemos D1 Pro](https://docs.wemos.cc/en/latest/d1/d1_mini_pro.html) and it has built-in resistors connected to the battery which makes the iSpindel requiring very minimal soldering.

https://wiki.wemos.cc/_media/products:d1:sch_d1_mini_pro_v2.0.0.pdf

The Voltage Factor is `5.109489 / 4.2 * 191.8 = 233.3333` which is **21.6 %** higher 😛 

[image]

beikeland, 2020-02-03:
You have 5.1v instead of 4.2v? Check your wiring, the d1 mini has a diode
removed and uses the lipo charger's output (4.2v) for 5v.

So my guess is that it's a sanity check that works as intended?

ldab, 2020-02-03:
> 
> You have 5.1v instead of 4.2v? Check your wiring, the d1 mini has a diode removed and uses the lipo charger's output (4.2v) for 5v. So my guess is that it's a sanity check that works as intended?
> […](#)

Hi, Thanks for the reply, but you probably didn't even bother to open the links I sent, if you did, you would notice that this board has a built-in charger and the Diode does not affect the battery path.

vytux-com, 2020-02-03:
I think this was asked before and the simple solution is to first calibrate it by faking the figures so it's just below 20% then after saving do a second real calibration

beikeland, 2020-02-03:
I most sincerely doubt any level of patronizing will fix the wemos ssl cert and allow me to open the link, picture is not worth a thousand words in this case.

In any case it doesn't really change the response, you asked why the code is limited to 20%; the answer remains because it's intended to work with max 4.2v.

I doubt having an extra 5v boost converter followed by a 3.3v is going to be beneficial for your runtime so maybe look into what the others using the pro is doing?

ldab, 2020-02-03:
@beikeland who talked about boost converter?

Here you have another schematic with valid SSL https://docs.wemos.cc/en/latest/_static/files/sch_d1_mini_pro_v2.0.0.pdf

As you will notice, this board has built-in resistor connected to the battery as well 130K which is different from the circuit diagram presented on iSpindel documentation. It's easy enough for me to change it in the code, the question was intended to help others in the future using the same board.

beikeland, 2020-02-03:
5.1v without schematic sounds like boost converter.

But why not just calibrate the adc when running on battery if it won't see more than max 4.2v  during use, should bring down the difference a bit?

ldab, 2020-02-03:
@beikeland thanks for quick reply, but please take your time to have a look at the links provided and the schematic of the official Wemos website. As explained the 5.1v is due the different resistor, which is built-in.

universam1, 2020-02-03:
The FAQ contain as section how to calculate the conversion factor, that should suffice

universam1, 2020-02-03:
In your case with a differencing board, why not adjust the resistor acting as voltage divider accordingly?

ldab, 2020-02-03:
@universam1 because the resistor is SMD built-in and it's a lot easier to adjust the voltage factor 1.3% more as apparently there's no reason for limiting it to 20%...

This is the official `wemos pro` board which has built-in battery charger and other whistles which makes neat to iSpindel in my humble opinion.

universam1, 2020-02-03:
Okay makes sense, the 20% is a convenience limit. Open for a PR

ldab, 2020-02-03:
Ok, will do after the Blynk one ;)

universam1, 2020-02-03:
There are open concerns with

ldab, 2020-02-06:
https://github.com/universam1/iSpindel/pull/341

closed 2020-02-06 by ldab