universam1/iSpindel #339: Will not boot to config when battery powered

Naesstrom, 2020-02-05:
I'm having a weird problem with one of my builds, when powered through the usb it works perfectly, it starts in config mode as it should. but when I power it through the battery it seems to crash when booting into it:
## USB BOOT
```
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"","PSK":"","Offset":[-1034,-5738,720,-211,-68,12]}
applying offsets: -1034:-5738:720
confirming offsets: -1034:-5738:720
Boot-Mode: Deep-Sleep Wake
...........
ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
```
## BATTERY BOOT
```
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"","PSK":"","Offset":[-1034,-5738,720,-211,-68,12]}
applying offsets: -1034:-5738:720
confirming offsets: -1034:-5738:720
Boot-Mode: External System
power-cycle or reset detected, config mode
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL
sd lܟ< ⸮l⸮<⸮l⸮c|⸮⸮⸮⸮r⸮#l⸮c⸮⸮gn⸮dog⸮⸮⸮c8ćlrd;l⸮;⸮⸮g⸮b⸮gb8 ⸮
```
It still seems to be on, the gyro LED is lit and if I do a reset of the wemos it starts in config mode again as it should.

ldab, 2020-02-06:
What battery and what board are you using?

Naesstrom, 2020-02-07:
Ah yeah... checked everything again and had turned the bat-43 the wrong way... my bad!

closed 2020-02-07 by Naesstrom