universam1/iSpindel #340: ispindel tilting issue

ftassini, 2020-02-06:
A recap of what I did with my ispindel:
1. soldered, assembled and connected: also verified the gyroscope was working as well as the temperature sensor;
2. put weights on until had a tilt of 24;
3. offset calibration as per ispindel menu "maintenance" where it says to do the offset before definitive calibration and result was a tilt of 89.80;
4. placed the ispindel back in plain water and the tilt was 60 whereas in the same position I had 24... With the apparatus in upright position it gives a tilt of 62...
Is that normal?
Do I have to start the calibration with sugar even if my starting tilt is 60?
Thanks to anyone that could help

ErikdBr, 2020-02-06:
You should have done point 3 before point 2, so first the offset calibration and after that put weights on until... in plain water.

ftassini, 2020-02-06:
Thanks Erik, so what should I do now?
Maybe reset the ispindel or re-flash the firmware?
What would you suggest?

ErikdBr, 2020-02-07:
Initializing the iSpindel(see the FAQ at What should I do if my iSpindel isn't working properly after flashing it with the firmware?) and flash the software again, is allways a good idea if you have any problems I would do the calibration again, first the offset calibration and after that add or remove weights to get a tilt of around 20.

ftassini, 2020-02-12:
Hi Erik,
I've reflashed the ispindel and I've done the calibration offset BEFORE putting the weights on, but I still have a minimum tilt of circa 60...
Any idea?

ErikdBr, 2020-02-12:
What was the angle in plain water after putting the weights on? And do you have some picture of your ispindel?

ftassini, 2020-02-12:
With angle you mean approx. degrees?
Wouldn't know exactly, I'll take a picture and upload it asap

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ftassini, 2020-02-12:
Position and tilt result with bolt and 2 nuts on
[image]
[image]

ErikdBr, 2020-02-12:
I assume and think I see it in the last picture the battery side is downwards facing the bottom of the pan. Did you do the offset calibration with the same side, the battery side, facing down to the flat and level surface? If you did I am lost, if you didn't then that could be the problem.

ftassini, 2020-02-12:
I actually did the calibration in the way you said...

ErikdBr, 2020-02-12:
@universam1 I am completely lost here. Do you have any idea?

ecsfang, 2020-02-13:
Hi,

For me it looks as if the accelerometer is mounted upside-down. Looking at the picture, it looks as if the components on the MPU-6050 are towards the pbc, and the Vcc pin is closest to the bottom.
But looking at the pictures on https://3d-mechatronics.de/en/ispindel-teilmontiert/-partially-assembled_962.html, it looks at if the components should be facing outwards, and Vcc closest to the lid.
Just my two cents - I don't have this board myself, so I'm just relying on the pictures ...
[image]

ftassini, 2020-02-13:
@ecsfang You're right, the picture you refer to is clear, but I soldered the MPU according with the "DIY iSpindel manual" which shows the opposite, on page 9, pictures 17 and 18....

ecsfang, 2020-02-13:
@ftassini Agree, looking at https://dl.dropbox.com/s/0a2jkkvf0ba9g0c/DIY_Spindel_Anleitung_PCB1_EN_compressed.pdf it shows just as you have mounted the module, so I guess that that is the correct way!
I was just fooled by the picture above ... ;)

ftassini, 2020-02-14:
@universam1 could you give me your opinion on this issue please?

ErikdBr, 2020-02-15:
Only thing I can add is that in the older design the accelerometer is facing the other way around(with the chips up) as can be seen in the pictures. Also I can see a emailadres from 3d-mechatronics where you can ask questions, maybe you can ask which side should face up?
[image]
[image]

majberger, 2020-02-15:
> 
> 
> @ftassini Agree, looking at https://dl.dropbox.com/s/0a2jkkvf0ba9g0c/DIY_Spindel_Anleitung_PCB1_EN_compressed.pdf it shows just as you have mounted the module, so I guess that that is the correct way!
> I was just fooled by the picture above ... ;)

I've just finished soldering two brand new iSpindel. Following the copper on the PCB you can see that VCC and GND is correct the way it's documented and the way ftassini have built it.

ErikdBr, 2020-02-15:
> I've just finished soldering two brand new iSpindel. Following the copper on the PCB you can see that VCC and GND is correct the way it's documented and the way ftassini have built it.

And yours are working correctly? Cause if they do maybe you can help.

ftassini, 2020-02-15:
@ErikdBr as I wrote @ecsfang I just followed instructions given by DIY_Spindel_Anleitung_PCB1_EN_compressed.pdf ....
I've already tried asking at 3d mechatronics but the guy Tjorben Eberle told me he's only responsible for hardware and for what he sees it is working; he sent me over here to open an issue...

majberger, 2020-02-15:
> And yours are working correctly? Cause if they do maybe you can help.

Will report back when I've had time to try them out. Haven't even flashed them yet, way too many other things that need my attention right now. Just wanted to report that the manual seems to be correct when "reading" the PCB.

Fatfil, 2020-05-24:
I have a different kit, from Colin on ebay, but have exactly the same issue. Did you fix this ghost??

Fatfil, 2020-05-27:
Fixed it. Make sure the gyro is facing upwards when running the offset calibration routine even if when you float it, it faces down. If you do this the problem goes away. I would recommend you play with the bouncy so that the iSpindel floats with the gyro pointing upwards.

whitetd, 2020-06-11:
had to fix mine, flipped the gyro board over, extended the wires, I will add a pic when I get a chance.

majberger, 2020-06-17:
Finally sold the house and have finished moving, sorry for not getting back to you guys earlier.

TL;DR - I have the exact same model that ftassini (account deleted, hence the name "ghost") was using. Can reproduce the error and confirm that Fatfil's solution (make sure the gyro face "upwards" during calibration) works. You can close this issue.

I was a bit at a loss first.
Had to get one of my iSpindels ready in a hurry, since I had promised my father to brew him a Czech Premium Lager for summer. It worked really well! The second one I popped into a fermenter last sunday, and it had the exact same issue as ftassini's had. Really weird!

Bottled my fathers Lager a few days ago and now I have tested with the one that worked. I simply had forgotten to use Maintenance->calibrate. As soon as I'd done that it too had the same issue. Using the serial debug output I could see that it calibrated to "new offsets: -966:-1020:5224"

Suspected that the calibration optimization was to blame, flashed it with older firmware with the old "this takes several minutes"-calibration. Same odd result.

Set up so that I could dig into the code, compile my own firmware and really start debugging. Then, on a hunch, I checked the comments and found Fatfil's solution. Many thanks for posting that, it probably saved me lots of time! So I reset the Wemos D1 and flashed the latest firmware, made sure that it was "upside down" so that the gyro faced upwards and calibrated once more: "new offsets: -988:-1156:1628". 
Hooray! Now vertical is ~90 degrees and horizontal is ~0 degrees.

If you want to implement a software solution when calibrating (if weird readings, assume it's an upside-down-kit and calibrate correctly) I guess it would be nice. But I opt for just mentioning the requirement of having the gyro facing upwards in the calibration documentation. Better spend the time coding other things. :-)

universam1, 2020-06-22:
Thank you @majberger  for that detailed feedback!
Would you like to add a PR with your findings to the FAQ, might be helpful for other in similar situations

universam1, 2020-12-20:
Try using the latest firmware version and report if that doesn’t improve the situation

closed 2020-12-20 by universam1