universam1/iSpindel #342: Calibration issue

arjepsen, 2020-02-07:
Hey.
I just put together an ispindel, using the ispindelpcb.
I'm having trouble getting the offset calibration done.
I can connect to the ispindel, and start the calibration - the blue diode lights up.... and stays on.
I tried waiting like 30 minutes, but it's still on solid. 
Any idea what I might have done wrong?

ErikdBr, 2020-02-07:
Try the procedure "What should I do if my iSpindel isn't working properly after flashing it with the firmware?" in the FAQ: https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md

ErikdBr, 2020-02-17:
> Hey.
> I just put together an ispindel, using the ispindelpcb.
> I'm having trouble getting the offset calibration done.
> I can connect to the ispindel, and start the calibration - the blue diode lights [...]

Did it work?

closed 2020-06-23 by stale[bot]