universam1/iSpindel #343: iSpindel stops after 1 calling TCP

mcmellow, 2020-02-08:
When switched on iSpindel sends out 1 message and stops. No activity at all.  Did a wipe-out of the memory according this example:
`		1st column		2nd column
1st Line:	INTERNAL://NODEMCU	0x00000
2nd Line:	INTERNAL://BLANK 	0x01000`
And a ref;ash of the firmware 6.20.So far no positive result.  See serial monitor output
`14:42:45.328 -> FW 6.2.0
14:42:45.328 -> 2.2.1(cfd48f3)
14:42:45.328 -> Worker run!
14:42:45.328 -> mounting FS... mounted!
14:42:45.328 -> reading config file
14:42:45.328 -> parsed config:
14:42:45.328 -> {"Name":"iSpindel-Groen","Token":"","Sleep":10,"Server":"192.168.xxx.yyy","API":6,"Port":9501,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"MySSID","PSK":"MyPSK","Offset":[2608,1860,1188,45,-3,2]}
14:42:45.551 -> applying offsets: 2608:1860:1188
14:42:45.551 -> confirming offsets: 2608:1860:1188
14:42:45.551 -> Boot-Mode: Deep-Sleep Wake
14:42:45.551 -> 
14:42:45.551 -> woken from deepsleep, normal mode
14:42:46.283 -> Samples:42 min:70.43 max:70.79 time:749
14:42:46.329 -> x: -14294 y: 6124 z: 26
14:42:46.329 -> Tilt: 70.58
14:42:46.329 -> Tacc: 23.06
14:42:46.329 -> Volt: 4.60
14:42:46.329 -> Temp: 22.44
14:42:46.329 -> Gravity: 23.71
14:42:46.329 -> IP: 192.168.xxx.yyy
14:42:46.329 -> 
14:42:46.329 -> calling TCP
14:42:46.329 -> {"name":"iSpindel-Groen","ID":6880147,"angle":70.57919,"temperature":22.4375,"temp_units":"C","battery":4.603754,"gravity":23.71437,"interval":10,"RSSI":-66}
14:42:47.275 -> Sender: TCP stream
14:42:47.275 -> 6
14:42:47.416 -> 
14:42:47.416 -> Final-sleep: 10s; RT: 2152`

What can I do or is there something wrong? 
[image]

mp-se, 2020-02-08:
Check the r1 resistor, to high value can stop the wakeup procedure.

mcmellow, 2020-02-08:
As the printing says on the PCB.  4k7 ohm

mp-se, 2020-02-08:
It should be 470ohm between d0 and rst. 4.7 will stop it from waking up after sleep

mcmellow, 2020-02-08:
Thanks, I found also issue #59 So I am checking out now.

closed 2020-02-19 by mcmellow