universam1/iSpindel #344: No Derived Variable Option in Ubidots?

Hairyjeremy0, 2020-02-09:
Hi,
Sorry to ask for help, but I have been struggling for many days now and after reading and rereading the available instructions, FAQs, past issue logs, etc, I am still unable to complete configuration of my device. Here's what I have done so far:
1. Completed initial calibration on level surface = 89 degrees
2. Checked firmware for updates = 9.2.0 installed already
3. Added calibration weights to achieve a 20 degree tilt angle in pure water (comparing my iSpindel SG value at this point with a pre-calibrated hydrometer shows a variation of 0.095)
4. Went through the calibration procedure to derive a range of tilt angles and SG values and entered these on the Excel sheet.
5. Went to add derived variable, but no such option available?
6. Also, once I have a new calibration formula, how do I enter this into the iSpindel? Do I update the Polynomial field?

Sorry for my questions, but I need assistance please.
[image]

ErikdBr, 2020-02-09:
5. You'll have to sign up at ubidots for education(https://app.ubidots.com/), there you will have that option
[image]
6. Yes. And you can then enter a derived variable in ubidots:
[image]

Hairyjeremy0, 2020-02-09:
Erik, 
Thank-you so much for your quick reply. Unfortunately it seems I am unable to login or sign up for that version of Ubidots; their site keeps redirecting me to industrial.ubidots.com, (now Ubidots STEM). This does not give an option for calculating a derived variable. Do you know of any other ways which I might perform this function please?
Jeremy

ErikdBr, 2020-02-09:
I use an android app brewspy, it is paid but you can only use it on a phone, not on a website.
https://play.google.com/store/apps/details?id=com.yiannnos.myspindel

Hairyjeremy0, 2020-02-09:
Thanks Erik. 
I also wish to use my device with another brewing app, but does BrewSpy provide the same functionality as Ubidots; calculating the complete calibration formula? I have downloaded the BrewSpy app but cannot see this on the free version of the app? Happy to buy the app but only if it would help me calibrate my device

ErikdBr, 2020-02-09:
Yes. I have in the polynomal of the iSpindel just the word "tilt", in the configuration of brewspy "default" in calibration units and the formula from the Excel sheet also in brewspy.
[image]

Hairyjeremy0, 2020-02-09:
Thanks Erik, I'll take a look.

Hairyjeremy0, 2020-02-10:
OK, for the benefit of anyone else struggling with this same problem (probably just me!), the ubidots configuration instructions need updating. There is no 'derived variable' option now, you have to select 'synthetic variable'. Then you copy and past the formula as instructed. However, when editing the formula, in addition you must delete the spaces and replace the exponentiation symbol with ** because ubidots does not recognise that symbol.

closed 2020-02-12 by Hairyjeremy0