universam1/iSpindel #345: Acc Test Connection ERROR

Nckls78, 2020-02-10:
I have soldered an Ispindel with version 4.0 of the board and with all components from this page: https://www.pcbway.com/project/shareproject/iSpndel_4_0.html

When I connect to it and go to the Info page I get 0.0 in Tilt:
[image]

Checked the log through PuTTY:

```
ÛcbC½Â¡1Áírd $ÜŸ< Ädà|Œlä#|‡ƒì“{“cŒc„ò'gžl'oœãäcpäÇds$s$8ûgàƒd„œcoã<lä‡bŒûg'ç dŒŽl o'd`g{“›gcœd;›Û'#œ$œÄ$lä„l üƒoÜ
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindelX01","Token":"","Sleep":900,"Server":"/ispindel?id=n3VjiZTxdAL1bc","API":3,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"Fluffnet","PSK":"MiniNora7147","Offset":[0,0,0,0,0,0]}
Acc Test Connection ERROR!
offsets not available
Boot-Mode: External System
power-cycle or reset detected, config mode

going to Config Mode
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: Scan done
*WM: DUP AP: Fluffnet
*WM: DUP AP: Fluffnet
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
```

Ok, it can't connect to MPU6050. Found some issues regarding this: https://github.com/universam1/iSpindel/issues/302

It says the sensor reports wrong address. I set up a Arduino board and connected the sensor and ran the code from this page: https://github.com/ElectronicCats/mpu6050/blob/master/examples/IMU_Zero/IMU_Zero.ino
It uses the same library MPU6050 ant the same method testConnection().

Got the log:
```
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
PID tuning Each Dot = 100 readings
>......>......
at 600 Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS      120,     286,    1664,      31,      12,      10

>.>.700 Total Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS      118,     286,    1662,      23,      10,      10

>.>.800 Total Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS      120,     284,    1662,      30,      11,      12

>.>.900 Total Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS      120,     284,    1662,      30,      11,      11

>.>.1000 Total Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS      118,     284,    1662,      31,      12,      11

 Any of the above offsets will work nice 

 Lets proof the PID tuning using another method:
averaging 1000 readings each time
expanding:
....	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
 [0,0] --> [-979,-968]	[0,0] --> [-2447,-2500]	[0,0] --> [1626,1624]	[0,0] --> [-129,-122]	[0,0] --> [-46,-45]	[0,0] --> [-41,-42]
.... [0,1000] --> [-970,7283]	[0,1000] --> [-2499,6001]	[0,1000] --> [1626,10502]	[0,1000] --> [-122,3875]	[0,1000] --> [-45,3951]	[0,1000] --> [-41,3953]
.... [0,1000] --> [-969,7280]	[0,1000] --> [-2501,5995]	[0,2000] --> [1627,19377]	[0,1000] --> [-120,3875]	[0,1000] --> [-41,3950]	[0,1000] --> [-38,3954]

closing in:
..	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
 [0,500] --> [-969,3153]	[0,500] --> [-2501,1749]	[1000,2000] --> [10501,19377]	[0,500] --> [-120,1878]	[0,500] --> [-41,1956]	[0,500] --> [-38,1960]
.. [0,250] --> [-969,1092]	[250,500] --> [-376,1749]	[1500,2000] --> [14944,19377]	[0,250] --> [-120,877]	[0,250] --> [-41,955]	[0,250] --> [-38,958]
.. [0,125] --> [-969,53]	[250,375] --> [-376,677]	[1500,1750] --> [14944,17162]	[0,125] --> [-120,377]	[0,125] --> [-41,455]	[0,125] --> [-38,458]
.. [62,125] --> [-461,53]	[250,312] --> [-376,146]	[1625,1750] --> [16042,17162]	[0,62] --> [-120,125]	[0,62] --> [-41,202]	[0,62] --> [-38,205]
.. [93,125] --> [-212,53]	[281,312] --> [-124,146]	[1625,1687] --> [16042,16600]	[31,62] --> [0,125]	[0,31] --> [-41,80]	[0,31] --> [-38,81]
..	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
 [109,125] --> [-80,53]	[281,296] --> [-124,13]	[1656,1687] --> [16326,16600]	[31,46] --> [0,61]	[0,15] --> [-41,16]	[0,15] --> [-38,18]
.. [117,125] --> [-16,53]	[288,296] --> [-58,13]	[1656,1671] --> [16326,16453]	[31,38] --> [0,29]	[7,15] --> [-15,16]	[7,15] --> [-13,18]
.. [117,121] --> [-16,16]	[292,296] --> [-20,13]	[1663,1671] --> [16380,16453]	[31,34] --> [0,12]	[11,15] --> [0,16]	[7,11] --> [-13,2]
.. [119,121] --> [-1,16]	[294,296] --> [-3,13]	[1663,1667] --> [16380,16415]	[31,32] --> [0,5]	[11,13] --> [0,8]	[9,11] --> [-6,2]
averaging 10000 readings each time
.................... [119,120] --> [-1,19]	[295,296] --> [-3,13]	[1663,1665] --> [16380,16396]	[31,32] --> [0,5]	[11,12] --> [0,2]	[10,11] --> [-1,2]
....................	XAccel			YAccel				ZAccel			XGyro			YGyro			ZGyro
 [119,119] --> [-1,1]	[295,296] --> [-4,13]	[1663,1664] --> [16380,16397]	[31,32] --> [0,5]	[11,12] --> [0,2]	[10,11] --> [-1,2]
.................... [119,119] --> [-6,1]	[295,295] --> [-4,8]	[1663,1664] --> [16379,16397]	[31,32] --> [0,5]	[11,12] --> [0,2]	[10,11] --> [-1,2]
-------------- done --------------
```

Works just fine.

Trying to check the code in latest release to see what I'm missing, but I really can't understand why it's still reporting a tilt of 0.0. I commented the code as in: https://github.com/universam1/iSpindel/issues/302#issuecomment-561919373, but that doesn't change the result. 

Any suggestions? Where should I look?

universam1, 2020-02-10:
The error message „ Acc Test Connection ERROR!” is unambiguous stating that you have a hardware issue on your board. You need to fix that before you can expect anything to work

Nckls78, 2020-02-11:
The error message is only called in the method testAccel(), so I assume it has something to do with the MPU6050 board? I have checked all the solder points, but I will make another iSpindel with identical components and see if the error appears on that board too.

jaisoh, 2020-02-29:
@Nckls78, Did you manage to resolve the issue?

I'm having the same problem and can I at least read the raw registers just fine and those also correlate to movement but indeed WHOAMI register reports 0x98 as it should report 0x68 back. Seems like the IMU modules I bought from local electronics shop is selling counterfeit goods bought from China.

closed 2020-06-23 by stale[bot]

NoWing31, 2020-08-24:
I'm experimenting the same issue, how did you solve it  ?

worgz01, 2020-09-27:
Same here with three of the boards I recently purchased. 

Is there any reliable supplier of the GY-521 boards, if this is the issue?

Assetsolution, 2020-12-14:
Would be cool to confirm what the issue is, are you guys saying that the its more than likely to be a fake 6050 as I am going through so many GY-521 boards some purchased from the UK and other from china.
I have attached am image of the board I'm using
[image]

ErikdBr, 2020-12-14:
> The error message „ Acc Test Connection ERROR!” is unambiguous stating that you have a hardware issue on your board. You need to fix that before you can expect anything to work

I had exactly the same issue last weekend with a built, I could connect to the configuration and saw the value of the tilt and the gravity on the info page read: 'nan' so a faulty gy-521 mpu-6050. About 50% of the batch of boards I bought are working fine the rest is garbage.

Assetsolution, 2020-12-15:
@ErikdBr Thanks for the info :)

MatthBou, 2020-12-24:
@ErikdBr I had exactly the same problem, 50% of the batch did not work, with "NAN" instead of the angle value.
Strangely enough, I directly tested these "faulty" GY-521 with a D1 mini and the ElectronicCats library and they all worked!
It's just to try to move the problem forward.

holzachr, 2021-04-13:
Guys, please stop posting your serial log; it contains your clear text WIFI password ;-)

vytux-com, 2021-04-13:
> Guys, please stop posting your serial log; it contains your clear text WIFI SSID and password ;-)

And unless I know where you live it's completely useless to anyone.

peterschrott, 2021-06-04:
I also get the NAN for the gyro in the WebUi. Same with `Acc Test Connection ERROR!`.

Nevertheless when I press the rest on the D1 mini there is an gyro value printed:

```
woken from deepsleep, normal mode
Samples:42 min:73.07 max:122.78 time:752
x: 14888 y: 7932 z: 128
Tilt: 89.66
Tacc: 41.44
Volt: 4.30
Temp: 25.63
Gravity: 33.39
IP: 192.168.0.181
```

So, not quite sure if it's broken hw.