universam1/iSpindel #348: How to calibrate device SG? ('Easy method' not working)

Hairyjeremy0, 2020-02-12:
Hi,
Can anyone help please? I have completed the off-set calibration, added tare weights to the device to tilt at 20deg in pure water and performed the 'easy calibration' method (three times now). For the incline experiment I have used my tried and trusted analogue hydrometer to provide accurate reference SG readings for the various tilt angles. However, after using the Excel formula computation and entering it into Ubidots, the Plato readings I get are significantly different to my analogue hydrometer (e.g. by up to 0.010SG), particularly acute around critical densities. Also, this means that the raw SG feed from the iSpindel is completely irrelevant, so it seems pointless. What am I doing wrong? Should I be calibrating the iSpindel SG using weights to get to the correct density reading? Or should I not expect the iSpindel to match an analogue hydrometer at all densities?

thegreatgunbantoad, 2020-02-12:
If it's the excel doc I'm thinking of, I found I needed to change the excel to use a 3rd order polynomial.  The cal that you get goes into the iSpindel config though, not Ubidots.

Hairyjeremy0, 2020-02-12:
Hi,
Thanks for your reply. I am using the excel spreadsheet provided in the iSpindel instructions. 
Yes, I did notice that the polynomial function already in the iSpindel i bought from you was quite different from the excel example. Not being an expert in such things though, would you be so kind as to tell me how to convert my formula in the way you describe please.

Many thanks,

Jeremy

thegreatgunbantoad, 2020-02-12:
I assume you ran through the quick start I put together: https://drive.google.com/open?id=19A07WmQ3VS27igYJXsUmOvKqMiIy_CmG.

There was a thing on facebook that I posted they will adjust from your starting angle, so far as I know it only works for the lastest cherryphilip board but it might work in general.

Do the senor cal on a flat surface.

Using this: https://drive.google.com/open?id=1DrNSo2Nlz9jMdAEGj9jOmz3cJEc5P8-H
Download the actual file though, google sheets seems to mess it up.

Chuck your unit in water roughly 20deg C ish and note the tilt (angle), also take a reading with a hydrometer.
Pop those results in the blue cells, and push go.
Copy all the text in the orange bit on the graph into the purple cell.
The calculation to use in the iSpindel setup is the text in the green cell.

If that isn't accurate enough there is the long way around

thegreatgunbantoad, 2020-02-12:
At some point those links may well die as they are to my own google drive.  I need to find a better place for them

Hairyjeremy0, 2020-02-13:
Many thanks @thegreatgunbantoad ! That is a _much_ better tool; simpler and more accurate that the original. However, I'm still not quite comfortable with the level of accuracy I'm getting; as towards the higher end of things my iSpindel is still around 0.006 SG low, compared with the analogue hydrometer. Presumably, the 'long way around' is to physically measure the tilt and SG, increasing 2 SG at a time, and manually input the observed readings into the table on your graph?

Hairyjeremy0, 2020-02-15:
Sorry to chase, but are you able to please advise me regarding the long calibration method, to help me improve the accuracy of calibration. You also made reference to a Facebook page, but it seems that Ebay have removed the link from your listing, so I haven't been able to access any information for myself, if it is publicly available? Thanks

thegreatgunbantoad, 2020-02-15:
The method is here: https://github.com/universam1/iSpindel/blob/master/docs/Calibration_en.md
The facebook group is this one: https://www.facebook.com/groups/BrewBeer/

You'll want to do the whole thing at roughly the temp that you'll brew at. Temperature will affect any Gravity reading.

Do bear in mind that that any of these sort of floaty things will never be 100% accurate. Mine usually comes in within about 3 gravity points. That said the accuracy is not bad and the iSpindel is probably as accurate as a Tilt.

How far have you got with it up to now?

Hairyjeremy0, 2020-02-16:
Thanks for your reply. 

The 'easy' calibration method is what I was following (I've done it three times, using salt and water, through salination then dilution), but using the scatter plot Excel spreadsheet in Github. However, that formula was producing less accurate results than your third order polynomial. My maths skills are not up to converting it into a third order polynomial and I tried entering the results into your Excel spreadsheet, but that just corrupted the chart (probably obvious but I thought it worth a shot).

In terms of current accuracy against my analogue hydrometer, my iSpindel seems to be about 6 points high at the lower end (near 1.000SG) and 3 points low at the high end (1.080SG). Reflecting upon your comment about comparative accuracy of tilting hydrometers and the data curve in your spreadsheet, perhaps it is now as accurate as I will get it. What do you think? How do you use yours; watch the data curve and then take a manual reading at the end of fermentation?

Thanks again and sorry for more questions. This is a learning curve for me.

Regards,

Jeremy

ErikdBr, 2020-02-16:
I use it just to follow fermentation so I can see when it is done and not so much to get the exact SG. So I know the tilt in water, I measure the OG after boiling with a hydrometer, I drop the iSpindel in the fermenting wort I then know the tilt in the wort and the relating SG. Then I insert these values in the spreadsheet and update the polynomial in the app that I am using. In the config of the iSpindel I have the word "tilt" in the polynomial field so it only sends the angle.  Sometimes when I measure the SG during fermentation I can then add the value in the spreadsheet and adjust the formula in the app instead of in the iSpindel to make it a bit more accurate.
I am happy with the fact that I don't have to guess when fermentation is done. The real measuring I do by hand.

Hairyjeremy0, 2020-02-16:
Thanks Erik

ErikdBr, 2020-02-16:
So to clarify this a bit more I show you a screenshot of a brew I did last year, in picture 1 are measurements done by the iSpindel and in picture 2 are the measurements I did by hand. At the end it is only 5 points off, accurate enough for me. And that with only 3 values in the Excel sheet, I measured once during fermentation. 
1.
[image]
2.
[image]

thegreatgunbantoad, 2020-02-16:
If you want to try a 3rd order poly generated from just your data points, try filling in the blue cells here, then copy from the whole of what's in the orange bit on the graph into the purple cell, your calc is then in the green cell.
[Manual generator.xlsx](https://github.com/universam1/iSpindel/files/4209827/Manual.generator.xlsx)

thegreatgunbantoad, 2020-02-16:
Just done a turbo Cider.  Real OG 1.077 FG 1.012.   iSpindel measured OG 1.079 FG1.012.  So not bad at all.

ErikdBr, 2020-02-16:
> If you want to try a 3rd order poly generated from just your data points, try filling in the blue cells here, then copy from the whole of what's in the orange bit on the graph into the purple cell, your calc is then in the green cell. I would be interested in what your data points are, it might help me make the quick generator more accurate.
> [Manual generator.xlsx](https://github.com/universam1/iSpindel/files/4209827/Manual.generator.xlsx)

I tried to find the excel with the iSpindel's current formula but couldn't find, sorry.

Hairyjeremy0, 2020-02-16:
Thanks both! It does seem like I may have the device as accurate as you do already, but in the interest of accuracy and sharing my learning with the community I shall run the experiment a fourth time and share the results with you. Thanks again for your on-going support.

Hairyjeremy0, 2020-02-22:
I think this graph, (which contains over 50 data points) illustrates well the challenge of accurate calibration. I have been careful to maintain a constant temperature and let the solution properly dissolve and settle before each reading, yet as you can see for some unexplained reason the tilt angle readings were erratic, compared with the analogue hydrometer. 
As you can see from the graph, at the bottom end of the scale my iSpindel's calculated SG is over 0.01 SG different from my true readings, but overall (above 20degs tilt) I am content with its accuracy. 
Big thanks for Erik & Martin for their support. I hope this thread helps anyone else battling with the same challenge. 
Happy brewing!
[image]

closed 2020-02-22 by Hairyjeremy0