universam1/iSpindel #349: iSpindel caught fire

henkexse, 2020-02-15:
I have a huge problem with my iSpindel. It caught fire and before ordering parts and building a new one I need to know if I can get help with what might have gone wrong so it can be prevented next time.
After the fermentation finished I moved the beer to a small cornelius keg and placed it in the fridge. I thought it would be a good thing to follow the temperatur in the fridge so after cleaning the iSpindel I placed it on top of the keg. When I noticed that it stopped sending data I when out to the fridge in the garage and saw that the iSpindel had caught fire that luckilly didn't spread. I have attached some pictures how it looks and appreceate if someone might give me a hint of what might have gone wrong.
When assembling the components I had to really push the iSpindel in the tube. Could there be some tensions on the boards/solderings tha broke when temeratur dropped quickly? Or do I just got a bad battery perhaps. 

[image]
[image]
[image]
[image]
[image]
[image]

universam1, 2020-02-15:
That’s impressive!
Possible the battery go overcharged, maybe by not removing the diode on the Wemos?

beikeland, 2020-02-15:
Curious to see the voltage history of the device if possibe?

(Glad you limited the damage to the iSpindel!)

henkexse, 2020-02-15:
I didn't charge it after fementation, just cleaned it and put it on the keg in the fridge. It worked fine the 2 week it floated in the beer before this. Also had it floating in waterin the fridge a week before that to see that the data gathering worked.
I think the voltage stated at 4.21V fully charged when the fermentation started and the last data I have stored is at 4.14V. These are the voltages sent from the iSpindel, don't know if they are accurate.

beikeland, 2020-02-15:
Yeah, I read it like it wasn't charged after fermenting, just curious if it could have been discharged too far as the DW01 chip goes down to 2.4v which is a bit low.

I also had an incident with a battery getting too hot for comfort but caught it before it failed catastrophically, but this was during charging, and mitigated by #335 

Another wakeup call I got was one cell holder somehow sliced the insulation of the cell causing it to short out; hard to evaluate in a postmortem, but possible the temperature drop made some minimal clearance turn into a short circuit.

Which cell and supplier did you use?

henkexse, 2020-02-15:
Ok I think it was an Panasonic ncr18650b but it might have been the ncr18650pf. I bought it from a gay making i big order from tinytronics.nl, so one of those.

beikeland, 2020-02-15:
not familiar with the supplier, but at least the pricing would suggest they're genuine cells; and using the socket should eliminate problems related to soldering wires to the cell. that pretty much leaves the charger module and or the wemos board. universam1 asked about the removal of the diode on the wemos board, might look like you have a version of the wemos d1 mini pro board are you certain its safe to retain the diode?

on the other hand there are ample postings in general about charger modules with fake TP4065 chips with varying conclusions about safety, could be worth looking into how the ones you get behave before leaving it unattended again.

henkexse, 2020-02-15:
Yeah both battery and charging module are suspects. But could be some bad soldering from my side. If the sockret + side losened and got contact with bat43 diod would that caused this?
I actually didn't know I was supposed to remove D1 diod when I built it, could that harm the battery while flashing the wemos?

thegreatgunbantoad, 2020-02-15:
With the accelerometer and the temp sensor mounted behind the battery, you have to be VERY careful to keep the legs very short when they go through the board, otherwise they can contact the battery and possibly pierce it.  I had an issue once where a temp sensor shorted across the battery casing that was a wake up call.

beikeland, 2020-02-16:
> I actually didn't know I was supposed to remove D1 diod when I built it, could that harm the battery while flashing the wemos?

The TP4065 module's output is connected to Vcc/5v on the Wemos. Removing that diode on the regular wemos d1 mini prevents 5V from the USB reaching the TP4065 ouput, and essentially the battery. Not sure if your board is the same. However he DW01 chip, should prevent damage to the battery due to overcharging, but if the cheap modules comes with fake TP4065 modules I wouldn't be surprised if the DW01 was subject to corners being cut as well. 

Would you by chance have more of those charging modules from the same supplier?

henkexse, 2020-02-16:
No I don't have any more of them, but I think it is a genuin board, it was also bought at tinytronics. Think the wemos might have been a 3.1 version. Should I aim for the 3.0?

beikeland, 2020-02-16:
For the charger the cheapest one at tinytronics says "Contains TP4056 or equivalent TC4056 (A) chip" so wouldn't be convinced they are in controll of the supply chain there; maybe consider the slightly more costly robodyn version? (Although it might not fit the pcb).

Not sure about the wemos either. V1.x has a ceramic antenna shown both at tinytronics and wemos's site - in your picture it has a pcb antenna. V2.0 does have a pcb antenna but it also has onboard lipo charger and jst connector for it + generally different size; so I'm guessing clone.

henkexse, 2020-02-16:
I think the se are the ones I got:
Wemos D1 v3. 1.0:
https://www.tinytronics.nl/shop/nl/communicatie/wi-fi/wemos-d1-mini-v2-esp8266-12f-ch340
Charger:
https://www.tinytronics.nl/shop/nl/batterij-en-accu/laders/tp4056-micro-usb-li-ion-lader-1a-met-li-ion-protection-circuit

I think it is a genuin wemos D1 v3.1.0.
Well I think I'll remove the wemos diod next time. Other than that, I don't know. Is there any effect req regarding the resistor?

beikeland, 2020-02-16:
> I think it is a genuin wemos D1 v3.1.0.

could be, between not removing the diode and it not using a ESP12 module I assumed it was the pro version. definitively remove the diode from that one.

ErikdBr, 2020-02-16:
If you don't remove the diode on the wemos you can't flash over usb with the battery attached because the battery will then get 5V unprotected.
However I don't see a relation to your iSpindel caughting fire in this situation because it was in fermentation for 2 weeks.

beikeland, 2020-02-16:
Isn't the DW01 protection ic supposed to disconnect the battery from overcharging in any case? His 4056 module did have the protection circutry.

(and a little OT, I went with removing the diode, as well as isolating the CH340 so it only gets 5v from USB allowing flashing when charging as well as reducing the stand by current a little, and debugging while running on battery)

henkexse, 2020-02-16:
I think my 4056 board did have the protection.

beikeland, 2020-02-16:
> I think my 4056 board did have the protection.

from the pictures it did; question is should it have protected the battery, and did it protect the battery,

henkexse, 2020-02-16:
As long as you don't have the battery attached when flashing there is no need of removing the diod on wemos right? Can't remember if I had it attached when flashing.

ErikdBr, 2020-02-16:
> As long as you don't have the battery attached when flashing there is no need of removing the diod on wemos right? Can't remember if I had it attached when flashing.

That I don't know for sure but once you flashed it the first time(without the battery attached), upgrades can be done over the air, so no need to use the usb on the wemos anymore, I plugged it with a silicone dust cover( https://nl.aliexpress.com/item/33019739380.html ) so I can not charge over the wrong usb port. (I can not remove it from the tube to unhook the battery cause it is fixed in the tube with epoxy)

cherryphilip74, 2020-02-21:
I think that burn point starts from BAT43 diode area or on positive BATTERY node. That's the first time and unfortunately I cannot give you any clear feedback because the burnt area hides other helpful information.

closed 2020-06-23 by stale[bot]