universam1/iSpindel #350: iSpindel will not fully boot!!!

peterd0404, 2020-02-17:
Hello All,
I currently flashed the firmware to the NodeMCU and it will not boot up completely.  I have not wired the ESP8266 board to the temperature sensor or gyro.  I wanted to test the board before doing it.  I have flashed the firmware via Windows 10 (ESP8266Flasher) and MacOS (esptool.py).  Below are the information I got from the ESP8266 from Terminal and Arduino Serial Monitor
------- Terminal------
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:f4:32:2d:ad:52
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 421296 bytes to 292016...
Wrote 421296 bytes (292016 compressed) at 0x00000000 in 26.1 seconds (effective 129.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
-----

------Serial Monitor in Arduino------
17:41:57.222 -> FW 6.0.4
17:41:57.222 -> 2.2.1(cfd48f3)
17:41:57.222 -> Worker run!
17:41:57.222 -> mounting FS... mounted!
17:41:57.258 -> scanning for OW device on pin: 5
17:41:57.296 -> No devices found!
17:41:57.509 -> scanning for OW device on pin: 12
17:41:57.545 -> No devices found!
17:41:57.791 -> scanning for OW device on pin: 116
17:41:57.791 -> No devices found!
17:41:58.051 -> scanning for OW device on pin: 121
17:41:58.051 -> No devices found!
17:41:58.271 -> ERROR: cannot find a OneWire Temperature Sensor!
17:41:58.271 -> Acc Test Connection ERROR!
17:41:58.304 -> Boot-Mode: Deep-Sleep Wake
17:41:58.304 -> ...........
17:41:59.401 -> ERROR no Wifi credentials
17:41:59.401 -> 
17:41:59.401 -> going to Config Mode
17:42:00.384 -> Acc Test Connection ERROR!
17:42:00.384 -> 
17:42:00.384 -> Exception (28):
17:42:00.384 -> epc1=0x4020e5c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
17:42:02.228 -> 
17:42:02.228 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
17:42:02.228 -> 
17:42:02.228 -> load 0x4010f000, len 1384, room 16 
17:42:02.265 -> tail 8
17:42:02.265 -> chksum 0x2d
17:42:02.265 -> csum 0x2d
17:42:02.265 -> vbb28d4a3
17:42:02.265 -> ~ld
17:42:02.303 -> 
17:42:02.303 -> FW 6.0.4
17:42:02.303 -> 2.2.1(cfd48f3)
17:42:02.303 -> Worker run!
17:42:02.341 -> mounting FS... mounted!
17:42:02.378 -> scanning for OW device on pin: 5
17:42:02.378 -> No devices found!
17:42:02.624 -> scanning for OW device on pin: 12
17:42:02.624 -> No devices found!
17:42:02.846 -> scanning for OW device on pin: 116
17:42:02.883 -> No devices found!
17:42:03.101 -> scanning for OW device on pin: 121
17:42:03.139 -> No devices found!
17:42:03.356 -> ERROR: cannot find a OneWire Temperature Sensor!
17:42:03.356 -> Acc Test Connection ERROR!
17:42:03.356 -> Boot-Mode: Exception
17:42:03.393 -> ...........
17:42:04.467 -> ERROR no Wifi credentials
17:42:04.467 -> 
17:42:04.467 -> going to Config Mode
17:42:04.467 -> reboot RFCAL
17:42:04.689 -> sd l⸮⸮| ⸮d⸮|⸮$⸮c|⸮⸮⸮⸮s⸮bd⸮#⸮⸮og⸮lng⸮⸮⸮c8⸮⸮ds$sd⸮s⸮⸮o⸮#⸮gb8 ⸮
------

Any input on this would be greatly appreciated.  I have 3 others ESP8266 that are doing the same thing.

I read where you have to build the iSpindel first for flashing the firmware.  Is this true?  Is it possible to do a reset of the iSpindel from Arduino Serial monitor or SSH?

beikeland, 2020-02-17:
> I have not wired the ESP8266 board to the temperature sensor or gyro.

won't boot until you wire it all up

peterd0404, 2020-02-23:
Thank you..  I wired the sensor up and it's working.  I think there's an issue with Philipcherry iSpindel PCB boards.  I shorted out two etch lines going to the temperature sensor on the PCB.

closed 2020-02-23 by peterd0404