universam1/iSpindel #354: Issues with configuration and calibration

jaisoh, 2020-02-29:
I'm facing two issues currently with iSpindel firmware and ESP and I think they might be related.

1. **deserializeJson() failed: NoMemory** is getting printed on the uart console after I have saved the configuration and ESP reboots. Settings are not saved e.g. voltage reading offset does not change. It seems that the configuration JSON file is not correctly written on the memory chip?

2. I'm also having issue with calibration routine. When pressing 'Calibrate' after that I keep getting asterisk printed in uart console for good and FW gets stuck. I haven't yet figured out what might be the issue.

PS. I have erased the whole memory numerous times already and it does not help with the issues mentioned above. I also tested the FW with NodeMCU V1.0 with all the sensors (different units) wired to it and I'm facing same issues with that setup also.

ragnsep, 2020-03-15:
I'm having these same issues.

I am able to load the 6.2.0 firmware on the device but unable to save settings.  Even after doing the INTERNAL://NODEMCU  // BLANK routine on 0x00000 and 0x01000 respectively.

I am using the D1 Mini 3.1.0 with the ESP8266EX chipset with 4MB flash space.

Here is a snippet.
`FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
deserializeJson() failed: NoMemory
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 69 96 16 A8 1 3C 5B
  Chip = DS18B20
  Data = 1 13 1 55 5 7F A5 A5 66 E3  CRC=E3
  Temperature = 17.19 Celsius, 
offsets not available
`

jaisoh, 2020-03-17:
I overcame the issue of NoMemory with changing line `DynamicJsonDocument doc(size * 2);` to `DynamicJsonDocument doc(size * 3);`

Don't know if it is right way to fix the issue but atleast I am able to save settings now.

I'm still having issues with calibration routine though.

thomasjec, 2020-03-29:
Any progress on this issue?
Yesterday I discovered that the wrong GY-521 chip caused all the trouble for me. I had the same deserializeJson() error showing up in the serial monitor, and the size*3 fixed it indeed.
But when I replaced the gyroscope with a new one (apparently the old one didn't have the right address, see also #302), all the errors vanished. Not only the _Acc Test Connection ERROR_, obviously, but also the NoMemory issue. The code works straight from GitHub. Calibration works without problems.
So my guess is that you might have the wrong type gyroscope?
Question remains why the gyroscope has such an impact on the memory etc.

thegreatgunbantoad, 2020-03-29:
I've had trouble with the gyros.  I tend to find that about 1 in 5 don't seem to work.  I think there are some bad batches of them coming through the supply chain at the moment.  Last lot I had were all good though.
If you hook a scope up there is a data stream but I think it's the internals that mess up sometimes.  I haven't run any of the debug stuff though.

jaisoh, 2020-03-29:
I was hoping that the original author of the project would have responded if there's any way to use these modules which are not performing as well as the authentic chip. What I found out is that the accelerometer in the MPU-6050 is probably drifting too much because when commenting out the ACC calibration method the calibration routine passes just fine.

I think that the GY-521 modules are using chips which hasn't passed quality checks and should have been scrapped. However in China nothing goes to waste and they have used those chips to build these modules which are selling cheap.

In western markets the chip itself costs roughly 7.5 euros and from China you can get the fully assembled board with 0.5 euros. Even in China the authentic chip costs more.

jaisoh, 2020-04-07:
I made another board and with genuine MPU-6050 IC and my iSpindel is working correctly now. There was no need for any tweaks on the firmware. 

However measured the DS18B20 with calibrated precision thermocouple at work and there is temperature difference of few degrees which I would like to compensate to get accurate reading. Easily done in code but as I'm going to make multiple iSpindel's I would like to use the web config to compensate for the temperature difference. That's topic for another issue/feature request though.

I'm also curious what is the link between the fake/faulty MPU-6050 IC and JSON document size (NoMemory fault on boot). It shouldn't affect at all for the memory size, any ideas?

gingerneil, 2020-06-02:
Looks like I am hitting this issue with the boards I have just built. V6.2.0 doesnt store the value and gives the same JSON error on the console. v5.x seems to work fine - and I think that was prior to the move to JSON. Is there any chance the doc size change could be included in the next release ? I could change and recompile, but that then sticks me to that 'bespoke' release.

closed 2020-08-08 by stale[bot]