universam1/iSpindel #358: Calibration and repeatability

TTornblom, 2020-03-09:
I bought a prebuilt iSpindel and I'm having problems with calibration and repeatability.

The builder had calibrated the unit in water and got a tilt of 28°.

When I try the same I get anywhere between 21 - 26° and the values at the lower end of the scale are very unreliable.

Is this normal?

I did a calibration with sugar tonight, ranging from an SG of 1.084 (67.9°) down to water (20.9°).

After I calculated a new polynomial and entered it into the unit I checked with a solution at 1.052 and the unit showed 1.057. I then tested it with water again, and got a tilt of around 25°, which with the new polynomial gave an SG of 1.015.

I believe part of the problem is that the PCB doesn't always sit in the same position in the tube when inserted and powered on. I had to open and reset the unit between the two series tonight and I believe that is part of the problem.

I also noticed a problem with calibrating the unit on a horizontal surface. If I calibrate it when horizontal and I roll it slowly between -90° and +90° I see that the tilt goes from around 90° at +-90° down to about 60° at +-45°. It seems that there is some issue with the tilt calculation and it seems like a bug to me.

As it works now I can only use the unit to see how fermentation  progresses, the absolute values are useless.

ErikdBr, 2020-03-09:
>1. I believe part of the problem is that the PCB doesn't always sit in the same position in the tube when inserted and powered on. I had to open and reset the unit between the two series tonight and I believe that is part of the problem.
> 
>2. I also noticed a problem with calibrating the unit on a horizontal surface. If I calibrate it when horizontal and I roll it slowly between -90° and +90° I see that the tilt goes from around 90° at +-90° down to about 60° at +-45°. It seems that there is some issue with the tilt calculation and it seems like a bug to me.
> 
> 3. As it works now I can only use the unit to see how fermentation progresses, the absolute values are useless.

1. The PCB should be in a fixed position within the tube so it can not shift position. I fixed it with a little bit of epoxy on the bottom of the tube. It can't be moved and it can't be taken out the tube.
2.The calibration on a flat surface should be done with the battery on the downside only and without moving the iSpindel. That should give you a tilt of about +90. When in fluid the tilt will allways be between the tilt measured in plain water and the +90, it will never reach any other tilt. So rolling it from +90 to -90(what is that even?) doesn't serve any purpose.
3. If you fix 1 and 2 and calibrate according to the "Improved method using fast fermenting" it will be pretty accurate. I get results of max. 5 points difference with only the 1.000sg measurement and the OG of the wort measured with a refractometer to calculate the polynomial. That is accurate enough for my purposes. FG I allways measure by hand after the iSpindel showed me that fermentation is done.

TTornblom, 2020-03-10:
1) Gluing the PCB to the tube makes it impossible to reach the reset button, which is a problem when I need to get it into configuration mode, no?
2) Agreed. I assume only two of the accelerators are used in the calculation and that it switches from one of the used ones to the third, unused one when I rotate it. I guess it simplifys the calculations when one doesn't have to use all three accelerators.
3) I think that the real problem is that the PCB isn't fixed in the tube. I took regular samples on my first brew with it and calculated a new polynom, but it didn't help as the readings were not repeatable.
I'll try to come up with something to keep the PCB and the tube in a fixed position.

ErikdBr, 2020-03-10:
Then you have another model of the iSpindel then I have I gather:
[image]
[image]

TTornblom, 2020-03-10:
Yes, mine looks like this https://www.opensourcedistilling.com/ispindel-assembly/

ErikdBr, 2020-03-10:
Yeah well now I get it, In the original design the d1 mini is the other way around so the usb and the reset button can be reached.
But can't you reach it with a small electrical test screwdriver?
[image]

TTornblom, 2020-03-10:
I'll see what I can do. I believe there are some wire ends near the USB connector that acts as a reset and that can be bridged with a screwdriver.

thegreatgunbantoad, 2020-03-11:
On those boards there are two pins at the end of the PCB.  Short those with a screwdriver to reset.  See my quick start guide to see a lebelled up picture: https://drive.google.com/file/d/19A07WmQ3VS27igYJXsUmOvKqMiIy_CmG/view?usp=drivesdk

thegreatgunbantoad, 2020-03-11:
That guide also has a link to an excel sheet that will give you a poly using just a water reading, based on the known profile of iSpindels I've built with those boards.  It's been very accurate for me.

TTornblom, 2020-04-03:
I think I'm closer now.
I used a bit of foam to fixate the PCB in the tube, and I used the rapid calibration calc sheet, to get a polynomial.
I also figured out that one of my problems was that the polynomial I used before was in SG, while Brewfather wants it in Plato, changing the name of the iSpindel  so that Brewfather also use SG made it work much better, although the values seems consistently low.
I started with a 1.060 wort, which was reported to be 1.050. I have taken regular samples and I've noticed that my manual readings seems high. Normally it takes about 4 days to get close to FG, but the manual reading showed 1.022. The spindel reported 1.017.
I just realized that the issue with the too high manual readings are probably due to the pressure fermentation I now use. I have about 1 bar of carbon dioxide in the beer, which is likely to skew my readings. The last batch was also about 10 points higher than expected and I was suspecting my mash to have been too warm, but the resulting beer tasted just as it should.
I will have to get rid of the CO2 in my samples I think.
I'll make a test with both a gassy and  a flat sample to see the difference.

closed 2020-06-23 by stale[bot]