universam1/iSpindel #359: Buying parts or project

i-wish-i-could-program123, 2020-03-10:
Hello,
I'm a newbie to programming and I really want a ispindle as I'm a keen brewer, where I can get all the components from, or maybe buy the project complete?

Kind regards,
Jamie

ErikdBr, 2020-03-10:
https://www.3d-mechatronics.de/en/ispindel-teilmontiert/-partially-assembled_962.html#5795048

ErikdBr, 2020-03-10:
Or you can figure out what parts you need and buy them at aliexpress.com or something

thegreatgunbantoad, 2020-03-11:
Keep your eyes peeled on eBay this weekend.  Assuming COVID-19 doesn't mess up parts deliveries again.  I've been putting them together based on the CherryPhilip PCB aimed at those who might not have the kit/time/confidence.

i-wish-i-could-program123, 2020-03-11:
Excellent let me know when they go onSent from my Huawei phone-------- Original message --------From: thegreatgunbantoad <notifications@github.com>Date: Wed, 11 Mar 2020, 08:31To: universam1/iSpindel <iSpindel@noreply.github.com>Cc: i-wish-i-could-program123 <jamieandjess54@gmail.com>, Author <author@noreply.github.com>Subject: Re: [universam1/iSpindel] Buying parts or project (#359)Keep your eyes peeled on eBay this weekend.  Assuming COVID-19 doesn't mess up parts deliveries again.  I've been putting them together based on the CherryPhilip PCB aimed at those who might not have the kit/time/confidence.

Colin66, 2020-03-28:
A full kit of all the parts can be purchased here : https://www.ebay.co.uk/itm/183849732350
Commercial sale of the completed product is in violation of the license.  
This is a DIY project and any lessons / improvements in build or code by DIY-ing can be shared to help the community.  If you have difficulties people will help :).

@Jamie you should probably remove your personal email address from that last post as it will attract spam / scammers.  
Cheers

thegreatgunbantoad, 2020-03-28:
Given that you could order a company to fabricate and and populate the cherryphilip PCB for you, selling a completed kit doesn't seem like it would be a licence breach, provided it is made clear that that is a source and build activity, the person takes no credit for the design and provides links to the source material.  That's generally the ethos of open source.  Some people would like to try the project but don't have the skills or tools to do the build stage.

The Licence only appears to hold to the software:
Amending the software is a definite no no, charging for the software is a definite no no.
Loading/sourcing software for a fee but making it clear that the software is actually free has some precedent, for example burning Open Office to a CD and charging a small fee whilst also linking to the free download option, in order to offer a physical media service.

Robbing the design concept and claiming it like this: http://www.brewbrain.nl/?fbclid=IwAR2yMxjKyeZdJ7XWBY9nHKdS-2VZCA-6fNkTKGLpGi4xo7y867oHPZxKiJE
rather takes the P, but isn't really breaching anything unless they robbed parts of the software.

Colin66, 2020-03-28:
.. and in a lot fewer words, you have built and sold around 25 of these on ebay in the last 48hrs or so for ~£750/€900 in breach of the license : "no commercial use".   
It's supposed to be a DIY, educational, non-commercial project.

i-wish-i-could-program123, 2020-03-29:
I haven't?Sent from my Huawei phone-------- Original message --------From: Colin66 <notifications@github.com>Date: Sat, 28 Mar 2020, 15:55To: universam1/iSpindel <iSpindel@noreply.github.com>Cc: i-wish-i-could-program123 <jamieandjess54@gmail.com>, Author <author@noreply.github.com>Subject: Re: [universam1/iSpindel] Buying parts or project (#359)
.. and in a lot fewer words, you have built and sold around 25 of these on ebay in the last 48hrs or so for ~£750/€900 in breach of the license : "no commercial use".
It's supposed to be a DIY, educational, non-commercial project.

thegreatgunbantoad, 2020-03-29:
@i-wish-i-could-program123  @Colin66  is not referring to you, He's upset with me.
Sorry if you got caught up in that none of this is in anyway directed at you.

thegreatgunbantoad, 2020-05-03:
Just a couple of quick notes for anyone buying or selling parts, kitting or building:

Wemos boards often tend to be reproductions if you are sourcing cheaply, in general they work just as well but so far I have had bad joints needing reflow, a diode missing, a transistor missing, so give them a check (generally if they will program from the PC you're good to go).

The gyro boards can be hit and miss I find about 1 in 10 doesn't function.  Testing these is more involved but if you are kitting/building I would definitely recommend to test every one before they are used.

The only other complex item is the charging board.  It's easy to get the wrong kind of USB fitting, but I've never had a bad board so far.

Never had a bad temp sensor, resistors, diodes, or switches.

Under the battery holder there are the back of the leads from the gyro board and the temp sensor.  Try to cut these flush before soldering or they can foul the battery holder or the battery itself.

I've thrown together some build instructions based on the Cherry Philip board (V4) attached.
[CherryPhilipV4_BuildInstructions_comp.pptx](https://github.com/universam1/iSpindel/files/4569421/CherryPhilipV4_BuildInstructions_comp.pptx)

closed 2020-07-09 by stale[bot]