universam1/iSpindel #361: Calibration question

Nr18, 2020-03-27:
So when I was calibrating the iSpindel I got stuck in the end, I have used the "Easy method" and I now have 7 tilt, gravity and temperature data points and the documentation states:

h4. Using the Online Gravity Calibration tool

> It will calculate a formula of 2nd or 3rd degree that will predict the density or the degrees of Plato given an tilt value. This formula has to be entered in Firmware to be able to get readings in density or degrees Plato.

Where my outcome is:

```
Degree 1:y = 0.005974075x + 0.849043961
Degree 2:y = -0.000286825x^2 + 0.02533112x + 0.527620881
Degree 3:y = 0.000031832x^3 + -0.003498627x^2 + 0.132475961x + -0.654489166
```

But I have 2 questions regarding this:
1. Where do I need to enter the formula?
1. The tool gives me 3 formulas and the only formula field that I could find in the spindel is in the configuration but that is only 1 field so what do I need to enter? And which field will than contain the `density` or `degrees Plato`?

h4. Using the Excel Sheet

When I enter all the data I get a nice diagram but the formula never changes, it's always: `-0.050364959*tilt^2 + 6.868698667*tilt - 185.8498673` so I did opened the excel in google sheets because I am working on a MacBook and I don't have excel so maybe the sheet got corrupted?

I have also 3 questions regarding this:
1. Why do you need to change the formula in the iSpindel firmware when you use the online tool and you don't need to when you use the excel sheet?
1. Is the excel sheet only functional in excel?
1. How can I get the formula using a MacBook/Google Sheet?

thanks in advanced!

thegreatgunbantoad, 2020-03-29:
The online tool gives you 3 polynomials.  Use the third one.  Replace all the x's with the word tilt insert * and brackets where needed (also take out the spaces):
(0.000031832*(tilt^3))-(0.003498627*(tilt^2))+(0.132475961*tilt)-0.654489166

If using the spreadsheet it must be used in excel due to it needing a macro.

I put one together that tries to guess your formula based on just a reading in water that I believe works in google sheets:
[iSpindelCPPolyEstimator.xlsx](https://github.com/universam1/iSpindel/files/4398829/iSpindelCPPolyEstimator.xlsx)

1 You need to change the formula due to the fact that each unit will hang slightly differently in the fluid and the tilt will be slightly different.
2 The excel sheet you mentioned in excel only
3 Try the sheet I posted above for non excel, if that doesn't work I'll try to do one that lets you add your own points in, you could just replace the values in columns A and E if you want to use all your points

thegreatgunbantoad, 2020-03-29:
Also on where to enter, you need to get the iSpindel to serve it's WiFi port, double tap the reset contacts.  Then connect via wifi, when you get a connection and it opens it's page go to 'Configuration' and put your calc in the 'Polynomial' box at the end.

Nr18, 2020-03-29:
@thegreatgunbantoad thanks a lot! and just in time, I have a batch ready for fermentation.

closed 2020-03-29 by Nr18

flyte, 2020-07-16:
@thegreatgunbantoad thanks for this spreadsheet! I'm surprised it's not referenced in the docs.

Can the spreadsheet be used to in lieu of full calibration in a pinch? How accurate is it likely to be compared to calibrating properly?

thegreatgunbantoad, 2020-07-16:
That's absolutely the intention.  The Data set is limited, so it's a work in progress, I have found it to be pretty close when using a CherryPhilip PCB.  Ultimately though nothing beats a true reading all these floaty things do have limited accuracy.

flyte, 2020-07-17:
Thanks for the response and the excellent work! I'll give it a shot in my next ferment and see how accurate it is on mine. :ok_hand: