universam1/iSpindel #362: Wemos stopped working after removing diode

thomasjec, 2020-03-27:
So I was upgrading my previous iSpindel build, which had an older version of Wemos on it (which I assume had too little memory and thus I had some fails on the serial monitor, like deserializeJson failed: NoMem etc.) 

But then, after I removed the diode, my computer wouldn't recognize the Wemos anymore. I'm sure this was due to the diode being removed, because I was able to put it back, and then things worked again.
So I'm guessing I removed the wrong diode, although I can't seem to find another one on the board.
Below is a picture of the board (you can see the old one in the background) and the diode in question...

Any thoughts on this or people having the same issue?

[image]

beikeland, 2020-03-27:
> But then, after I removed the diode, my computer wouldn't recognize the Wemos anymore.

that is normal as removing the Diode effectively cuts the USB power to the Wemos, you'll need to power it from the battery fully assembled after removing the diode.

thomasjec, 2020-03-27:
oh wow
that's quite definitive :)
guess I'll remove it when I get everything working then.....

closed 2020-03-27 by thomasjec